dresden-elektronik/deconz-rest-plugin #1891: Question: Case for Raspberry Pi 4 and Raspbee

ScharV, 2019-09-20:
Has anyone an idea for a case which fits for Raspberry Pi 4 and Raspbee module and which supports cooling options for the Pi?
Thanks

FezVrasta, 2019-09-20:
I got this one https://www.amazon.de/gp/product/B07QTXQ8CZ

You just need to mount the fan on the top of the ceiling rather than inside, so basically the fan will stay outside the case.

The fan has two separate pins for ground and power, so you can connect it to the availble 3.3v GPIOs

It's working good for me.

ScharV, 2019-09-20:
Thank you for your feedback

millie75, 2019-12-10:
> I got this one https://www.amazon.de/gp/product/B07QTXQ8CZ
> 
> You just need to mount the fan on the top of the ceiling rather than inside, so basically the fan will stay outside the case.
> 
> The fan has two separate pins for ground and power, so you can connect it to the availble 3.3v GPIOs
> 
> It's working good for me. (50 degrees inside a TV cabinet)

Thanks for your advice. I can only see one 3.3v pin free (no 17) as the other 3.3v and 5v are used by the raspbee component. Therefore I can’t get the fan to work. Could you point me in the right direction. Cheers

closed 2020-04-16 by stale[bot]

emresumengen, 2020-06-12:
> Thanks for your advice. I can only see one 3.3v pin free (no 17) as the other 3.3v and 5v are used by the raspbee component. Therefore I can’t get the fan to work. Could you point me in the right direction. Cheers

Hi @millie75 , I’m looking for the same answer. Did you have any updates or solutions?