dresden-elektronik/deconz-rest-plugin #1892: Cannot create a backup

nunoary, 2019-09-20:
With ConbeeI FW 26330500. On a Windows 10. deConz latest Version .69

I have problems with the Backup of my Setup via the Gateway tab in Phoscon.
I get an error "Backup failed" the header turns light red.

At the moment I'm on the latest Version .69, but I could never "backup" with previous versions. 
deCONZ is installed in C:\Program Files\deconz and the Zll.db it's on ~/local/dresden-elektronik/deCONZ. I have the userrights for this folders.

How can I find this issue and possibly solve it?

wvuyk, 2019-09-20:
Nuno,
Is this a ConBee or ConBee II? Guessing a ConBee. Tested the same config here, but with ConBee II on FW 26490700. Not sure if the current version FW for ConBee is the latest (not found on Phoscon.de)

Wim

nunoary, 2019-09-20:
Hi Wim,

It's a Conbee, and the firmware is the latest (26330500). I can't know what the problem is because there is no error log for deconz. In previous versions of deconz also could not backup, so I think the problem should be the premises or settings in the installation of deconz.

nunoary, 2019-09-27:
Sorry to insist but no one can help me with this question?
@ebaauw @manup is there any way i can check what is causing this error?

nunoary, 2019-10-09:
@jose-d
I noticed you already solved this, you can help me solve mine. I read the answer in your post and I think my problem is the same as yours but I can't understand how you solved it.

My deCONZ is installed on Windows 10 in

C:\Program Files\deconz 

and the Zll.db it's on 

~/local/dresden-elektronik/deCONZ

 I think i have the userrights for this folders, how can i confirm that? In the folder properties i have the permissions all.

manup, 2019-10-10:
Is the firmware version of the ConBee shown in the "Phoscon App > Menu > Settings > Gateway" page, meaning it is connected?

This is currently required in order to be able to create a backup.

closed 2020-02-14 by stale[bot]