dresden-elektronik/deconz-rest-plugin #1894: Aqara Motion Sensor, Aqara Switch, Xiaomi Mi Switch

merdok, 2019-09-21:
Hello,

i have some issues with those devices. I have a Conbee 2 and i paired today those devices with the stick. That works fine and the device are visible int he Phoscon app. Now i have some issues:

1. The Aqara Motion Sensor detects motion a few times and then it stops working. No motion is detected anymore and the lux value is frozen. I need then to remove the sensor from Phoscon and add it again, but after a short time again the same. Is there something i am doing wrong?
Also the temperature is not passed to HomeKit with the homebridge-hue plugin. 

2. The Aqara Switch is missing the shake gesture. The switch has 4 gestures (single press, double press, long press and shake) but shake is not present in HomeKit (homebridge-hue plugin). Can this be fixed?

3. For the Xiaomi Mi Switch the long press gesture is not exposed to HomeKit (homebridge-hue plugin). Could that be fixed?

Also it is possible to update the firmware of the Aqara devices using Conbee 2?

ebaauw, 2019-09-21:
> 1. [...]
Also the temperature is not passed to HomeKit with the homebridge-hue plugin.

By design, homebridge-hue doesn't expose `config.temperature`.  The Xiaomi on-device temperature sensor is too far off to be useful.  It should expose `state.temperature` of ZHATemperature sensors.

> 2. The Aqara Switch is missing the shake gesture. The switch has 4 gestures (single press, double press, long press and shake) but shake is not present in HomeKit (homebridge-hue plugin). Can this be fixed?

No, HomeKit only supports _Single Press_, _Double Press_, and _Long Press_.  There is no way to expose shake.  If the sensor is not recognised by homebridge-hue at all, please open an issue there copy/pasting the warning message.

> 3. For the Xiaomi Mi Switch the long press gesture is not exposed to HomeKit (homebridge-hue plugin). Could that be fixed?

Please open an issue in homebridge-hue, listing the sensor resource.  There's many different Xioami devices with the same (commercial) name

jfhenriques, 2019-09-21:
Same problem here with the Aqara Switch.

The button mapping in the Aqara Switch image is all wrong.
- Selecting "1x" button, adds empty icon of the switch, and doesn't appear to do any action
- Selecting "2x" button, adds "1" icon and does single click action
- Selecting shake button, adds "2" icon and does double click action

I can see the events arriving correctly to home assistant tho.

merdok, 2019-09-21:
@ebaauw Thanks. Will open an homebridge-hue issue regarding the switch.
But what about the Aqara Motion Sensor? That is my main issue right now. I cannot get it to work properly. That is how it looks like in Phoscon. The values are not updating anymore, i need to delete and add the sensor back again to make it work for a few times:

[image]

And is updating te firmware of Aqara devices possible using the Conbee 2?

ebaauw, 2019-09-21:
I don’t know the Xiaomi motion sensor.

If the device is upgradable (has a client _OTAU_ cluster) and you can find the firmware files somewhere, you should be able to upgrade it through deCONZ.  I don’t think I’ve seen any reports of upgrading Xiaomi devices though.

merdok, 2019-09-21:
It is the Xiaomi Aqara Motion sensor **RTCGQ11LM** and it is marked as supported here [Compatibility list](https://phoscon.de/en/conbee2/compatible), yet it doesn't fully work in my case. So i am not sure if i am doing something wrong or is this a bug?

closed 2019-09-26 by merdok