dresden-elektronik/deconz-rest-plugin #1895: Phoscon app Aqara button mapping is wrong

jfhenriques, 2019-09-22:
Same problem here with the Aqara Switch.

The button mapping in the Aqara Switch image in phoscon app is all wrong.

Selecting "1x" button, adds empty icon of the switch, and doesn't appear to do any action
Selecting "2x" button, adds "1" icon and does single click action
Selecting shake button, adds "2" icon and does double click action
I can see the events arriving correctly to home assistant tho.

Version 2.05.69 / 06/09/2019
Firmware 26330500

YKO-de, 2019-09-24:
Could you tell me the number on the back of the switch? It starts with WXKG.

jfhenriques, 2019-09-25:
I've 2 switches and the number is the same: WXKG12LM

When I first added them in deconz and configured in phoscon (I can't remember the firmware version), the mapping was right.

This issue happens only in new configurations, one of the switches is still configured in one of the groups and shake action and icon still shows ok, but if I try to configure it in another group I face the same mapping issue.

YKO-de, 2019-09-25:
Thank you for pointing that out. We will get the device and look at the error. Please excuse that the correction of the error will take some time.

gskbyte, 2019-12-01:
I think I am having similar issues, I can see the following logged when I press my linked Aqara mini switch:

```
19:51:38:970 no button map for: lumi.remote.b1acn01 ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 005,
19:51:38:971 ZCL attribute report 0x00158D0001E87E79 for cluster 0x0000, ep 0x01,
19:51:38:971 0x00158D0001E87E79 skip Xiaomi attribute 0x0000,
19:51:38:971 0x00158D0001E87E79 extract Xiaomi special attribute 0xFF01
```

My configuration is as follows:
- Aqara mini switch (Smart SwitchWXKG11LM Rev.2)
- Running deCONZ 2.05.70
- ConBee II firmware 26490700.

WhistleMaster, 2020-03-28:
Same here. Just added a new Xiaomi Aqara Switch WXKG12LM (version 20180629) in Phoscon, and the mapping is wrong in Phoscon.

Thanks !

dvbavel, 2020-04-10:
Same issue here with an WXKG11LM 2018 Smart switch. Any solution would be more then welcome! 

If anyone needs more data, please ask.