dresden-elektronik/deconz-rest-plugin #1897: Conbee II does not work

kmplngj, 2019-09-22:
I have a Conbee II Stick since some month and never got it working correctly.

Setup:
ESXi Host with CentOS Guest and Docker Stack. Newest Deconz Docker Version. I use USB Extension-Cable for the stick. 

The Gateway Information does not show a firmware version in the WebGUI.

`root@centos:/# GCFFlasher_internal -l
GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  |            | ConBee II 

root@centos:/# lsof /dev/ttyACM0
COMMAND PID USER   FD   TYPE DEVICE SIZE/OFF  NODE NAME
deCONZ   56 root   13u   CHR  166,0      0t0 45250 /dev/ttyACM0`

An Ideas what I could do?

manup, 2019-09-23:
Does host system has read/write access to the USB device? I don't have experience with CentOS but on Ubuntu/Debian the user needs to be in the dialout group to gain read/write access.

Can you provide the output of:

    ls -l /dev/ttyACM0

kmplngj, 2019-09-23:
`ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Sep 23 07:48 /dev/ttyACM0`

manup, 2019-09-23:
>Is his the problem? discovery network reply error: Error transferring http://dresden-light.appspot.com/discover - server replied: Service Unavailable - Seems to be fixed

This is just the discovery service (currently worked on), it shouldn't be the problem here.

```
deconz                 | 09:14:50:351 GW firmware version: 0x26490700
deconz                 | 09:14:50:351 GW firmware version is up to date: 0x26490700
```

That's interesting, the logs indicate that the connection is actually active and should show the firmware version in the Phoscon App.

Is it possible that multiple containers/deCONZ instances are running?
Your Browser might connected to one which isn't connected to the ConBee II.

    docker ps -a

kmplngj, 2019-09-23:
No, just one container is running.

Is it a problem that I have an Hue Bridge running in the same house?

manup, 2019-09-23:
No, that's ok. I have an idea :)

Can you please login into the container via VNC and check if the deCONZ gui shows eventually shows "Not In Network".

[image]

If so please press the "Join" button to start the network.

https://github.com/marthoc/docker-deconz#viewing-the-deconz-zigbee-mesh-with-vnc

kmplngj, 2019-09-23:
[image]

"not on network" - when I click _Join_ nothing happens.

```
deconz                 | 10:11:44:909 Current channel 11
deconz                 | 10:11:44:931 CTRL ANT_CTRL 0x03
deconz                 | 10:11:44:961 Device protocol version: 0x010B
deconz                 | 10:11:45:027 Current channel 11
deconz                 | 10:11:45:048 CTRL ANT_CTRL 0x03
deconz                 | 10:11:45:080 NET ZDO start network status 0xC4
```

manup, 2019-09-23:
0xC4 means startup failure status. This usually happens when the radio can't get a good signal, perhaps due interference with either other 2.4 GHz devices or I assume here from the PC and it's USB devices/slots itself. Can you please try another USB slots preferably a USB 2.0 one.

kmplngj, 2019-09-23:
Thanks a lot. I bought yet another usb cable and it seems to work.

closed 2019-09-23 by kmplngj