dresden-elektronik/deconz-rest-plugin #1898: Ikea Symfonisk Controller

kmplngj, 2019-09-23:
Yesterday I bought an Ikea Symfonisk Controller. Is there already support for it in Deconz?

Could I help to add support?

Bye
Jan

Kane610, 2019-09-23:
Follow the wiki to share what relevant information is needed to add support for it

kmplngj, 2019-09-23:
What if I could not connect the device to Deconz?

ebaauw, 2019-09-23:
> Could I help to add support?

Yes, please provide the info described here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support.

Intriguing little device: apparently it talks ZigBee, as it requires a Trådfri gateway, but the Symfonisk (Sonos) speakers talk SOAP over http.  I wonder how much intelligence is in the device versus the gateway.  You won’t be able to use the device to control the speakers through deCONZ (instead of through the Trådfri gateway), but it should be able to get some button events for click and double click.  Not sure where we are on exposing turn gesture (with angle).  Otherwise we could expose an analogue button event value, cf. the second endpoint of the Xiaomi cube.

kmplngj, 2019-09-24:
Turning works for dimming the Ikea Lights here, that is connect right now.

kmplngj, 2019-09-24:
Question is how to connect it to the cluster?

I tried to add it in the Phoscon WebGUI by adding it as a Switch from an "Other" manufacturer. Pressing the connection button on the Symfonisk does not do anything here.

kmplngj, 2019-09-24:
[image]
[image]

Do you need anything else?

ebaauw, 2019-09-25:
The _Basic_ cluster.  Please read the attributes before taking the screenshot.

kmplngj, 2019-09-25:
[image]

ebaauw, 2019-09-25:
I found (the last?) one at IKEA Amsterdam today, even thought they're not yet on ikea.nl.  It joins deCONZ's ZigBee network without issues.  However, it's only responsive for a very brief period after joining the network, and then goes incommunicado.  It won't wake on click, turn, or briefly pressing the reset button.

After several attempts, bombarding it with _Read Attributes_ commands to keep it awake after joining the network, I managed to bind the client _OnOff_ cluster to a group.  And then the client _Level Control_ cluster.  After that, it now behaves normally, waking up to send commands on click and turn.  It sends the following commands:
- _Toggle_ on click;
- _Step Up_ on double click;
- _Step Down_ on treble click;
- _Move Up_ when starting a right (clockwise) turn;
- _Move Down_ when starting a left (counter-clockwise) turn;
- _Stop_ when stopping a turn.

Exposing click (1002), double click (1004) and treble click (1005) will be straightforward.
As far as I can tell, the move rate is constant. The timing between the _Move_ and _Stop_ commands seems to indicate how long you're turning, but not how quickly nor how far.  Probably easiest to expose left/right turning as two long press buttons (2001/2003 and 3001/3003).

It's far from trivial to translate these ZigBee commands to Sonos commands.  There's no Sonos equivalent of Toggle, to the Trådfri hub needs to keep the play/pause state of the Sonos player.  Also there's no _Move_ / _Stop_ equivalent for volume.  There's a _RampToVolume_ command, but I don't think that can be interrupted.

I think I managed to touchlink it to my Trådfri hub (the IKEA Home Smart app showed a popup that a new controller was found).  However, the app won't find my Sonos speakers, so I cannot setup the controller to sniff the (SOAP over HTTP) commands the Trådfri hub sends to the Sonos player.

ebaauw, 2019-09-27:
Above commits add support for the sound controller.
```json
{
  "config": {
    "alert": "none",
    "battery": 16,
    "group": "1",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "390a8f3dfff393f27db48b6d845550a4",
  "manufacturername": "IKEA of Sweden",
  "mode": 1,
  "modelid": "SYMFONISK Sound Controller",
  "name": "SYMFONISK Sound Controller ",
  "state": {
    "buttonevent": 2003,
    "lastupdated": "2019-09-27T09:15:06"
  },
  "swversion": "2.1.022",
  "type": "ZHASwitch",
  "uniqueid": "14:b4:57:ff:fe:66:48:62-01-1000"
}
```
For reference, here's the _Basic_ cluster with all attributes read:
[image]

Kane610, 2019-09-27:
@ebaauw would you mind giving a detailed description of what button events the different actions generate?

ebaauw, 2019-09-27:
As I mentioned above:
- 1002 click;
- 1004 double click;
- 1005 treble click;
- 2001/2003 start/stop counter clockwise turn;
- 3001/3003 start/stop clockwise turn.

Kane610, 2019-09-27:
There are no events between start and stop rotate?

ebaauw, 2019-09-27:
No, as far as I can tell, the controller only sends _Move_ when starting to turn and _Stop_ when ending it.  Depending how smoothly you turn (or not), you get multiple _Move_/_Stop_ combos per turn.  I did see some messages in the deCONZ log that it did drop buttonevent notifications because they happened too quickly in succession, but I don’t know what to do about that (other than ignore them ;-).

I’m dying to find out what Sonos commands the Trådfri hub sends, but the IKEA Home Smart app doesn’t find my Sonos (nor Symfonisk) players, even though the Sonos app does.  Also the app crashes on my iPhone Xr (I think since iOS 13.1), but not on my iPad (iPadOS 13.1).

Kane610, 2019-09-27:
Well you can pair sonos with home assistant and bind them together :)

ebaauw, 2019-09-27:
I've been doing the same with HomeKit; I control my Sonos speakers (through homebridge-zp) using the 5-button Trådfri remote (through homebridge-hue).

gforschi, 2019-10-04:
How exactly do you pair this with deCONZ?
I'm not able to get that working

paolotremadio, 2019-10-12:
Does the Symfonisk Controller fire an event on long press?

stoneobscurity, 2019-10-24:
when will this be in the release build?

Noah-UI, 2019-10-31:
> How exactly do you pair this with deCONZ?
> I'm not able to get that working

Can't get my remote connected either. Can anyone help?

Keesromkes, 2019-11-01:
@Noah-UI - as far as I know, you can connect to it and see that it 'works' in deconz, but you can't really act on it just now. That only works if you have the IKEA bridge and take it from there.

[image]

ebaauw, 2019-11-01:
> but you can't really act on it just now

While it's not possible to control Symfonisk/Sonos players without the Trådfri hub, the controller can be used just fine with deCONZ to control other ZigBee devices.  It sends regular ZigBee commands, which are picked up by deCONZ v2.05.70 to create `buttonevent` values, which can be used in gateway rules.  In addition, you can add lights to the controller's group and control them directly (even when deCONZ is down).

> Does the Symfonisk Controller fire an event on long press?

No, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1898#issuecomment-536069107.

> How exactly do you pair this with deCONZ?

Note that deCONZ doesn't support touchlink-pairing (which is what the Trådfri hub uses).  You need to search for new devices in Phoscon/open the network in the old web app, and reset the device (pressing the reset button four times - the LED should blink).

As I mentioned before (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1898#issuecomment-535090364), the device shuts off it's radio very quickly after joining the network - it's critical that deCONZ has setup the bindings before that happens, or the device won't power back on its radio on clicking/turning.  The chances of successful pairing can be increased by powering down all routers, and pair the controller close to the RaspBee/ConBee.  If needed, keep the radio awake by sending commands to it from the deCONZ GUI while pairing.

@Keesromkes in your screenshot, deCONZ didn't receive the simple descriptors, causing the right dropdown button to be missing.  See https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2002#issuecomment-547985838 for details.

lbouriez, 2019-11-05:
Hello, I just updated to the last version of deconz (270), but I cant seems to figure how to add the switch.
I am using the UI, in switch, add other. 4x click quickly but it doesnt add to deconz.
Could you help me ?

ACrazyConcept, 2019-11-06:
What an interesting gizmo!

stoneobscurity, 2019-11-06:
> While it's not possible to control Symfonisk/Sonos players without the Trådfri hub

not true. you can have Node-Red watch for the event clicks/rotate, and then have a flow that adjusts your Sonos devices based on which action it detected.

lbouriez, 2019-11-06:
> > While it's not possible to control Symfonisk/Sonos players without the Trådfri hub
> 
> not true. you can have Node-Red watch for the event clicks/rotate, and then have a flow that adjusts your Sonos devices based on which action it detected.

I am trying to do it using appdaemon and hass automation, everything is almost working, I am just looking for the volume now, since it's just 1 event to start/stop... ([see here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2040#issue-518135765))

paolotremadio, 2019-11-06:
> I am trying to do it using appdaemon and hass automation, everything is almost working, I am just looking for the volume now, since it's just 1 event to start/stop... ([see here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2040#issue-518135765))

For that you have to up/lower the volume repeatedly until the stop event is fired. 

I’ve opted for +/- 7% of volume every 500ms. It works quite nicely.

gforschi, 2019-11-06:
It seems I have successfully paired my sonos controller, can see it in phoscon app (via VNC) and in the old webapp.
However when I listen for events (in HASSIO) I get nothing :(

gforschi, 2019-11-06:
Oh! It suddenly works! After fiddling a lot in Phoscon and trying to include it several times.
Not really sure how, but now it works and I get the events in HASSIO

stoneobscurity, 2019-11-07:
wrote a how-to for NodeRed

https://www.reddit.com/r/homeassistant/comments/dssysn/nr_rabbit_hole_part_10_sonic_attack_using_the/

Noah-UI, 2019-11-07:
> Oh! It suddenly works! After fiddling a lot in Phoscon and trying to include it several times.
> Not really sure how, but now it works and I get the events in HASSIO

Do you have a solution to correctly interpret the rotational movement in HASSIO, for example to increase/ decrease volume?

gforschi, 2019-11-07:
> > Oh! It suddenly works! After fiddling a lot in Phoscon and trying to include it several times.
> > Not really sure how, but now it works and I get the events in HASSIO
> 
> Do you have a solution to correctly interpret the rotational movement in HASSIO, for example to increase/ decrease volume?

No, that's the next thing. Hoping to see others come up with something for that 😊

ebaauw, 2019-11-08:
See #2040.  Better close this issue.

lbouriez, 2019-11-11:
> No, that's the next thing. Hoping to see others come up with something for that 😊

Try this: https://github.com/lbouriez/hassio-home-assistant_config/blob/master/appdaemon/apps/modules/symfonisk_sonos.py
For me it works pretty well

lbouriez, 2019-11-11:
Does someone know how to make the switch appears in the phoscon new app ?
I have all my switch but the symfonisk just appears in the old app.

siggi85, 2019-11-16:
Cant get the symfonisk in phoscon. I am using deconz 2.5.70 and conbee with 26330500. When I try to connect as an ikea switch just show the 5 button remote and the old dimmer. When i try to connect as "other" it doesn't work. (push connect button 4 times on symfonisk remote until led is blinking).
What am I doing wrong? I paired several other devices (ikea and xiaomi) and had not these problems before. :/ Please help.

Keesromkes, 2019-11-17:
@siggi85 @lbouriez  - in short, it won't you show up in the new phoscon (until they will support it). You can address the device through hass.io or NodeRed (both I've been too busy with other things to set up)

siggi85, 2019-11-19:
@Keesromkes Thanks for your answer. Yes you are right. In deconz directly i can see the switch and can use it via API.

But unfortunatly the usage of the wheel is not usable at the moment. Event 3001 when you start spining and when it stops event 3003. Additionaly Event 3003 comes not every time when you stop. And you can't check how long or fast you spin. Just start and stop is not enough to use it effective. Don't know how it work smooth with Symfonisk speakers directly.

lbouriez, 2019-11-19:
@siggi85 check my appdaemon script, it's not perfect, I still have trouble with the wheel sometimes but it gives you an idea on how to do until someone comes with a better solution.

ebaauw, 2019-11-19:
> But unfortunatly the usage of the wheel is not usable at the moment.

The buttonevents are derived 1:1 from messages sent by the controller.  It doesn’t report how long you spin.  See my post above.

> until someone comes with a better solution.

That some-one will have to be IKEA delivering new firmware for the Symfonisk controller, making it behave differently.  Not very likely, in my estimation.

stoneobscurity, 2019-11-19:
@siggi85 you are being too literal. it simply does not send a continuous signal as you turn it. and it probably never will. all you need to look for is a START spin code. like i said above, if you turn slowly, you get multiple START and STOP spin codes. for each START code then do volume up. (i do a +2 in my NodeRed flow, works great.)

siggi85, 2019-11-19:
I thought maybe of a stop event with the value of the spin. Maybe 3001 for start and 3500 for a half spin or something like that. But it is not. I will try the +2 Option but my idea of using it was another. I will see wether this is suitable for me. Thanks for your replys guys.

plantoschka, 2019-12-14:
Another question. I got two of those and paired them succesfully.

[image]

But now they have the same name and I can not figure out how to change the name because they don't appear as an entity in HomeAssistant and also not in the Phoscon Web App. Maybe somehow trough the Rest API?

kmplngj, 2019-12-21:
FYI

I now use the controller to control my IOTAVX Amp with NodeRed and Home Assistant via a Broadlink Remote Controller. Thanks for the work.

[image]

atze09, 2019-12-26:
Hi guys,
Is there any update in this topic?
I also have the setting hassio/conbee2/deconz/symfonisk remote and would like to control my Sonos speakers.
Regards,
Dominik

atze09, 2019-12-28:
> Another question. I got two of those and paired them succesfully.
> 
> [image]
> 
> But now they have the same name and I can not figure out how to change the name because they don't appear as an entity in HomeAssistant and also not in the Phoscon Web App. Maybe somehow trough the Rest API?

Have you already got a solution for this topic?

Smanar, 2019-12-28:
> Have you already got a solution for this topic?

You can try using the API yes https://dresden-elektronik.github.io/deconz-rest-doc/sensors/ "Update sensor", but 1 device in deconz can have multiple entries in API so IDK how it will work.

lbouriez, 2019-12-28:
If you still can access the old app (in help>old app) you will see you device.
I don't know why we can't see it in the new app :(

atze09, 2019-12-29:
> If you still can access the old app (in help>old app) you will see you device.
> I don't know why we can't see it in the new app :(

I will try it trough the API.
My old App looks exactly Like the new, what am i doing wrong?

atze09, 2020-01-05:
> You can try using the API yes https://dresden-elektronik.github.io/deconz-rest-doc/sensors/ "Update sensor", but 1 device in deconz can have multiple entries in API so IDK how it will work.

I tried it with through the API, but i was Not successful maybe i am too dumb.
I have enabled the Port 40850 in deconz. If i Type it in the Browser i get to the new webapp.
I am Missing the Connection between hass.io entity to deconz symfonisk Controller.
I tried to read the doc, but didn' t get it right.
Can you please give me a few hints, or a short Tutorial what i need to so?
Thank you in advance!

mbrrg, 2020-01-07:
> Can you please give me a few hints, or a short Tutorial what i need to so?

I managed to pair a second controller yesterday but had to rename the device through the API. Try something like this:

`curl -u 'username:password' http://x.x.x.x:7080/api -X POST -H "Content-Type: application/json" -d '{ "devicetype": "curl" }'` (fetch API token, returned as "username")

`curl http://x.x.x.x:7080/api/<api token>/sensors -X GET -H "Content-Type: application/json"` (list all sensors to find out the ID of the sensor to rename)

`curl http://x.x.x.x:7080/api/<api token>/sensors/<sensor id> -X PUT -H "Content-Type: application/json" -d '{ "name": "<new name>" }'` (rename sensor)

atze09, 2020-01-26:
> Another question. I got two of those and paired them succesfully.
> 
> [image]
> 
> But now they have the same name and I can not figure out how to change the name because they don't appear as an entity in HomeAssistant and also not in the Phoscon Web App. Maybe somehow trough the Rest API?

I renamed it via the App Hue Essentials

ajdezigns, 2020-02-03:
> FYI
> 
> I now use the controller to control my IOTAVX Amp with NodeRed and Home Assistant via a Broadlink Remote Controller. Thanks for the work.
> 
> [image]

Can you post your NodeRed Code?

kmplngj, 2020-02-04:
> > FYI
> > I now use the controller to control my IOTAVX Amp with NodeRed and Home Assistant via a Broadlink Remote Controller. Thanks for the work.
> > [image]
> 
> Can you post your NodeRed Code?

Sure. I copied just the stuff that is really needed.

```[
    {
        "id": "d952ec68.792e6",
        "type": "deconz-input",
        "z": "cac2275d.5aab68",
        "name": "Symfonisk Controller",
        "server": "9705a63e.b575c",
        "device": "14:b4:57:ff:fe:69:37:45-01-1000",
        "device_name": "SYMFONISK Sound Controller  : ZHASwitch",
        "state": "0",
        "output": "always",
        "outputAtStartup": false,
        "x": 110,
        "y": 440,
        "wires": [
            [
                "a1449ad.1aa1068"
            ],
            []
        ]
    },
    {
        "id": "a1449ad.1aa1068",
        "type": "switch",
        "z": "cac2275d.5aab68",
        "name": "Events",
        "property": "payload.buttonevent",
        "propertyType": "msg",
        "rules": [
            {
                "t": "eq",
                "v": "1005",
                "vt": "num"
            },
            {
                "t": "eq",
                "v": "1004",
                "vt": "num"
            },
            {
                "t": "eq",
                "v": "1002",
                "vt": "str"
            },
            {
                "t": "eq",
                "v": "2001",
                "vt": "num"
            },
            {
                "t": "eq",
                "v": "2003",
                "vt": "num"
            },
            {
                "t": "eq",
                "v": "3001",
                "vt": "num"
            },
            {
                "t": "eq",
                "v": "3003",
                "vt": "num"
            }
        ],
        "checkall": "true",
        "repair": false,
        "outputs": 7,
        "x": 290,
        "y": 460,
        "wires": [
            [
                "3a5dcdc8.481b92"
            ],
            [
                "1f8c9412.286444"
            ],
            [
                "2ca5c685.4dabba"
            ],
            [
                "acbd3ab8.036f08"
            ],
            [
                "a4e4c412.1a3fb"
            ],
            [
                "b0800e96.d8f64"
            ],
            [
                "a4e4c412.1a3fb"
            ]
        ],
        "outputLabels": [
            "Triple",
            "Double",
            "Single",
            "Spin Left Start",
            "Spin Left Stop",
            "Spin Right Start",
            "Spin Right Stop"
        ]
    },
    {
        "id": "a4e4c412.1a3fb",
        "type": "function",
        "z": "cac2275d.5aab68",
        "name": "Stop",
        "func": "var newMsg = { payload: \"stop\" };\nreturn newMsg;\n",
        "outputs": 1,
        "noerr": 0,
        "x": 490,
        "y": 600,
        "wires": [
            [
                "b0800e96.d8f64",
                "acbd3ab8.036f08"
            ]
        ]
    },
    {
        "id": "acbd3ab8.036f08",
        "type": "looptimer",
        "z": "cac2275d.5aab68",
        "duration": "1",
        "units": "Second",
        "maxloops": "10",
        "maxtimeout": "1",
        "maxtimeoutunits": "Minute",
        "name": "",
        "x": 680,
        "y": 560,
        "wires": [
            [
                "ec10c519.7484a8"
            ],
            []
        ]
    },
    {
        "id": "b0800e96.d8f64",
        "type": "looptimer",
        "z": "cac2275d.5aab68",
        "duration": "1",
        "units": "Second",
        "maxloops": "10",
        "maxtimeout": "1",
        "maxtimeoutunits": "Minute",
        "name": "",
        "x": 680,
        "y": 600,
        "wires": [
            [
                "18b688c.60c6877"
            ],
            []
        ]
    },
    {
        "id": "9705a63e.b575c",
        "type": "deconz-server",
        "z": "",
        "name": "Deconz",
        "ip": "192.168.188.122",
        "port": "80",
        "apikey": "91E977E002",
        "ws_port": "443",
        "secure": false,
        "polling": "15"
    }
]
```

skynet01, 2020-03-26:
@kmplngj did you change something in that code when I try to import it says its not a valid json format.

skynet01, 2020-03-26:
@ebaauw did you do something special to connect Symfonisk to Deconz? I am having trouble pairing it. I am using Home Assistant Addon and Phoscon web interface.   Here is what I am doing:

- Devices > Switches > Add new switch > Other 
- Then I inserted the battery, pressed the button next to it 4 times. 
- Symfonisk is right next to Conbee 2, pretty much touching it (plugged in into usb3 port of pi4)
- The light blinks a few times then stays solid. 

Nothing happens in the UI. Even if do an API request 
`curl http://core-deconz:40850/api/\F04AC28AFD/sensors -X GET -H "Content-Type: application/json"`
I only get Phillips device, which i assume is the virtual one that's already there
`{"1":{"config":{"configured":true,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":"36afb24d0ddea3297e6077583506aee3","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":false,"daylight":true,"lastupdated":"2020-03-26T13:52:46","status":160,"sunrise":"2020-03-26T13:20:22","sunset":"2020-03-27T01:40:48"},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:05:a1:3e-01"}}#`

I have 2 Symfonisk remotes and none of them can connect. Any idea on what could be the issue? Is it that I have a Aeotech zwave stick nearby?

ebaauw, 2020-03-26:
> plugged in into usb3 port of pi4

That’s asking for problems.  Inadequately shielded (read: almost all) USB-3 controllers and cables interfere with 2.4GHz radio, and ZigBee is especially sensitive to this.  Plug the ConBee into a USB-2 port and do **not** use the USB-3 ports.  I’ve seen some reports where people managed to use a USB-3 SSD while ConBee connected to USB-2 using an extension cable, and others where they didn’t.  Using an extension cable is a good idea anyways, as is turning off WiFi and Bluetooth on the Raspberry Pi.

> Is it that I have a Aeotech zwave stick nearby?

Not sure if ZWave also uses 2.4GHz, but “nearby” doesn’t sound good.  Again use a USB extension cable to create some distance to the ConBee.

> Nothing happens in the UI.

Do you mean in Phoscon or in the deCONZ GUI?  You want to be looking at the GUI.  I don’t know if Phoscon yet supports the Symfonisk controller (I don’t use Phoscon), but if it isn’t listed in the API, Phoscon won’t see it anyways.

skynet01, 2020-03-26:
Thanks for quick response, ill try all that!
How do you connect to the GUI (I've been using Phoscon web interface)? I know there is a VNC port for the Hassio Deconz addon but looks like it's not working when i try to use the my HA ip.

ebaauw, 2020-03-26:
You need to be running deCONZ with the GUI enabled.  Use the deconz-gui service instead of the deconz service.

skynet01, 2020-03-26:
Yep that did the trick, all paired up now. I think it was the combination of turning off wifi, bt on pi4 + using USB extension and deCONZ with GUI enabling. I left the remote next to conbee2 for about a minute, if you move it away it won't fully pair. Thanks again for quick help!

[71 more replies]

closed 2020-12-19 by close-issue-app[bot]