dresden-elektronik/deconz-rest-plugin #1899: Trådfri open/close remote does not work

pattyland, 2019-09-23:
Hi there,

 dc0d508d6a8c49d4cf59a3b72791f31f726d801b said that it fixed the Trådfri open/close remote  but no matter how often I reset the remote I cannot get to work. When start a search for new buttons I chose "other" because "IKEA" only listed the the round remote and the dimmer. Was hat right?

Greets, Sören

nunoary, 2019-09-23:
Yes it's right. 
I think it doesn't matter but I added mine as a sensor. I have four and they all work.
If you've done well, the steps are to put the phoscon app into new sensor search mode, click the command button four times and wait for it to be added. Then you can use the old web application to turn on the command with the blind.

pattyland, 2019-09-23:
Thank for the answer! I'm a bit confused with all the different ways to get things in and out of deCONZ. 
I would prefer to do all my automations in Home Assistant (for example don't close the blinds if the windowsis open) and if I understand this right, this only possible if the device is exposed in Phoscon

ebaauw, 2019-09-23:
It needs to be exposed by the REST API plugin.  Phoscon is “just” another web app using the REST API.  There’s several devices exposed by the API, but not yet supported by Phoscon.

pattyland, 2019-09-24:
@ebaauw Thanks for the clarification. How can verify if a device is exposed by the API? I always thought Phoscon would show me that.

ebaauw, 2019-09-24:
Do a GET of the /sensors resource and check whether the response contains the remote.  Alternatively, check whether you get notifications for the remote on the websocket.  I use `ph` to interact with the API from the command line, and `dc_eventlog` to monitor the websocket.  Both tools are included in homebridge-hue.  Alternatively, you could use `curl` to issue API calls.

pattyland, 2019-10-07:
On the same port as phoscon? http://phoscon-url:phoscon-port/sensors gives me a 404

closed 2020-02-11 by stale[bot]