dresden-elektronik/deconz-rest-plugin #1902: Request device support EweLink

Cyrox49, 2019-09-24:
Hi,

Could you add support for the on/off module EWELINK.

I can add other screen if you need.

Thanks

[image]
[image]
[image]
[image]

ebaauw, 2019-09-24:
I would think it’s already supported as a `/lights` resource?

Cyrox49, 2019-09-24:
Yes it is, but it show in unknow device in deconz jeedom plugin.

closed 2020-01-30 by stale[bot]