dresden-elektronik/deconz-rest-plugin #1904: Migrate from Raspbee to Conbee II

posixx, 2019-09-25:
Guys,

I want to migrate form raspbee to my ubuntu 18.04 server running on fitlet2 and use marthoc docker image for deconz. As i'm  going to use another zigbee device; will this work using backup save / load functionality within phoscon?? I want to confirm before screwing things up and be left in the dark, literally:-)

Smanar, 2019-09-26:
It works at least for standard migration > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Backup-&-Restore

posixx, 2019-09-28:
thx this worked without any issues. I just needed to power toggle some lights to be recognised again.

closed 2019-09-28 by posixx

roboticsalign, 2020-08-01:
Hi Guys,
I'm re-opening, because it seems to be broken now. 

I tried to migrate from Raspbee to Conbee 2, but it did not work with version 2.05.78. The decives showed up after import via web (conbee incl. firmware, and all devices with old status), but there was no update and no switching action. 
Poweroff, exchange conbee by old raspbee and poweron worked, so the installation itsels seems to be ok.

Smanar, 2020-08-02:
I don't think it s something important, but have you same deconz version on both machine ?
Can you show me a capture from the "hidden config" with the old config history ?

Have you tried to run deconz with debug enabled ?
>deCONZ --dbg-error=2 --dbg-info=2 

You need to add "-platform minimal" for headless version.

roboticsalign, 2020-08-02:
Thanks for your ideas. Here is my current status:

1. Rpi Zero & Raspbee: OK
2. Rpi 4 & Raspbee: OK
Here is the screenshot from the [hiden menu](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues)
[image]
3. Rpi 4 & Conbee2: NOK

- The stick comes up and is shown in the configuration:

[image]
 

- Debug Messages (some selections):
```
    12:58:47:903 Master: read param with arg 0x19
    12:58:47:911 Current channel 15

12:58:47:915 CTRL got nwk update id 0
12:58:47:919 Device TTL 2968 s flags: 0x7
12:58:49:895 Idle timer triggered
12:59:03:978 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
12:59:03:978 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
12:59:03:979 COM: /dev/ttyS0 :  (0x0000/0x0000)
12:59:03:979 dev /dev/ttyS0 (/dev/ttyS0)
12:59:03:979 GW firmware version: 0x26580700
12:59:03:979 GW firmware version is up to date: 0x26580700
```

It is also shown in the hiden menu:
[image]

roboticsalign, 2020-08-02:
That's the solution: Manually change the channel after the import in the Web UI:

[image]

Now it works. 

@Smanar: Thanks for your help!

Smanar, 2020-08-02:
Ok So I think you have a problem in yout backup file.
The backup is not working, you need to have same channel, same PANID, same Network key (the first one is grayed but Im sure it 's not the same.

So yes, you can set all the config by hand, and not just the channel.

Or try to make another backup on the woring system, and retry it on the second one.

roboticsalign, 2020-08-02:
I had to update the channel manually. The log tells me, that it is not implemented to automatically change the channel...

15:33:42:936 channel 15 does not match channel mask 0x02000000 (TODO)
15:33:42:936 **channel is 15 but should be 25, start channel change**
15:33:42:936 **Skip automatic channel change, TODO warn user**

Smanar, 2020-08-02:
And it was enough ? I don't understand how devices can work with different network ID

roboticsalign, 2020-08-02:
Probably I was not clear enough. Steps to reproduce, that worked for me:

1) backup via web ui from old raspi and Raspbee
2) plug off old raspi
3) plug on new raspi
4) restore on new raspi & new Conbee2 via web ui
**5) manually change zigbee channel via web ui**
6) reboot new raspi

The fifth step is the key.