dresden-elektronik/deconz-rest-plugin #1906: Philips Hue Filament Globe (E27)

SimplyGardner, 2019-09-25:
I have just received the new [Philips Hue Filament Globe (E27)](https://www.argos.co.uk/product/2057312) and connected it to my ConBee II stick. 

Wondering what I can do to help test to ensure it functions correctly and can be added to the support devices list?

It seems to operate fine through Home Assistant and the Deconz GUI. Although there is a 2/3 second delay when operating through HomeKit (Home Assistant >> [HomeKit Component](https://www.home-assistant.io/components/homekit/)).

Kane610, 2019-09-25:
There are some issues with Hass HomeKit integration (I've noticed poor color mappings) that I will investigate further down the road

SimplyGardner, 2019-09-25:
Awesome - thanks @Kane610. Great to hear that the HomeKit component is still getting maintained :)

Kane610, 2019-09-25:
Well Im doing it from the angle of making sure that the deCONZ integration works as good as possible :)

SimplyGardner, 2019-09-25:
Understood and you are doing an awesome job - saw your contribution to the Ikea blinds too!

Kane610, 2019-09-25:
Haha, thanks! I was mostly harassing deCONZ upstream developers :P

closed 2020-01-31 by stale[bot]