dresden-elektronik/deconz-rest-plugin #1907: Aqara Motion Sensor stops detecting motion and not responding

merdok, 2019-09-26:
Hi,

I have an Aqara Motion Sensor (**lumi.sensor_motion.aq2**) and i can without problems connect it to my Conbee 2. The sensor works and detects motion.
Unfortunately after some time (usually a few hours) it stops detecting motion and a after a few more hours it is unreachable in the Phoscon app. No motion is detected anymore and the lux value is frozen. I need to start a search for new sensors and reset the sensor with the physical button to make it reachable and detect motion again for a few hours.
Is this a bug with the sensor or am i doing something wrong?

Smanar, 2019-09-27:
Are you using headless version or GUI version ? If it's with GUI, check connexion on deconz.
Are you using a router for this device ? a plug ?
If the sensor is mobile try to put it closer to the gateway to check if it's not a connexion problem.

merdok, 2019-09-27:
I am using headless and have no access to the GUI.
There are currently no routers connected to the Conbee 2. Just the one Aqara Motion Sensor and 2 Aqara switches.
But the sensor works initially, so i do not think that this is a distance problem, right?

Smanar, 2019-09-27:
It can, if you have a bad connexion, the device can connect but will disconnect after some time.
There is some tips to try to improve connection, like using an USB extension cable.

merdok, 2019-09-27:
Are bulbs(ikea or philips hue) considered as routers? I plan to add those to my setup.

Smanar, 2019-09-27:
Yep, and the better one.
But Xiaomi device don't change parent, so if you add a bulb between the gateway and the sensor, remake a pairing procedure.

merdok, 2019-09-28:
Hmm, will check that out. But still the sensor is not that far away from the Conbee 2 so it is kind of a strange behavior that it just doesn't detect motion anymore after some time...
Isn't this an issue that the sensor is going to sleep and does not connect anymore to the Conbee 2?
I mean the switches work fine even when they are much more further away from the Conbee 2.

Smanar, 2019-09-28:
All xiaomi devices (battery one)  are sleeping, motion sensors and switches, it's something normal. As you don't have router, it's can't be the problem (some device are bad router).
Hard to check link quality without GUI, but to make tries, just put the sensor close the gateway to make test.

BTW are you sure you have the last version (firmware and deconz) ?

merdok, 2019-09-29:
So i have put the sensor straight next to the Conbee 2 and still the same issue, so it can't be a distance problem...

Yes, i am using the latest firmware and deconz.

ebaauw, 2019-09-29:
Xioami end devices stick to their parent router like... well, something sticky.  You need to reset and re-pair them after moving so they connect to a router close to their new location.

merdok, 2019-09-29:
I did the repair and placed it next to the Conbee 2 and it is not responding anymore. I have to again repair the sensor to make it work for a few hours... I mean if this would a distance issue then the sensor should not be even able to connect to the Conbee 2, but it connect fine. It just stops working after some time...

wilfredsmit, 2019-10-07:
Just to be sure: Have you checked the battery level?

merdok, 2019-10-08:
Yes, battery level is full. But i think that i have found the issue. My Conbee 2 was connected directly to the Raspberry Pi. It seems this did not affect the pairing since all devices did pair without issue, but i guess it has trouble to maintain the connection. Anyway i got a usb extension cable and works without issues since a few days already so i think this was the issue.

closed 2019-10-08 by merdok

alonjr, 2020-04-12:
Hello, @merdok, have you found a solution for this issue?

I've been having this issue for a while. Now, after weeks of resetting the sensor and re-binding it to the zigbee network, this is for sure a problem. 

I have access to the deConz zigbee network GUI and was monitoring the nature of the sensor's connection status all along. Here are the steps I take to get the sensor up an running after it fails to respond:
1. I delete the sensor through the Phoscon WebUI
2. I press scan for new sensor via the Phoscon WebUI
3. I reset the sensor by punching in the reset hole on the device for 10 seconds.
4. Sensor is detected by deConz and starts functioning immediately.

Anywere between a few hours to a couple of days, the sensor no longer reacts to motion. In the GUI graph of the Zigbee network, you can see that the sensor is still connected (to Conbee directly) but does not show any signs of life. Then a little later, hours or maybe a day, the sensor becomes orphan and not connected to any device on the network, see image below
[image]

I was unable to find any information on this behavior and was hoping that someone here can give some tips. I'm happy to work with whomever to get to the bottom of this!

thank you
Alon

dxxxm, 2020-04-12:
conbee I, deConz 2.5.75, FW 0x26330500 
Got two of them. Just double checked event history for both.  Cannot confirm either of the issue.
Both responsive as long as I looked back in history. 
Known Xiaomi zigbee implementation quirk: battery powered devices cling to the parent router node paired with during initial search phase forever. If parent node drops out linked Xiaomi sensor won't try to search for an alternative one. Better explanation of this behaviour explained in numerous posts by @ebaauw.

merdok, 2020-04-12:
I did not fix my issue with the Aqara motion sensor. I got myself the xiaomi motion sensor (the one without lux measurement) and this one works stable since i got it a few months ago.

alonjr, 2020-04-12:
@dxxxm, I have the exact conbee/deConz configuration and the Xiaomi motion sensor (RTCGQ01LM). You referenced @ebaauw as having explained the behavior. I'd appreciate it if you could point me to one of his explanations as I could not find any. tnx in advance.

dxxxm, 2020-04-13:
@alonjr, random picks:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/677#issuecomment-402997990
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2527#issuecomment-593636770
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2527#issuecomment-594165921

alonjr, 2020-04-14:
@dxxxm, thanks i appreciate it :)

Jafalex, 2021-10-09:
Hey, did anyone find any solution to this issue? Still happening two years later 🤷🏼‍♂️

benblb, 2021-11-02:
I'm also running into trouble with Aqara motion sensors. Sometimes the motion sensors respond, sometimes they don't work for a minute or two. I do not, however, need to repair so I believe mine stay connected. No errors in log.

Tonnie-P, 2021-11-28:
Same problem here, seems to be worst the larger my network grows
it worked sort-of fine when i had like 20 lamps connected, but now with about 40+ nodes a lot of issues.
Also with the Xiaomi buttons.

Replaced the buttons with the Ikea buttons and they work perfect, probably going to change the motion and temp humidity sensorts for other brands.

pedropaz-pt, 2022-10-16:
same problem here. Randomly disconnects and stops reacting.
Anyone found the solution?

H35am, 2024-03-19:
I'm also running into trouble with Aqara motion sensors, after two years of usage without any problems! Battery levels are above 75% (that one is the lowest the other two are above 80%).
Sometimes the motion sensors respond, sometimes they don't work for a minute or two. The quick fix for the day is to switch the lightbulb manually off and on again.

Smanar, 2024-03-19:
Hello, you act on a lightbulb to repair a motion sensor ?
Try replacing batteries, at 70% they are dead.

H35am, 2024-03-19:
Hi, thanks for your advice. But 70% remaining is not even close to half, how is this an empty indication?

Smanar, 2024-03-19:
>But 70% remaining is not even close to half, how is this an empty indication?

Battery discherge is not something proportionnal. You have 100% during 2 years, 90 % during one year and 90 to 20% in the next year for exemple.
But what are you calling the switch ? the button on the device ? I don't see the "connection" sensor/switch.

H35am, 2024-03-19:
Thanks for your explanation.
I mean a physical light switch; all my 3 Aqara sensors are behaving very weird and not like 2 months ago. .

Smanar, 2024-03-19:
So you have issue on the sensor AND on a switch device ?

What is your deconz version ? The code have moved a lot recently.
Conbee version ?
Do you are using the last firmware ?

Make a try on one sensor with new batterie.