dresden-elektronik/deconz-rest-plugin #1909: zigbee: 'Install codes' support

imammedo, 2019-09-27:
I wonder if it's possible to add to deconz  'Install codes' as per BDBv1.0 spec (chapter 10.1),
(coordinator part of it)?

manup, 2019-09-27:
We have a internal prototype for Install Codes which already works with a few devices.

The current REST-API interface was added in commit https://github.com/dresden-elektronik/deconz-rest-plugin/commit/24c7ff9d3047b7a6bde698e56d734120b63025ee. Note that it requires the `hashing-cli` executable from the Zigbee Alliance.

We hope to release it in 2020 with a simple user interface in the Phoscon App.

Adminiuga, 2019-11-01:
@manup 
two questions:
1. what Zigbee devices with install code have you used to test with?
2. For deCONZ serial protocol, how do I add the derived key to ConBee?

closed 2020-03-07 by stale[bot]