dresden-elektronik/deconz-rest-plugin #1910: TRADFRI motion sensors strange behavior

A320Peter, 2019-09-27:
I observed strange behavior of IKEA motion sensors in the last weeks. I am not sure since which Deconz update, but I didn't have this problem before and I doubt that all of my IKEA motion sensors are 'dying' at the same time while other motion sensors like Xiaomi Aquara work perfectly on the same network. I have models `E1525` they are set to `1 minute` and `DAY` mode. It looks like randomly but very frequently they are detecting motion all day, while there is actually no motion. Although they detect true motions as well without problem. They are in different places in the house. All of them have the same behavior but not all of them turning on with the same frequency. In average every 15 minutes they detect motion at least once. Previously I never had problems with them. Here is a screen snippet one of the sensor in a few hours interval: 

[image]

I tried to reset them, even removed the batteries. I removed/added them to the network again. They work when they should but they also detect motion when they should not.

I don't have IKEA Gateway or any other IKEA stuff except a FYRTUR directly connected to Deconz which works perfectly. I have `ConBee` with version `2.05.69` and firmware `26330500`. I run hassio 0.99 on a Pi 3b+.

I have no idea if this could be a Deconz problem or my TRADFRI sensors are dying after a year or there is some interference with something else.

ebaauw, 2019-09-27:
Most likely battery related.  Don’t be fooled by the percentage reported.  Cheapest replacement batteries (by far!) are actually at IKEA.

A320Peter, 2019-09-29:
I have replaced the batteries with quality ones and problem solved. Cheap battery with cheap sensor case... thanks a lot!

closed 2019-09-29 by A320Peter