dresden-elektronik/deconz-rest-plugin #1911: GLEDOPTO 2ID RGB-CCT Controller -- Scenes don't work properly

ab1470, 2019-09-28:
#### Issue

I'm using a 2ID GLEDOPTO GL-C-008 controller in my smart home setup, which runs off of a Raspberry Pi with a RaspBee module, and connects to Homekit using Homebridge and Homebridge-Hue. Because I'm using the 2ID version, the controller is exposed to deCONZ as two separate devices: a GL-C-006 for controlling the white spectrum LEDs, and a GL-C-007 for controlling the RGB LEDs.

The controller does not seem to handle simultaneous commands properly when using the Phoscon app. It's configured in Phoscon as "Wall Light White" (for the white spectrum LEDs) and a "Wall Light Color" (for the RGB leds), and I have two test scenes to control these lights. The scenes are as follows:

[image]
[image]

When I activate the TestColorOnly scene in Phoscon, Wall Light Color comes on as expected, but Wall Light Color _also_ comes on. Initially, the Phoscon GUI indicates that the Wall Light Color is turned off, but after about 5-10 seconds the GUI updates and shows (correctly) that Wall Light Color is on. I'm getting similar behavior with the TestWhiteOnly scene.

I verified that the scene was set properly in deCONZ by querying the API, which gave me the following response:
```
{
    "lights": [
        {
            "bri": 255,
            "colormode": "xy",
            "id": "8",
            "on": true,
            "transitiontime": 10,
            "x": 0.207599,
            "y": 0.0729992
        },
        {
            "bri": 100,
            "colormode": "ct",
            "ct": 158,
            "id": "9",
            "on": false,
            "transitiontime": 10
        }
    ],
    "name": "TestColorOnly",
    "state": 0
}
```
(I also checked the TestWhiteOnly scene with the API, and it returns the expected result).

You might have noticed that the JSON above shows that, for the TestColorOnly scene, the Wall Lights White is set to 'off', but the brightness is set to 100. I did try manually setting the brightness to 0 (via the API). I then activated the TestColorOnly scene from Phoscon after doing so, and it set Wall Lights Color to 0%, and Wall Lights White to 4%.

Interestingly, I have the same two test scenes configured in the Apple Home app, and the lights behave as expected when I activate those scenes via the Home app or via Siri (i.e., TestColorOnly turns on the RGB LEDs and turns off the White LEDs).

#### Log Messages

I set the brightness to 50% for both Wall Light White and Wall Light Color, then activated the TestColorOnly scene from Phoscon. Homebridge returned the following log:
```
[9/27/2019, 11:12:20 PM] [Hue] Wall Light Color: state changed event: {"alert":null,"bri":254,"colormode":"xy","ct":417,"hue":45824,"on":true,"reachable":true,"sat":245,"xy":[0.2076,0.073]}
[9/27/2019, 11:12:20 PM] [Hue] Wall Light Color: light bri changed from 127 to 254
[9/27/2019, 11:12:20 PM] [Hue] Wall Light Color: set homekit brightness from 50% to 100%
[9/27/2019, 11:12:20 PM] [Hue] Wall Light White: state changed event: {"alert":null,"bri":254,"colormode":"ct","ct":158,"on":false,"reachable":true}
[9/27/2019, 11:12:20 PM] [Hue] Wall Light White: light bri changed from 127 to 254
[9/27/2019, 11:12:21 PM] [Hue] Wall Light White: set homekit brightness from 50% to 100%
[9/27/2019, 11:12:21 PM] [Hue] Wall Light White: light on changed from true to false
[9/27/2019, 11:12:21 PM] [Hue] Wall Light White: set homekit on from 1 to 0
[9/27/2019, 11:12:22 PM] [Hue] Phoscon: gateway request 201: get /sensors
[9/27/2019, 11:12:22 PM] [Hue] Phoscon: gateway request 201: ok
[9/27/2019, 11:12:22 PM] [Hue] Phoscon: gateway request 202: get /lights
[9/27/2019, 11:12:22 PM] [Hue] Phoscon: gateway request 202: ok
[9/27/2019, 11:12:27 PM] [Hue] Wall Light White: state changed event: {"alert":null,"bri":254,"colormode":"ct","ct":158,"on":true,"reachable":true}
[9/27/2019, 11:12:27 PM] [Hue] Wall Light White: light on changed from false to true
[9/27/2019, 11:12:27 PM] [Hue] Wall Light White: set homekit on from 0 to 1
[9/27/2019, 11:12:27 PM] [Hue] Phoscon: gateway request 203: get /sensors
[9/27/2019, 11:12:27 PM] [Hue] Phoscon: gateway request 203: ok
[9/27/2019, 11:12:27 PM] [Hue] Phoscon: gateway request 204: get /lights
[9/27/2019, 11:12:27 PM] [Hue] Phoscon: gateway request 204: ok
```
Notice that Wall Light White initially gets set to 'off', then 6 seconds later (with no action from me), the status changes to 'on'

Next, I repeated the same procedure as above, but this time I activated the scene via the Apple Home app. Here's the log:
```
[9/27/2019, 11:18:06 PM] [Hue] Wall Light Color: homekit brightness changed from 50% to 100%
[9/27/2019, 11:18:06 PM] [Hue] Wall Light White: homekit on changed from 1 to 0
[9/27/2019, 11:18:06 PM] [Hue] Phoscon: gateway request 384: put /lights/8/state {"bri":254}
[9/27/2019, 11:18:06 PM] [Hue] Phoscon: gateway request 385: put /lights/9/state {"on":false}
[9/27/2019, 11:18:06 PM] [Hue] Phoscon: gateway request 384: ok
[9/27/2019, 11:18:06 PM] [Hue] Wall Light Color: state changed event: {"alert":null,"bri":254,"colormode":"xy","ct":417,"hue":45824,"on":true,"reachable":true,"sat":245,"xy":[0.2076,0.073]}
[9/27/2019, 11:18:06 PM] [Hue] Phoscon: gateway request 385: ok
[9/27/2019, 11:18:06 PM] [Hue] Wall Light White: state changed event: {"alert":null,"bri":127,"colormode":"ct","ct":158,"on":false,"reachable":true}
[9/27/2019, 11:18:08 PM] [Hue] Phoscon: gateway request 386: get /sensors
[9/27/2019, 11:18:08 PM] [Hue] Phoscon: gateway request 386: ok
[9/27/2019, 11:18:08 PM] [Hue] Phoscon: gateway request 387: get /lights
[9/27/2019, 11:18:08 PM] [Hue] Phoscon: gateway request 387: ok
```

Thiesy65719, 2019-09-28:
Looking at the screenshots I see the slider for single light / all lights 
in position 'All' for TestColorAll and
in position 'Single' for TestWhite.
Is that done on intention?

ab1470, 2019-09-28:
Not intentional, but it doesn't make a difference either way. I'm experiencing the same problems as in my OP regardless of whether the slider is set to single or all lights.

Thiesy65719, 2019-09-28:
I am using 3 pieces of GLEDOPTO GL-C-008S
The instructions seem more or less generic, I can't seem to find detailed infos. Do you have a recommended source for more documentation?
Mine shows up with single ID. Did you need to activate 2 ID mode?

[image]

Thiesy65719, 2019-09-28:
I have an problem to get my zigbee light control running and need to change the raspi sd card to an basic setup to experiment with it. Using raspian buster, deconz headless and phosconapp.
I have the same problem with defining scenes. Needed to change my setup.

Using my 3 pieces of GL-C-008S I defined three scenes:

a: minimum bright,   max bright red, max bright green      (slider single + all selection in rgb mode)
b: max bright blue,   minimum bright     minimum bright  (slider single + all selection in rgb mode)
c::  all three lights in white mode full bright mid colortemp (slider all selection in white mode)

when defining the secenes I pushed store after each single setting.
after I was done defining one scene I went over the lights in the scene again and the settings of that light were confirmed.

After I was done with scene c  I rechecked scene a again and they now showed white mode iso the rgb mode I had defined.

I have read the full configuration from rest-api and the scenes dictionary is empty. In the groups are the names of the scenes. I am presently running out of time, need to leave home.

ab1470, 2019-09-28:
> I am using 3 pieces of GLEDOPTO GL-C-008S
> The instructions seem more or less generic, I can't seem to find detailed infos. Do you have a recommended source for more documentation?
> Mine shows up with single ID. Did you need to activate 2 ID mode?
> 
> [image]

GLEDOPTO sells two versions of its RGBCCT controllers: a 1ID version and a 2ID version. If you have the former, it's not possible to have it expose two endpoints. There's an explanation of the difference here: [r/Gledopto](https://www.reddit.com/r/Gledopto/comments/8spz43/which_gledopto_controller_do_i_need/).

Thiesy65719, 2019-09-30:
I have been exploring scenes with phoscon and get get it working either. I configure and store a couple of scenes. When reopening them they have changed. Also the restapi reveals nothing that looks similar to the scenes previously stored.

I noticed that GLEDOPTO is not listed in the supported devices.  
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices

MrMiracles, 2019-11-02:
I have the some problem. Weird thing is some scenes work and others don't. I have made two scenes, both are identical. Scene with ID 3 works perfectly (and turns both GLEDOPTO lights outs) and the other does nothing. 

Nothing as in: it changes the on flag to "false" but the lights stay on and after like 5 seconds the on flag is returned to "true". Super weird. Same as with @ab1470  

This is the scene I use which works on ID 3 but not on ID 4:
```
{
  "lights": [
    {
      "bri": 248,
      "colormode": "xy",
      "id": "1",
      "on": false,
      "transitiontime": 10,
      "x": 0.602976,
      "y": 0.343694
    },
    {
      "bri": 126,
      "id": "2",
      "on": false,
      "transitiontime": 10
    }
  ],
  "name": "kitchen out",
  "state": 0
}
```

Can't figure out why at the moment, someone an idea?

closed 2020-03-08 by stale[bot]

ab1470, 2020-06-17:
Reopening this because I'm still experiencing this issue.

ab1470, 2020-06-17:
I'm still experiencing this issue. Can this be reopened?