dresden-elektronik/deconz-rest-plugin #1912: Lots of errors in logs

Loader23, 2019-09-28:
Hi,

Iam having alot of these errors in my logs, can you tell me what they mean?
Most is 0xE9 but sometimes also 0xA7
 Thanks!

12:11:49:185 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:11:54:274 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:11:58:699 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xE9 on task 
12:12:02:997 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:12:08:489 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:12:09:294 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xE9 on task 
12:12:15:777 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xE9 on task 
12:12:25:779 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xA7 on task 
12:12:28:386 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:12:31:895 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:12:42:647 Current channel 15 
12:12:42:650 Device TTL 1283 s flags: 0x7 
12:12:45:778 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xE9 on task 
12:12:46:486 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xE9 on task 
12:12:48:185 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xE9 on task 
12:12:48:884 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:12:53:396 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xE9 on task 
12:12:54:195 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:12:58:680 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xE9 on task 
12:13:04:891 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:13:14:983 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xE9 on task 
12:13:17:684 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:13:24:796 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:13:25:594 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xE9 on task 
12:13:28:354 0x7CB03EAA0A003CDF error APSDE-DATA.confirm: 0xE9 on task 
12:13:40:097 0x7CB03EAA00B26CE7 error APSDE-DATA.confirm: 0xE9 on task

manup, 2019-09-29:
I've added a new wiki page [Zigbee Error Codes in the Log](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log) to summarise the error codes.

Loader23, 2019-09-29:
Thanks alot, I have changed the position of the stick a little bit and don't see these errors anymore. 👍

closed 2019-09-29 by Loader23