dresden-elektronik/deconz-rest-plugin #1914: How to find new lights via the api?

nodefeet, 2019-09-28:
I updated my conbee/deconz to the latest firmware/software. 

Now I can no longer get the new lights after I factory reset them (with touchlink reset) via the api. I used to get them via a GET all lights requests. It did work with previous versions of deconz (I think is was version 2.05.58. I can provide more detailed information if necessary)

I am still able to find new lights via the phoscon app. But do you have any explanation or reasoning why it is no longer possible via the api?

bdickhaus, 2019-09-28:
Not sure how it worked before, but I do this to regain access to Hue bulbs that were powered off/on and get stuck as unreachable. I have a script that issues  /config {"permitjoin":120} to the API.

nodefeet, 2019-09-29:
Thank you  @bdickhaus .
I totally forgot about the „permitoin“ option.
Sorry for all the fuss.

closed 2019-09-29 by nodefeet