dresden-elektronik/deconz-rest-plugin #1915: Problems after moving Conbee over to Raspbian

StoricU, 2019-09-30:
Hi,

I have been running Conbee on my windows computer together with JowieHue/Homeseer.
Has been working great.

But with a new server, I need to move the Conbee to a raspbian and better location.

I have gotten everything set up and at first sight everything looked to be working, but not at a closer look.

The connection between the lights and remotes are very unstable. Some of the lights won't connect at all.
Some of the lights connect, but after a while they just disconnect and the gateway does not respond when using my IKEA remotes.

Everything worked before the transfer on the windows gateway.

I'm totaly green on raspbian and linux, so hoping that someone may have some tips regarding getting it as stable as it was on the windows computer.

StoricU, 2019-09-30:
Have tested some more and it seams that even though the gateway shows there is a connection, when trying to operate the lights I only get timeout.
So I'm rarely able to operate the lights.

StoricU, 2019-09-30:
After hitting another wall, I have now moved the ConBee back to the computer and restored the backup I took before moving it the first time.
Have the same problem as on the Pi. Not able to add any new lights and the lights that states connected, only timesout when trying to operate.

I'm really at a dead end here. Any help is very much appreciated!

Smanar, 2019-09-30:
Try an older backup if you have one, or take a look on hidden configuration https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1849#issuecomment-530287817

But first check on Phoscon the gateway state.

StoricU, 2019-09-30:
> Try an older backup if you have one, or take a look on hidden configuration [#1849 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1849#issuecomment-530287817)
> 
> But first check on Phoscon the gateway state.

[image]
[image]

I don't have a older backup. I have just a few lights and switches for testing until I got deconz over to a Raspberry Pi.
What do I have to do to start completely clean?
Can I reset the ConBee in addition to deConz?

Don't care if I loose anything now, I just want it back online again

Smanar, 2019-09-30:
I m sorry, I don't see something anormal.
Perhaps can you compare your deconz setup with this one https://user-images.githubusercontent.com/4129909/55972478-daa18380-5c83-11e9-8ad4-761e5c75c65c.jpg

StoricU, 2019-09-30:
> I m sorry, I don't see something abnormal.
> Perhaps can you compare your deconz setup with this one https://user-images.githubusercontent.com/4129909/55972478-daa18380-5c83-11e9-8ad4-761e5c75c65c.jpg
> 
> You haven't changed something close to the gateway ? a new Wifi router ?

The settings is the same, except PANID , tc address and network key.

The only difference is that I have tried to use the ConBee on 2 different windows machines and 1 Raspberry Pi now that I'm trying to sort this out.

I followed this guide: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/516#issuecomment-429564761 but no luck. Even with the ConBee 10cm from the lightbulb, I can include it.

Smanar, 2019-09-30:
So all is same.
Yes, but it was working on first machine (the windows one) so no reason for it stop working.
Perhaps a problem in the backup, but I can't help you here.
It's like you have a connection problem (timeout / disconnect / problem for pairing) but it was working previously.

When you said you can't include it, the bulb doesn't blink or nothing appear in phoscon ?

StoricU, 2019-09-30:
> When you said you can't re-include it, the bulb doesn't blink or nothing appear in phoscon ?

Correct. Nothing happens at all.

StoricU, 2019-10-01:
Uninstalled deconz and went back to 2.05.65 and everything is working fine. Just going from .69 to .67 did not fix it.
Did not try .66. 
Now I'm not touching anything..

closed 2019-10-01 by StoricU

Smanar, 2019-10-01:
Lol, I can understand, but something strange.
On your 2 screenshoots, on phoscon you are on 2.05.69 but on the hidden configuration the last one memorised is 2.05.66 and this part is updated at every deconz update.

StoricU, 2019-10-01:
Yes, I have no idea what has happened.
When I had the trouble on Raspbian, it was using .67.
As far as i know I have never installed .66 on any of my machines.
Just happy it works now.
Good thing IKEA's new screens is sold out. So I don't need to think about upgrading ;P