dresden-elektronik/deconz-rest-plugin #1917: Zigbee network stopped working

joch, 2019-10-01:
I’m having an issue where everything suddenly stopped responding after having been solid for more than a year. Now getting lots of these in the log. A few A7, but lots of D0:

```
deconz       | 07:19:25:335 0x90FD9FFFFE030A51 error APSDE-DATA.confirm: 0xD0 on task
deconz       | 07:19:36:226 0x7CB03EAA00AF303B error APSDE-DATA.confirm: 0xD0 on task
deconz       | 07:19:47:122 0x84182600000FC393 error APSDE-DATA.confirm: 0xD0 on task
deconz       | 07:19:58:032 0x84182600000FFA39 error APSDE-DATA.confirm: 0xD0 on task
deconz       | 07:20:14:501 Current channel 25
deconz       | 07:20:14:533 Device TTL 3342 s flags: 0x7
deconz       | 07:20:22:684 0x00178801042D5F01 error APSDE-DATA.confirm: 0xA7 on task
deconz       | 07:20:47:045 0x001788010425DB8E error APSDE-DATA.confirm: 0xA7 on task
deconz       | 07:20:57:916 0x84182600001034ED error APSDE-DATA.confirm: 0xD0 on task
deconz       | 07:21:08:822 0x000D6F001065708D error APSDE-DATA.confirm: 0xD0 on task
deconz       | 07:21:14:499 Current channel 25
deconz       | 07:21:14:528 Device TTL 3282 s flags: 0x7
deconz       | 07:21:19:735 0x000B57FFFED965EA error APSDE-DATA.confirm: 0xD0 on task
deconz       | 07:21:30:623 0x086BD7FFFE2F6786 error APSDE-DATA.confirm: 0xD0 on task
deconz       | 07:21:55:401 0x086BD7FFFE2E156A error APSDE-DATA.confirm: 0xA7 on task
deconz       | 07:22:14:495 Current channel 25
deconz       | 07:22:14:527 Device TTL 3222 s flags: 0x7
deconz       | 07:22:19:753 0x84182600000D9CCD error APSDE-DATA.confirm: 0xA7 on task
deconz       | 07:22:44:219 0x7CB03EAA00A5B1C6 error APSDE-DATA.confirm: 0xA7 on task
deconz       | 07:23:08:702 0x000B57FFFED22DB6 error APSDE-DATA.confirm: 0xA7 on task
deconz       | 07:23:14:498 Current channel 25
deconz       | 07:23:14:533 Device TTL 3162 s flags: 0x7
deconz       | 07:23:33:146 0x000B57FFFED8E662 error APSDE-DATA.confirm: 0xA7 on task
deconz       | 07:23:57:274 0x0017880103896FC4 error APSDE-DATA.confirm: 0xA7 on task
```

Anyone know what the issue could be, or any way of debugging this further?

(added this as a comment in an old issue at first, but moved it to a new one)

manup, 2019-10-01:
For a description of the error codes refer to https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

Can you also please provide more details about your system:

* ConBee, ConBee II or RaspBee?
* Operating system?
* Do you use a USB extension cable?
* Is it a Raspberry Pi and if so are there any other USB devices attached?
* Is the device located nearby any Wi-Fi router?

joch, 2019-10-01:
Thanks for the link to the wiki. Now I at least know what it means.

* It's a Conbee running on Linux and Docker.
* No extension cable. Tried using one after seeing similar issues, but made no difference.
* Not a Pi, using an old Mac.
* Located fairly close to a wi-fi router, but it's been there for more than 1,5 years without these kind of issues. Can try moving the entire computer somewhere else tonight.

I have attached a debug run of deconz in [deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3675243/deconz.log) and looking through it, I saw something interesting.

```
08:40:45:580 CTRL. 3.16.208:40:45:606 COM: /dev/ttyUSB0 :  (0x0403/0x6015)
08:40:45:606 COM: /dev/ttyACM0 :  (0x1CF1/0x0030)
```

`/dev/ttyACM0` is my z-wave hub. Again, the setup has been the same for ages, so why this started happening now is a mystery. I'll try disconnecting it when I get home and see if that changes anything.

manup, 2019-10-01:
Ah ok I guess the device file names are just dynamically switched and your Z-Wave stick now might have become /dev/ttyACM1 or visa versa.  

I'm not sure if it is already supported by HA, but it would be better to use the predictable device names instead of /dev/ttyACM*.

`/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE1948474-if00`

https://community.home-assistant.io/t/how-to-fix-predictable-usb-device-names/118427/4

joch, 2019-10-01:
I'm not actually using the z-wave stick, since I have moved everything over to Zigbee, so I'll just remove the stick as soon as I'm home and see if that makes a difference.

manup, 2019-10-01:
Yes in this case this should do the trick if the HA configuration also uses /dev/ttyACM0

joch, 2019-10-01:
Sadly getting the same errors even with the z-wave hub disconnected:

```
deconz       | 15:01:26:828 0x00178801032A32F2 error APSDE-DATA.confirm: 0xF0 on task
deconz       | 15:01:26:828 APS-DATA.confirm id: 6 status: transaction expired (8 seconds)
deconz       | 15:01:42:708 scan finished
deconz       | 15:01:49:226 0x84182600000D9CCD error APSDE-DATA.confirm: 0xD0 on task
deconz       | 15:02:00:116 0x7CB03EAA00A5B1C6 error APSDE-DATA.confirm: 0xD0 on task
deconz       | 15:02:11:064 0x000B57FFFED22DB6 error APSDE-DATA.confirm: 0xD0 on task
deconz       | 15:02:16:492 Current channel 25
```

Any way of debugging this further?

joch, 2019-10-08:
I'm back from vacation to look into this further. :) I tried moving the Conbee and running it on another machine in a slightly different location, but with the same results.

Some observations:
- all lights are unreachable (greyed out). They start out as reachable, but become unreachable after a while when deconz has been running.
- most sensors (not all) are reachable in the sense that they are not greyed out in the UI, but they don't update after initially being polled. These start out as greyed out but become reachable after a while (still don't work).
- the latest entry in the zbconf table had a networkKey with all `0`!? Updated to the correct one together with the version and firmware, but didn't make a difference.

This issue looks identical to #1876 as well. I tried searching for lights using touchlink (scan for devices), and then they do show up with the correct channel and network id. Clicking the blink button works on the light I tested on as well.

Looking at the log, some devices (assuming the sensors) are at least found in some form:
```
deconz       | 10:40:56:513 Node 0x00158D0001A99C42 is known by 1 neighbors, last seen 3 s
deconz       | 10:40:56:993 Node 0x00158D00015DA7C9 is known by 1 neighbors, last seen 6 s
deconz       | 10:40:57:473 Node 0x00158D0001C2AB69 is known by 1 neighbors, last seen 3 s
deconz       | 10:40:57:953 Node 0x00158D0001A94565 is known by 1 neighbors, last seen 2 s
deconz       | 10:40:58:440 Node 0x00158D00015DA7AF is known by 1 neighbors, last seen 6 s
```

While the lights are just completely gone:
```
deconz       | 10:41:10:913 Node 0x0017880104E536F8 is known by 0 neighbors, last seen 0 s
deconz       | 10:41:11:400 Node 0x0017880104E78320 is known by 0 neighbors, last seen 0 s
deconz       | 10:41:11:874 Node 0x000B57FFFE963688 is known by 0 neighbors, last seen 0 s
deconz       | 10:41:12:354 Node 0x000B57FFFE40DEA6 is known by 0 neighbors, last seen 0 s
deconz       | 10:41:12:833 Node 0x000B57FFFE928EB7 is known by 0 neighbors, last seen 0 s
```

Attached the full log since last night: [deconz.log.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3702013/deconz.log.zip)

@manup do you have any suggestions for a next step?

Not having proper lights at home is becoming an issue, so anything to help in debugging would be appreciated. :)

joch, 2019-10-08:
As mentioned in the previous post, the newest entry in zbconf had a networkKey of 0000..., so I fixed the entry. I assumed that it would be picked up on, but I now clicked the Load button in the Zigbee config page, and things suddenly started working again!

closed 2019-11-05 by joch

sorryusernameisalreadytaken, 2020-07-20:
@joch, I had a very similar mistake the day before yesterday. Unfortunately we had a lot of guests & I couldn't avoid to put my deconz back.

This was my first incident of this kind. So the complete loss of all Zigbee actuators. Nothing was working at all anymore.

My Raspberry-Pi 4 with the attached Raspbee module is standing next to my router which also distributes WLAN but I had no problems for 2 years.
My docker deconz  log was showing a lot of **error APSDE-DATA.confirm: 0xD0** errors.

I'm already worried about that. I absolutely need a stable system and I would have to know how to prevent this or why it happened.

I still have my old deconz config, the whole content of the folder:
/root/.local/share/dresden-elektronik/deCONZ

There's one there:
config.ini
session.default
zcldb.txt
and
zll.db

I suppose you had opened zll.db and removed the one entry with the many 0's?

joch, 2020-07-20:
> @joch, I had a very similar mistake the day before yesterday. Unfortunately we had a lot of guests & I couldn't avoid to put my deconz back.
> 
> This was my first incident of this kind. So the complete loss of all Zigbee actuators. Nothing was working at all anymore.
> 
> My Raspberry-Pi 4 with the attached Raspbee module is standing next to my router which also distributes WLAN but I had no problems for 2 years. I even repositioned my pi once with a longer LAN cable 6-7 meters away and did a reboot and choose another pi-usb-power-adapter. But also here no improvement.
> My docker deconz log was showing a lot of **error APSDE-DATA.confirm: 0xD0** errors.
> 
> I am worried that this could happen again.
> I'm already worried about that. I absolutely need a stable system and I would have to know how to prevent this or why it happened.
> 
> I still have my old deconz config, the whole content of the folder:
> /root/.local/share/dresden-elektronik/deCONZ
> 
> There's one there:
> **config.ini**
> **session.default**
> **zcldb.txt**
> and
> **zll.db**
> 
> I suppose you had opened zll.db and removed the one entry with the many 0's?
> 
> And where can I find the Zigbee configuration page introduced in 2.05.59, mentioned here:
> [#739 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/739#issuecomment-468574532)

Alt click on advanced to restore a previous setting: 

[image]