dresden-elektronik/deconz-rest-plugin #1919: Req for device support. Heiman gas sensor HS1CG-E

Asbjoern, 2019-10-01:
It looks like there is a new Heiman Combustible gas sensor HS1CG-E. The non-E is on the supported list.
This one does is not detected by the Phoscon app.
[image]
[image]
[image]

andrem99, 2019-10-09:
Could you please add support for this sensor model in Deconz.
Many thanks!

milkwithfish, 2019-10-10:
I've got the same problem. It would be awesome to add this sensor to supported list.

andrem99, 2019-10-11:
Hi @manup! Thanks for paying attention to the request. How long do you think it will take to integrate this version of the sensor to deconz?

michis0806, 2019-11-15:
How do you set the Heiman to pairing mode?

andrem99, 2019-11-16:
With a long press, nothing special.

michis0806, 2019-11-16:
Yes, I didn't manage to connect it to a deconz/phoscon gateway...

SCoe2k, 2020-01-05:
Same problem. Cannot connect hs1cg-e. Thank you for taking it up to devoce integration.
@manup : any indication about timing?

michis0806, 2020-01-17:
> With a long press. I tried all possible ways. Is there a trick, you mean?

Still no luck here. I think I tried any possible combinations of pressing the buttons of the Heiman, but phoscon does not recognize it :/

andrem99, 2020-01-18:
@manup Hi, Manuel, any update on this issue? quite a few people are suffering already:) would appreciate if you could give at least some update. thank you upfront!

SwoopX, 2020-02-02:
This post might hold the solution: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2403#issuecomment-581164140

flavio20002, 2020-03-06:
No way to pair this device....the post was not helpful...any hint?
Thanks

flavio20002, 2020-03-06:
Sorry, it works perfectly

`
    "43": {
        "config": {
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 1,
        "etag": "a35e061f17a70c456da16f7520e85040",
        "manufacturername": "Heiman",
        "modelid": "GASSensor-EM",
        "name": "Gas Sensor",
        "state": {
            "carbonmonoxide": false,
            "lastupdated": "none",
            "lowbattery": false,
            "tampered": false
        },
        "swversion": "2018.3.26",
        "type": "ZHACarbonMonoxide",
        "uniqueid": "xxxxxxx"
    },
`

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix