dresden-elektronik/deconz-rest-plugin #1922: Conbee II "not connected" to Phoscon app, stick is seen by the s

HikinenM, 2019-10-01:
Hi,

I have an issue with this Conbee II stick running on OMV and using deconz. The problem started when I rebooted deconz because deconz suddenly started using a lot of CPU cycles. Below is graph of the usage.
[https://drive.google.com/file/d/1K4d8fo-rGaAn17OfEsy80WAuvnBsoWrV/view?usp=sharing](url)

Anyway, after deconz had restarted I didn't have any devices working. I quickly saw that Phoscon app said firmware was not connected. Later on I haven't been able to connect to the phoscon app anymore. Below is a picture to explain the current situation:
[https://drive.google.com/file/d/11gEmKAnQcHFr1-FBGp88MXlmmeNluawP/view?usp=sharing](url)

I have had similar issue before where deconz didn't recognize the firmware and got it fixed by putting a usb extension cable between the server and the stick. Now I tried different usb slots etc. All of them were USB 2.0 btw. but that didn't help.

Here are some basic info about the stick. Root user and my own Mika user are in group dialout.
Mika@openmediavault:~$ lsusb | grep Dresden
Bus 006 Device 011: ID 1cf1:0030 Dresden Elektronik
Mika@openmediavault:~$ ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Oct  2 01:37 /dev/ttyACM0

Here is a partial log of deconz after the issue started. You can eventually see only "wait reconnect" messages:
`[marthoc/deconz] Starting deCONZ...
[marthoc/deconz] Current deCONZ version: 2.05.69
[marthoc/deconz] Web UI port: 80
[marthoc/deconz] Websockets port: 443
[marthoc/deconz] VNC port: 5900
Killing Xtigervnc process ID 23... which was already dead
Cleaning stale pidfile '/root/.vnc/openmediavault.local:0.pid'!
Cleaning stale x11 lock '/tmp/.X0-lock'!
Cleaning stale x11 lock '/tmp/.X11-unix/X0'!

New 'openmediavault.local:0 (root)' desktop at :0 on machine openmediavault.local

Starting applications specified in /etc/X11/Xvnc-session
Log file is /root/.vnc/openmediavault.local:0.log

Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd openmediavault.local:0 to connect to the VNC server.

libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
libpng warning: iCCP: known incorrect sRGB profile
01:52:39:802 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/
01:52:39:802 CTRL. 3.16.201:52:39:851 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
01:52:39:932 parent process /bin/sh
01:52:39:932 gw run mode: docker
01:52:39:932 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
01:52:39:933 DB sqlite version 3.16.2
01:52:39:934 DB PRAGMA page_count: 32
01:52:39:934 DB PRAGMA page_size: 4096
01:52:39:934 DB PRAGMA freelist_count: 0
01:52:39:934 DB file size 131072 bytes, free pages 0
01:52:39:934 DB PRAGMA user_version: 6
01:52:39:934 DB cleanup
01:52:39:935 DB create temporary views
01:52:39:950 don't close database yet, keep open for 900 seconds
01:52:39:951 started websocket server at port 443
01:52:39:955 found node plugin: libde_rest_plugin.so - REST API Plugin
01:52:39:957 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
01:52:39:965 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
01:52:40:883 wait reconnect 15 seconds
01:52:41:912 wait reconnect 14 seconds
01:52:42:909 wait reconnect 13 seconds
01:52:43:909 wait reconnect 12 seconds
01:52:44:910 wait reconnect 11 seconds
01:52:45:253 Announced to internet
01:52:45:909 wait reconnect 10 seconds
01:52:46:909 wait reconnect 9 seconds
01:52:47:909 wait reconnect 8 seconds
01:52:48:909 wait reconnect 7 seconds
01:52:49:909 wait reconnect 6 seconds
01:52:50:909 wait reconnect 5 seconds
01:52:51:909 wait reconnect 4 seconds
01:52:52:909 wait reconnect 3 seconds
01:52:53:909 wait reconnect 2 seconds
01:52:54:909 wait reconnect 1 seconds
01:52:54:972 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
01:52:55:909 wait reconnect 15 seconds
01:52:56:908 wait reconnect 14 seconds
01:52:57:909 wait reconnect 13 seconds
01:52:58:909 wait reconnect 12 seconds`

Trying to run the marthoc/deconz Firmware Flashing Script fails. Interesting here is that the **device serial number is not seen by the script**.
`Mika@openmediavault:~$ sudo docker run -it --rm --entrypoint "/firmware-update.sh" --privileged --cap-add=ALL -v /dev:/dev -v /lib/modules:/lib/modules -v /sys:/sys marthoc/deconz
-------------------------------------------------------------------

             marthoc/deconz Firmware Flashing Script

                       Version: 0.5

-------------------------------------------------------------------

Listing attached devices...

GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  |            | ConBee II

Enter the full device path, or press Enter now to exit.

Device Path : /dev/ttyACM0

-------------------------------------------------------------------

Firmware available for flashing:

deCONZ_ConBeeII_0x26490700.bin.GCF
deCONZ_ConBeeII_0x264a0700.bin.GCF
deCONZ_Rpi_0x26330500.bin.GCF

Enter the firmware file name from above, including extension,
or press Enter now to exit.

File Name : deCONZ_ConBeeII_0x26490700.bin.GCF

-------------------------------------------------------------------

Device: /dev/ttyACM0

Firmware File: deCONZ_ConBeeII_0x26490700.bin.GCF

Are the above device and firmware values correct?
Enter Y to proceed, any other entry to exit: Y

Flashing...

GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
deCONZ firmware version 264A0700
failed reset 10001

1607: Error: uart reset failed, check retry

Flashing Error! Please re-run this script...
`
dmesg messages after running the marthoc/deconz Firmware Flashing Script:
`[ 1407.936770] usb 6-2: USB disconnect, device number 5
[ 1408.524187] usb 6-2: new full-speed USB device number 6 using ohci-pci
[ 1408.731420] usb 6-2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 1408.731428] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1408.731432] usb 6-2: Product: ConBee II
[ 1408.731436] usb 6-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 1408.731440] usb 6-2: SerialNumber: DE2118846
[ 1408.734192] cdc_acm 6-2:1.0: ttyACM0: USB ACM device
[ 1411.812314] usb 6-2: USB disconnect, device number 6
[ 1412.440203] usb 6-2: new full-speed USB device number 7 using ohci-pci
[ 1412.649415] usb 6-2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 1412.649423] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1412.649427] usb 6-2: Product: ConBee II
[ 1412.649431] usb 6-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 1412.649434] usb 6-2: SerialNumber: DE2118846
[ 1412.655837] cdc_acm 6-2:1.0: ttyACM0: USB ACM device`

Could I have some help please? Is my stick busted or recoverable somehow?

HikinenM, 2019-10-02:
Update on the issue. I moved the Conbee II stick to my Windows 10 PC. After numerous removable device plugging and device plugging sounds, the stick was seen in Device manager as USB Serial Port (COM4) and could log into VNC instance and phoscon app saw it as new device to setup.

I then used command GCFFlasher -l and it saw the USB stick normally with serial number attached to it. I then ran command GCFFlasher -d COM4 -f ../firmware/deCONZ_ConBeeII_0x264a0700.bin.GCF and the tool successfully updated the firmware.

After all the success I happily moved the Conbee II stick back to OMV server. Nothing changed. After fiddling around I found out that docker configuration had "DECONZ_DEVICE=0", instead of "DECONZ_DEVICE=/dev/ttyAMC0". I'm not sure how this can be happened but I replaced the line to correct one.  Still the device is not discovered.

`08:52:15:498 wait reconnect 2 seconds
08:52:16:497 wait reconnect 1 seconds
08:52:16:521 COM: --dev: /dev/ttyAMC0 (RaspBee)
08:52:17:498 wait reconnect 15 seconds
08:52:17:534 COM: --dev: /dev/ttyAMC0 (RaspBee)
08:52:18:498 wait reconnect 14 seconds
08:52:19:497 wait reconnect 13 seconds
08:52:20:497 wait reconnect 12 seconds
08:52:21:498 wait reconnect 11 seconds
08:52:22:497 wait reconnect 10 seconds
08:52:23:498 wait reconnect 9 seconds
08:52:24:498 wait reconnect 8 seconds
08:52:25:497 wait reconnect 7 seconds
08:52:26:498 wait reconnect 6 seconds
08:52:27:498 wait reconnect 5 seconds
08:52:28:497 wait reconnect 4 seconds
08:52:29:498 wait reconnect 3 seconds
08:52:30:497 wait reconnect 2 seconds
08:52:31:498 wait reconnect 1 seconds
08:52:31:519 COM: --dev: /dev/ttyAMC0 (RaspBee)
08:52:32:497 wait reconnect 15 seconds
08:52:32:528 COM: --dev: /dev/ttyAMC0 (RaspBee)
08:52:33:497 wait reconnect 14 seconds
08:52:34:497 wait reconnect 13 seconds
08:52:35:497 wait reconnect 12 seconds
08:52:36:498 wait reconnect 11 seconds`

HikinenM, 2019-10-02:
Ok. The last one was a typo. /dev/ttyAMC0 -> /dev/ttyAMC0 and I get a bit more info. But still

`08:55:33:028 COM: /dev/ttyACM0 / serialno:
08:55:33:028 COM: --dev: /dev/ttyACM0 (ConBee II)
08:55:33:997 wait reconnect 15 seconds
08:55:34:029 COM: /dev/ttyACM0 / serialno:
08:55:34:029 COM: --dev: /dev/ttyACM0 (ConBee II)
08:55:34:997 wait reconnect 14 seconds
08:55:35:997 wait reconnect 13 seconds
08:55:36:998 wait reconnect 12 seconds
08:55:37:997 wait reconnect 11 seconds
08:55:38:997 wait reconnect 10 seconds
08:55:39:998 wait reconnect 9 seconds
08:55:40:901 scan finished
08:55:40:998 wait reconnect 8 seconds
08:55:41:997 wait reconnect 7 seconds
08:55:42:998 wait reconnect 6 seconds
08:55:43:998 wait reconnect 5 seconds
08:55:44:997 wait reconnect 4 seconds
08:55:45:997 wait reconnect 3 seconds
08:55:46:003 Websocket disconnected 192.168.1.12:14937 (state: 0)
08:55:46:997 wait reconnect 2 seconds
08:55:47:502 New websocket 192.168.1.12:14943 (state: 3)
08:55:47:875 Websocket disconnected 192.168.1.12:14943 (state: 0)
08:55:47:997 wait reconnect 1 seconds
08:55:48:027 COM: /dev/ttyACM0 / serialno:
08:55:48:028 COM: --dev: /dev/ttyACM0 (ConBee II)
08:55:48:997 wait reconnect 15 seconds
08:55:49:028 COM: /dev/ttyACM0 / serialno:
08:55:49:028 COM: --dev: /dev/ttyACM0 (ConBee II)`

`Listing attached devices...

GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  |            | ConBee II

Enter the full device path, or press Enter now to exit.

Device Path : /dev/ttyACM0

-------------------------------------------------------------------

Firmware available for flashing:

deCONZ_ConBeeII_0x26490700.bin.GCF
deCONZ_ConBeeII_0x264a0700.bin.GCF
deCONZ_Rpi_0x26330500.bin.GCF

Enter the firmware file name from above, including extension,
or press Enter now to exit.

File Name : deCONZ_ConBeeII_0x264a0700.bin.GCF

-------------------------------------------------------------------

Device: /dev/ttyACM0

Firmware File: deCONZ_ConBeeII_0x264a0700.bin.GCF

Are the above device and firmware values correct?
Enter Y to proceed, any other entry to exit: Y

Flashing...

GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
deCONZ firmware version 264A0700
deCONZ firmware version 264A0700

1607: Error: uart reset failed, check retry

Flashing Error! Please re-run this script...`

manup, 2019-10-02:
Can you please try following sequence:

* Stop the container with deCONZ
* Unplug ConBee II
* Wait 10 seconds
* Attach ConBee II again
* **wait 15 seconds**
* Start deCONZ container

Also highly recommend to use the USB extension cable you mentioned above.

HikinenM, 2019-10-02:
I tried that now but to changes.

[marthoc/deconz] Starting deCONZ...
[marthoc/deconz] Current deCONZ version: 2.05.69
[marthoc/deconz] Web UI port: 80
[marthoc/deconz] Websockets port: 443
[marthoc/deconz] VNC port: 5900
Killing Xtigervnc process ID 24... which was already dead
Cleaning stale pidfile '/root/.vnc/openmediavault.local:0.pid'!
Cleaning stale x11 lock '/tmp/.X0-lock'!
Cleaning stale x11 lock '/tmp/.X11-unix/X0'!

New 'openmediavault.local:0 (root)' desktop at :0 on machine openmediavault.local

Starting applications specified in /etc/X11/Xvnc-session
Log file is /root/.vnc/openmediavault.local:0.log

Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd openmediavault.local:0 to connect to the VNC server.

libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
libpng warning: iCCP: known incorrect sRGB profile
17:42:16:372 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/
17:42:16:372 CTRL. 3.16.217:42:16:445 COM: /dev/ttyACM0 / serialno:
17:42:16:445 COM: --dev: /dev/ttyACM0 (ConBee II)
17:42:16:446 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
17:42:16:534 parent process /bin/sh
17:42:16:534 gw run mode: docker
17:42:16:534 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
17:42:16:535 DB sqlite version 3.16.2
17:42:16:536 DB PRAGMA page_count: 32
17:42:16:536 DB PRAGMA page_size: 4096
17:42:16:536 DB PRAGMA freelist_count: 0
17:42:16:536 DB file size 131072 bytes, free pages 0
17:42:16:537 DB PRAGMA user_version: 6
17:42:16:537 DB cleanup
17:42:16:537 DB create temporary views
17:42:16:552 don't close database yet, keep open for 900 seconds
17:42:16:552 started websocket server at port 443
17:42:16:556 found node plugin: libde_rest_plugin.so - REST API Plugin
17:42:16:558 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
17:42:16:567 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
17:42:16:584 COM: /dev/ttyACM0 / serialno:
17:42:16:584 COM: --dev: /dev/ttyACM0 (ConBee II)
17:42:17:470 wait reconnect 15 seconds
17:42:17:501 COM: /dev/ttyACM0 / serialno:
17:42:17:501 COM: --dev: /dev/ttyACM0 (ConBee II)
17:42:18:499 wait reconnect 14 seconds
17:42:19:498 wait reconnect 13 seconds
17:42:20:497 wait reconnect 12 seconds
17:42:21:411 New websocket 192.168.1.12:1167 (state: 3)
17:42:21:498 wait reconnect 11 seconds
17:42:21:874 Announced to internet
17:42:22:497 wait reconnect 10 seconds
17:42:23:497 wait reconnect 9 seconds
17:42:24:498 wait reconnect 8 seconds
17:42:25:498 wait reconnect 7 seconds
17:42:26:498 wait reconnect 6 seconds
17:42:27:497 wait reconnect 5 seconds
17:42:28:498 wait reconnect 4 seconds
17:42:29:498 wait reconnect 3 seconds
17:42:30:498 wait reconnect 2 seconds
17:42:31:498 wait reconnect 1 seconds
17:42:31:528 COM: /dev/ttyACM0 / serialno:
17:42:31:529 COM: --dev: /dev/ttyACM0 (ConBee II)
17:42:31:560 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
17:42:32:498 wait reconnect 15 seconds
17:42:32:527 COM: /dev/ttyACM0 / serialno:
17:42:32:527 COM: --dev: /dev/ttyACM0 (ConBee II)
17:42:33:497 wait reconnect 14 seconds
17:42:34:497 wait reconnect 13 seconds
17:42:35:498 wait reconnect 12 seconds
17:42:36:498 wait reconnect 11 seconds
17:42:37:498 wait reconnect 10 seconds
17:42:38:497 wait reconnect 9 seconds
17:42:39:498 wait reconnect 8 seconds
17:42:40:497 wait reconnect 7 seconds
17:42:41:497 wait reconnect 6 seconds
17:42:42:065 scan skip host .4
QNetworkReplyImplPrivate::error: Internal problem, this method must only be called once.
17:42:42:498 wait reconnect 5 seconds
17:42:43:498 wait reconnect 4 seconds
17:42:44:497 wait reconnect 3 seconds
17:42:45:497 wait reconnect 2 seconds
17:42:46:497 wait reconnect 1 seconds
17:42:46:528 COM: /dev/ttyACM0 / serialno:
17:42:46:528 COM: --dev: /dev/ttyACM0 (ConBee II)
17:42:47:498 wait reconnect 15 seconds
17:42:47:528 COM: /dev/ttyACM0 / serialno:
17:42:47:528 COM: --dev: /dev/ttyACM0 (ConBee II)
17:42:48:497 wait reconnect 14 seconds
17:42:49:498 wait reconnect 13 seconds

HikinenM, 2019-10-02:
I'm really at a loss here. I tried to start fresh and uninstalled the docker container and image, then started the installation again by following this guide: https://phoscon.de/en/conbee2/install#docker. Still the same issue of deCONZ not seeing the device at all and unable to update the firmware (and not seeing serial of the device on the device listing). The device and it's is still seen properly by the Linux system.

Just in case I moved the stick and extension cable to my Windows 10 PC took a screenshot of the Conbee II working fine.
https://drive.google.com/file/d/1Dk5I0v9yZvu9IEnySHWUVqv0DGtR1uud/view?usp=sharing

So there is definitely something going on with the interaction between the Conbee II stick and the deCONZ docker container of mine or the linux server it's hosted on but I can't figure out what.

manup, 2019-10-02:
Hmm yes, I'n not sure yet what it is but lets check a few things.

The log indicates that deCONZ can't open the serial interface. Do you have any other USB devices connected to your system which may get accessed by some service?

I have no experience with OMV but maybe USB pass through needs to be enabled separately?

HikinenM, 2019-10-02:
> The log indicates that deCONZ can't open the serial interface. Do you have any other USB devices connected to your system which may get accessed by some service?
I have a External USB 3.0 hard drive connected.

> 
> I have no experience with OMV but maybe USB pass through needs to be enabled separately?
I don't think so unless something changed without my knowledge. My setup was working for almost a week before this started.

I'm not sure if this is related but as the Conbee II stick has been connected to my Windows 10 PC and I woke it up from sleep mode, after a while my PC started making device uplugging and plugging sounds. After some minutes of figuring out what was happening I fired up USBLogView and found out that the Conbee II was making those. After a while it stopped and is now working (on Windows).
https://drive.google.com/file/d/1hLYUhY9olnyUgVvtmRywSx8InrGAO7cT/view?usp=sharing

HikinenM, 2019-10-02:
I have no idea what again happened. I closed my OMV server, connected the Conbee II stick back to it and turned it on. For reason unknown deCONZ now recognised Conbee II stick. It looks to be working but for how long?

1:55:34:422 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/
21:55:34:448 CTRL. 3.16.221:55:34:543 COM: /dev/ttyACM0 / serialno:
21:55:34:543 COM: --dev: /dev/ttyACM0 (ConBee II)
21:55:34:543 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
21:55:34:761 parent process /bin/sh
21:55:34:762 gw run mode: docker
21:55:34:762 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
21:55:34:776 DB sqlite version 3.16.2
21:55:34:777 DB PRAGMA page_count: 30
21:55:34:777 DB PRAGMA page_size: 4096
21:55:34:777 DB PRAGMA freelist_count: 0
21:55:34:777 DB file size 122880 bytes, free pages 0
21:55:34:778 DB PRAGMA user_version: 6
21:55:34:778 DB cleanup
21:55:34:781 DB create temporary views
21:55:34:792 don't close database yet, keep open for 900 seconds
21:55:34:797 started websocket server at port 443
21:55:34:816 GW firmware update not supported on x86 linux headless
21:55:34:817 found node plugin: libde_rest_plugin.so - REST API Plugin
21:55:34:820 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
21:55:34:836 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
This plugin does not support propagateSizeHints()
21:55:34:857 COM: /dev/ttyACM0 / serialno:
21:55:34:857 COM: --dev: /dev/ttyACM0 (ConBee II)
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
PROTO: CRC error
PROTO: CRC error
21:55:34:905 DEV config changed event
21:55:34:978 Device firmware version 0x264A0700
21:55:34:989 unlocked max nodes: 200
21:55:35:080 Device protocol version: 0x010B
21:55:35:087 new node - ext: 0x00212effff04d5bb, nwk: 0x0000
21:55:35:153 Current channel 25
21:55:35:173 CTRL ANT_CTRL 0x03
21:55:35:206 Device protocol version: 0x010B
21:55:35:277 Current channel 25
21:55:35:298 CTRL ANT_CTRL 0x03
21:55:35:328 Device protocol version: 0x010B
21:55:35:389 Current channel 25
21:55:35:409 CTRL ANT_CTRL 0x03
21:55:35:440 Device protocol version: 0x010B
21:55:35:506 Current channel 25
21:55:35:526 CTRL ANT_CTRL 0x03
21:55:35:586 APS-DATA.indication from unknown node 0x7CB03EAA0A00AB4A
21:55:35:586 ZCL attribute report 0x7CB03EAA0A00AB4A for cluster 0x0006, ep 0x03
21:55:35:625 don't close database yet, keep open for 900 seconds
21:55:37:683 new node - ext: 0x000d6ffffeb8d34a, nwk: 0xE89B
21:55:38:163 new node - ext: 0x7cb03eaa0a00ab4a, nwk: 0x54E9
21:55:38:643 new node - ext: 0x000d6ffffebcdad8, nwk: 0x8915
21:55:39:590 discovery network reply error: Host dresden-light.appspot.com not found
21:55:40:351 don't close database yet, keep open for 900 seconds
21:55:48:518 saved node state in 0 ms
21:55:48:732 sync() in 213 ms
21:55:49:535 don't close database yet, keep open for 900 seconds
21:55:54:001 Set bridgeid to 00212EFFFF04D5BB
21:56:00:270 don't close database yet, keep open for 900 seconds
21:56:00:504 scan finished
21:56:34:524 Current channel 25
21:56:34:534 Device TTL 3533 s flags: 0x7
21:56:35:159 don't close database yet, keep open for 900 seconds
21:56:35:238 don't close database yet, keep open for 900 seconds
21:56:35:369 ZDP discovery done in 59575 ms21:56:35:606 don't close database yet, keep open for 900 seconds
21:56:35:666 don't close database yet, keep open for 900 seconds
21:56:35:825 don't close database yet, keep open for 900 seconds
21:56:35:929 remove discovery request duplicate
21:56:35:929 remove discovery request duplicate
21:56:35:929 remove discovery request duplicate
[... 369 lines cut ...]
21:58:10:885 ZCL attribute report 0x7CB03EAA0A00C70A for cluster 0x0300, ep 0x03
21:58:13:593 ZCL attribute report 0x7CB03EAA0A00AB4A for cluster 0x0008, ep 0x03
21:58:15:476 ZCL attribute report 0x7CB03EAA0A006FFE for cluster 0x0008, ep 0x03
21:58:15:598 ZCL attribute report 0x7CB03EAA0A00AB4A for cluster 0x0300, ep 0x03
21:58:34:522 Current channel 25
21:58:34:532 Device TTL 3413 s flags: 0x7
21:58:48:815 ZCL attribute report 0x7CB03EAA0A0033B8 for cluster 0x0006, ep 0x03
21:58:58:223 ZCL attribute report 0x00158D0003F41249 for cluster 0x0000, ep 0x01
21:58:58:223 0x00158D0003F41249 extract Xiaomi special attribute 0xFF01
21:58:58:223 	01 battery 3035 (0x0BDB)
21:58:58:223 	03 temperature 28 °C
21:58:58:224 	04 unknown 17320 (0x43A8)
21:58:58:224 	05 RSSI dB (?) 19 (0x0013)
21:58:58:224 	06 LQI (?) 8589934592 (0x0200000000)
21:58:58:224 	0a unknown 6594 (0x19C2)
21:58:58:224 	64 on/off 0
21:58:58:224 	0b lightlevel 0 (0x0000)
21:59:12:525 don't close database yet, keep open for 900 seconds
21:59:31:618 ZCL attribute report 0x7CB03EAA0A00AB4A for cluster 0x0300, ep 0x03
21:59:34:519 Current channel 25
21:59:34:529 Device TTL 3353 s flags: 0x7
21:59:49:659 ZCL attribute report 0x00158D00036BA1A8 for cluster 0x0400, ep 0x01
21:59:49:664 ZCL attribute report 0x00158D00036BA1A8 for cluster 0x0406, ep 0x01
22:00:00:493 ZCL attribute report 0x7CB03EAA0A006FFE for cluster 0x0300, ep 0x03
22:00:26:627 ZCL attribute report 0x7CB03EAA0A00AB4A for cluster 0x0006, ep 0x03

HikinenM, 2019-10-02:
Again the device suddenly stopped to been seen by deCONZ. I changed DECONZ_VNC_MODE=0 -> DECONZ_VNC_MODE=1 and after restart this is the log.

[marthoc/deconz] Starting deCONZ...
[marthoc/deconz] Current deCONZ version: 2.05.69
[marthoc/deconz] Web UI port: 80
[marthoc/deconz] Websockets port: 443
[marthoc/deconz] VNC port: 5900
/usr/bin/xauth:  file /root/.Xauthority does not exist

New 'openmediavault.local:0 (root)' desktop at :0 on machine openmediavault.local

Starting applications specified in /etc/X11/Xvnc-session
Log file is /root/.vnc/openmediavault.local:0.log

Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd openmediavault.local:0 to connect to the VNC server.

libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
libpng warning: iCCP: known incorrect sRGB profile
23:49:32:174 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/
23:49:32:174 CTRL. 3.16.223:49:32:226 COM: /dev/ttyACM0 / serialno:
23:49:32:226 COM: --dev: /dev/ttyACM0 (ConBee II)
23:49:32:226 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
23:49:32:308 parent process /bin/sh
23:49:32:308 gw run mode: docker
23:49:32:308 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
23:49:32:309 DB sqlite version 3.16.2
23:49:32:310 DB PRAGMA page_count: 30
23:49:32:310 DB PRAGMA page_size: 4096
23:49:32:310 DB PRAGMA freelist_count: 0
23:49:32:310 DB file size 122880 bytes, free pages 0
23:49:32:310 DB PRAGMA user_version: 6
23:49:32:310 DB cleanup
23:49:32:310 DB create temporary views
23:49:32:324 don't close database yet, keep open for 900 seconds
23:49:32:325 started websocket server at port 443
23:49:32:329 found node plugin: libde_rest_plugin.so - REST API Plugin
23:49:32:331 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
23:49:32:341 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
23:49:32:372 COM: /dev/ttyACM0 / serialno:
23:49:32:372 COM: --dev: /dev/ttyACM0 (ConBee II)
23:49:33:157 wait reconnect 15 seconds
23:49:33:190 COM: /dev/ttyACM0 / serialno:
23:49:33:190 COM: --dev: /dev/ttyACM0 (ConBee II)
23:49:34:107 wait reconnect 14 seconds
23:49:35:058 wait reconnect 13 seconds
23:49:36:007 wait reconnect 12 seconds
23:49:36:001 wait reconnect 11 seconds
23:49:37:808 discovery network reply error: Error transferring http://dresden-light.appspot.com/discover - server replied: Service Unavailable
23:49:37:002 wait reconnect 10 seconds
23:49:38:001 wait reconnect 9 seconds
23:49:39:001 wait reconnect 8 seconds
23:49:40:536 void MainWindow::devConnectClicked() choose com /dev/ttyACM0
23:49:40:001 wait reconnect 15 seconds
23:49:41:033 COM: /dev/ttyACM0 / serialno:
23:49:41:033 COM: --dev: /dev/ttyACM0 (ConBee II)
23:49:41:001 wait reconnect 14 seconds
23:49:42:020 New websocket 192.168.1.4:32960 (state: 3)
23:49:42:001 wait reconnect 13 seconds
23:49:43:003 wait reconnect 12 seconds
23:49:44:572 void MainWindow::devConnectClicked() choose com /dev/ttyACM0
23:49:44:957 void MainWindow::devConnectClicked() choose com /dev/ttyACM0
23:49:44:001 wait reconnect 15 seconds
23:49:45:033 COM: /dev/ttyACM0 / serialno:
23:49:45:033 COM: --dev: /dev/ttyACM0 (ConBee II)
23:49:45:166 void MainWindow::devConnectClicked() choose com /dev/ttyACM0
23:49:45:340 void MainWindow::devConnectClicked() choose com /dev/ttyACM0
23:49:45:535 void MainWindow::devConnectClicked() choose com /dev/ttyACM0
23:49:45:738 void MainWindow::devConnectClicked() choose com /dev/ttyACM0
23:49:45:001 wait reconnect 15 seconds
[... 35 lines cut ...]
23:50:06:001 wait reconnect 15 seconds
23:50:07:033 COM: /dev/ttyACM0 / serialno:
23:50:07:033 COM: --dev: /dev/ttyACM0 (ConBee II)
23:50:07:001 wait reconnect 14 seconds
23:50:08:001 wait reconnect 13 seconds
23:50:09:001 wait reconnect 12 seconds
23:50:10:105 scan finished
23:50:10:001 wait reconnect 11 seconds
23:50:11:001 wait reconnect 10 seconds
23:50:12:001 wait reconnect 9 seconds
23:50:13:001 wait reconnect 8 seconds
23:50:14:001 wait reconnect 7 seconds
23:50:15:001 wait reconnect 6 seconds
23:50:16:001 wait reconnect 5 seconds
23:50:17:001 wait reconnect 4 seconds
23:50:18:001 wait reconnect 3 seconds
23:50:19:001 wait reconnect 2 seconds
23:50:20:001 wait reconnect 1 seconds
23:50:21:032 COM: /dev/ttyACM0 / serialno:
23:50:21:032 COM: --dev: /dev/ttyACM0 (ConBee II)
23:50:21:001 wait reconnect 15 seconds
23:50:22:048 COM: /dev/ttyACM0 / serialno:
23:50:22:048 COM: --dev: /dev/ttyACM0 (ConBee II)
23:50:22:001 wait reconnect 14 seconds
23:50:23:002 wait reconnect 13 seconds

HikinenM, 2019-10-02:
Ok, this has been finally solved. When I first started using Home Assistant as docker image I installed the deCONZ addon from their store but didn't really understand what it does. I thought it was for integration for the deCONZ software. So basically that addon has been snatching up the Conbee II stick at times and sometimes not.

Sorry for bothering and thank you manup for the help.

closed 2019-10-02 by HikinenM

manup, 2019-10-04:
Cool glad it works now :)