dresden-elektronik/deconz-rest-plugin #1923: VNC devices can't be removed

hellcry37, 2019-10-02:
Hello,

The problem I have is not recent and is very annoying.

I've removed a sensor in Phoscon app and all seem removed and ok, then i login in VNC to find that the sensor is still there, connected and transmitting data as usual.

- tried to remove the sensor many times from VNC without any success
- restarted ConBee II
- removed again sensor

Sensor keeps appearing and i cant get rid of it, this happens with all the devices i've removed.

Images:

https://i.postimg.cc/FzD1kHd3/1.png

https://i.postimg.cc/Y9tpNww3/2.png

ebaauw, 2019-10-02:
Did you reset the sensor?

hellcry37, 2019-10-02:
this might be stupid but no, i did not reset the sensor(s) never did that, i did not think is necessary, i imagined this is necessary to be done only to join a zigbee network not to leave it.
it is mandatory to reset the sensor when you delete it in VNC?

ebaauw, 2019-10-02:
As long as the device is still a member of the network, it will be re-discovered and re-appear in the GUI.  I'm not even sure if removing a device from the GUI should make it leave the network, but in my experience it doesn't.  Factory-resetting the device does make it forget the network altogether.

Even then, the device might re-appear in the GUI (but now not communicating), because it's still in the neighbour tables of other devices.  I'm not sure when these tables are supposed to be cleared, but typically after a week or so, the device no longer re-appears.  Alternatively, you might try and power cycle the other devices.

Note that deleting a device from GUI typically won't delete the REST API resources for the device, but deleting the resource will also delete the device from the GUI.

hellcry37, 2019-10-02:
I'll try to fallow your guides, see what happens. But in my opinion delete should delete the device forever and right away, not after one week and it should not rejoin the network even if I did not reset it.

So basically delete it should:
- make the device leave the network
- remove the device and all related things from deconz
- all this without me resetting that device

This is what DELETE does in a normal scenario.

hellcry37, 2019-10-08:
Yah well after resetting old IKEA bulbs and removed them they still appear in deconz, bulbs are not even paired or even powered anymore and they still appear. I've started to have delays when I push buttons to turn on a bulb light doesn't turn on as it should.

I started to think this is not a good solution.

hellcry37, 2019-12-01:
Fallow the hue bridge example, when I delete a bulb, bridge leaves that bulb directly in unpaired state. 

example:
Removed Osram bulb from hue bridge and I could pair it directly in deconz with only a search. If i delete it in deconz, god have mercy, I have to manually reset it and even then is still present in deconz vnc as a node with no name.

hellcry37, 2019-12-01:
also, ebaauw say that factory reseting the deconz conbee will remove all devices, as a side note, this is not the case. After resetting conbee 2 a few times i still have all devices connected in vnc, they just there, connected, none of them have their respective name assigned, but they are there.

hellcry37, 2020-03-31:
this still needs solving

ebaauw, 2020-03-31:
> ebaauw say that factory reseting the deconz conbee will remove all devices

No, I said:
> As long as the device is still a member of the network, it will be re-discovered and re-appear in the GUI. I'm not even sure if removing a device from the GUI should make it leave the network, but in my experience it doesn't. Factory-resetting the device does make it forget the network altogether.

I'm referring to the device that you're trying to remove, not the RaspBee or ConBee.  I don't even know how to factory-reset a RaspBee or ConBee; `GCFFlasher -r` only does a regular reset, preserving the network configuration in non-volatile memory.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

hellcry37, 2020-06-05:
it's fine I moved away from deconz, to many bugs that are not solved for way to long time

Mimiix, 2020-06-05:
Thats what i'm here for to bring some structure :)