dresden-elektronik/deconz-rest-plugin #1924: How do i pair a Philips Hue Fair lamp ?

donnib, 2019-10-02:
It comes with a philips remote but i want to connect it to deconz and i can't seems to figure out how to pair it. Any ideas ?

bastshoes, 2019-10-02:
I guess as any other hue bulb. Hold the remote within 4 inches of the bulb. Press On and Off buttons simultaneously and hold for 10sec. It should blink and you can start light discovery using deconz.

Philje123, 2019-10-06:
Just follow the instructions in Phoscon.

It shows you what to do (resetting the remmote etc).

Philje123, 2019-10-06:
Oh first you might need to reset the light as they come paired with the remote.

Turn the lamp off (at the switch) and back on again and hold the dimmer close to it. Hold the on and off buttons until the light starts to blink and wait for it to stop blinking.

Then pair the lamp and then the remote (following the instructions in phoscon for resetting the remote with a paperclip etc).

closed 2020-02-10 by stale[bot]