dresden-elektronik/deconz-rest-plugin #1925: Mannually(physically) changin a device value takes time to update the api with the good value

MattL0, 2019-10-03:
see this thread . https://github.com/andreypopov/node-red-contrib-deconz/issues/35

MattL0, 2019-10-05:
Quote from my other thread :

In this case ‘’manually’’ meant with the hue dimmer switch associated to 5 lights via deconz. (So I can control lights event if my server is down) .

If i press the button to increase the light ... it will take ~10 sec to seet the value in node red .

Same thing with my osram switch if i press the on/off button mannually on the switch.

played with postman a little ... seems like a deconz limitation

@manup

MattL0, 2019-10-12:
up

MattL0, 2019-10-29:
@manup?

MattL0, 2020-02-27:
@manup is that a normal zigbee behavior? Maybe do not understand something

ebaauw, 2020-02-27:
How did you “associate” the dimmer to the lights?

MattL0, 2020-02-27:
With Phoscon web app. But I saw the same update latency with the osram wall plug when I physically press the button on the physical wall plug.

ebaauw, 2020-02-27:
The problem is that you change the plug state, unknown to deCONZ.  It needs to get an updated state from the plug, either thru attribute reporting, or by polling the plug.  Polling causes the delay, which gets worse with larger networks.  I think the OSRAM plug does support attribute reporting, you could try and re-pair it to see if the REST API plugin will set this up, or set it up manually.

Hue lights don’t support attribute reporting, but the REST API plugin should eavesdrop on the group commands the dimmer sends to the Hue lights, updating the cached light state.  However if you paired the dimmer to the lights directly, it will send unicast commands to the lights, which the coordinator might not see.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix