dresden-elektronik/deconz-rest-plugin #1926: Migration between systems and devices (docker container)

lukicsl, 2019-10-03:
I have currently the productive system running inside a docker container (2.05.69) on a raspberry pi with raspbee.
I would like to migrate to another system (Rock64 - new production) also inside a docker container with conbee II.
Neither copying the config directory between containers nor creating and importing backup succeeded in having a valid network on the new system.
I played with all different options, though I do not want to mess up my current production system.
Is there a way to do this?

jo-me, 2019-10-05:
I did it recently by means of backup/restore but afaik, the network config is on the stick - the backup contains only the custom names for the nodes, scenes and so on.
Make sure to have the same deconz version and double-check the mapped device name.

lukicsl, 2019-10-06:
Bad, really bad situation!
I moved away from Tradfri Gateway because there was no backup. I had a situation that the gateway dropped dead. I had to do a lot of manual work to bring the network up again.
I hoped to have a hot/called backup system that could just load the config and take over.
Even thinking about docker cluster config. So that seems to be impossible.
To understand it right: The network from the stick cannot be back upped?
Nor is there a solution to move the whole network from one system to another?

lukicsl, 2019-10-06:
I found this wiki:
[backup/restore](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Backup-&-Restore)
Will close the issue

closed 2019-10-06 by lukicsl