dresden-elektronik/deconz-rest-plugin #1927: Connecting to a Siterwell GS361

larskohlenberg, 2019-10-03:
Hi, 

I got a Siterwell GS361 radiator themostat and see it in the deCONZ:
[image]

[image]

But I do not get any further cluster infos and also do not see this device in the phoscon webinterface.

Any ideas?

Thanks and best regards
Lars

Shurov, 2019-10-10:
Sorry for off-topic, but where could you find it available for purchase?

BTopbas, 2020-01-15:
I am interesting this TRV. @larskohlenberg did you find solution?

BTopbas, 2020-01-15:
@larskohlenberg and where did you purchase it?

sylvaing, 2020-05-11:
Hi, No news about gs361 and Deconz support ?

closed 2020-07-04 by stale[bot]

Wonkavator, 2020-09-19:
Winter is coming and I ordered the same device.
Would appriciate, if there could be a support by deconz. Is there some way how I could support development?

There is one article that was sniffing the communication, might that help:
https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-2-5492707e882d

kylegordon, 2020-09-28:
@BTopbas @Shurov I've found it at https://www.aliexpress.com/item/1005001349186776.html

Also wondering if there's support or not, or upcoming

Mimiix, 2020-09-29:
@Smanar Is this something for you?