dresden-elektronik/deconz-rest-plugin #1928: No automatic binding and attribute reporting

SwoopX, 2019-10-03:
Since I got a boatload of questions, I hope some light can be cast on this, especially by @ebaauw and @manup. This all originated from the Develco smoke sensor integration, which seemed to work despite the cumbersome device pairing (that might be somehow connected but coming to that later).

I thought all the time that there's still somethng not right since I never got any temperature or battery readings. Out of curiosity, I integrated the device in zigbee2mqtt and it went all got. Pairing, discovery, reporting as it should be while sniffing the traffic. Then I played around with the deconz gui and noticed the (from my perspective) totally forgotten binding dropbox. As it couldn't have hurt, I did that for the two clusters and promptly, reporting was there. What I also noticed before was, that reporting for IAS zone, however, was working, but coming from the light node, not a sensor node.

That brings me to a couple of questions:

- Can this be due to the fact, that for the device's siren, a light node had to be created while the other stuff is exposed as sensors?
- As I read the code, binding and reporting should be automatically handled due to to the code amendments for the device, right? However, I cannot see and binding or reporting requests.
- Could that also be related to the pairing and discovery process? I mean, I can see node descriptor requests and active endpoint requests which aren't picked up. Consequently, after successfull pairing, I don't see any endpoints at all and also not, e.g., the correct manufacturer code in the node info pane. The endpoints only appear after manually issuing a simple descriptor read and that only, if I manually set the duration in "permit join" to 0.

I'm just wondering and can't wrap my head around it. Any input and discussion is most welcome. Thanks!

ebaauw, 2019-10-04:
> I hope some light can be cast on this

I'm afraid "some" light is all I can do...

>  but coming from the light node, not a sensor node.

In ZigBee, nor in the deCONZ GUI, there's no such thing as a light node vs a sensor node.  There's the ZigBee coordinator (blue node), ZigBee routers (yellow nodes), and ZigBee end devices (grey nodes).  Routers (incl. the coordinator) have their radio on all the time; end devices power down their radio most of the time.  Routers form the mesh network; end devices connect to a single router (their parent).  A ZigBee command (message) for an end-device is actually sent to its parent.  The parent caches the command, until the end device wakes up and checks in to (polls) its parent.  Unfortunately, the parent only caches the command for like 7 seconds.  This is no problem for "light sleepers", that poll their parent every 5 seconds or more frequently.  It does cause an issue for "deep sleeper", that poll the parent at longer intervals.  Neither deCONZ nor the REST API plugin knows whether an end device is a light sleeper or a deep sleeper, and assumes the worse.  They only send commands when they know the end device to be awake, i.e. when they just received a command from the end device.

> Can this be due to the fact, that for the device's siren, a light node had to be created while the other stuff is exposed as sensors?

A ZigBee device contains multiple end points (logical devices); each end point contains multiple clusters (services).  Clusters come in pairs: server (blue numbers) vs client (grey number).  Typically the server provides the service, keeping state, and the client consumes or controls the service.  Roughly speaking, the REST API plugin creates a REST resource per cluster, but there are exceptions.  If you check the `uniqueid` of the resource, you see the device (mac address), endpoint and cluster.  For `/lights` resources, the _On/Off_, _Level Control_, and _Color Control_ clusters are combined in a single resources.  For switches/controllers that send commands to control lights directly, the client clusters are combined in a single resource.

In the good old days when deCONZ and the REST API plugin were created, the world was simple.  All actuators were mains-powered lights and ZigBee routers, and all sensors were battery-powered and ZigBee end devices.  This is reflected in the REST API setup of `/lights` resources vs `/sensors` resources, with different code handling the two different cases.  Nowadays we see lights (or in-wall switches) that are end devices, even though mains powered; we see mains-powered sensors and we see other actuators (curtains controllers, thermostats, locks, ...).

To address these changes, the code contains all sorts of exceptions for known devices.  E.g. to create a `/lights` resource for an end device.  In the area of binding there's a lot of these exceptions.  If you check `bindings.cpp`, you'll see different methods for checking and creating the bindings for `/lights` resources vs `/sensors` resources, and even for setting up bindings for attribute reporting (from a _server_ cluster on the device to the coordinator) vs for client clusters (from a _client_ cluster on the device to a group), each with their own hard-coded whitelists.

> As I read the code, binding and reporting should be automatically handled due to to the code amendments for the device, right? However, I cannot see and binding or reporting requests.

In my experience, adding REST API support for the devices I own, there's always one more place to whitelist.  It doesn't help that some whitelists are on _Model Identifier_, others on _Manufacturer Code_, other on mac address prefix, others on ZigBee cluster or even on ZigBee attribute.  Especially the code to create the bindings for end devices seems cursed, as it needs a command from the end device before being sent.

> I can see node descriptor requests and active endpoint requests which aren't picked up.

Aren't picked up as in: don't receive a response?  Could be because the request doesn't reach the end device, or because the response doesn't reach the coordinator.  You really need to sniff the MAC layer to check whether the requests are sent to the parent router.  Best place your sniffer dongle next to the end device, to pick up MAC traffic between end device and parent.  Also note that different manufacturers interpret the standard differently, and some combinations of end device and parent simply won't work.  Try forcing the RaspBee or ConBee as parent, by powering down the other routers, and keeping the end device close the the Rasp/ConBee while pairing.

> The endpoints only appear after manually issuing a simple descriptor read 

End devices tend to be wake for some time after joining a network, so the coordinator/gateway can configure them.  Manually reading the descriptors (while triggering the device, forcing it to wake up) helps in populating the endpoints and clusters, but REST API plugin requests might still be dropped, because the device is back to sleep, or deCONZ might not send these, because it thinks the device is asleep.  You could try to reset and repair the device (_without_ deleting the node); deCONZ tends to use the previously read descriptors the second time around.

It could also be that the REST API plugin drops the request before even sending it, because the device (or cluster) fails the whitelist.  In my experience, fixing this is a trial and error process, adding debug [... cut ...]
> and that only, if I manually set the duration in "permit join" to 0.

Note that the REST API plugin doesn't know when you open the network from the GUI.  It won't create REST resources, unless, the network is opened through the API (easiest from the Phoscon or old web app).

> What I also noticed before was, that reporting for IAS zone, however, was working

I think _IAS Zone_ devices don't use bindings; setting the _IAS CIE Address_ seems to be enough to get reports (or rather _IAS Zone Status_ commands).

SwoopX, 2019-10-05:
@ebaauw, I really apprechiate your extensive response on this! Although I knew a couple of things you mentioned before, it gave me some new ideas or directions. However, other people will also have valuable information here.

Allow me to respond to some points:

> The parent caches the command, until the end device wakes up and checks in to (polls) its parent. Unfortunately, the parent only caches the command for like 7 seconds. This is no problem for "light [...]

According to vendor technical manual, it seems the device is a "light" sleeper. That's what the values from the poll cluster would also support (7.5 secs or below for long poll).

> In my experience, adding REST API support for the devices I own, there's always one more place to whitelist. It doesn't help that some whitelists are on Model Identifier, others on Manufacturer [...]

Jep, still hunting for the last remnant in terms of whitelisting. Not exactly the fun part.

   >> I can see node descriptor requests and active endpoint requests which aren't picked up.

> Aren't picked up as in: don't receive a response? Could be because the request doesn't reach the end device, or because the response doesn't reach the coordinator. You really need to sniff the MAC [...]

My bad, I was now precise enough here. the node descriptor and active endpoint responses do not seem to be picked up by the code. I believe that especially the unavailability of the node descriptor information is responsible is responsible for the binding to fail.

> It could also be that the REST API plugin drops the request before even sending it, because the device (or cluster) fails the whitelist. In my experience, fixing this is a trial and error process, [...]

Totally agree. A new API version could significantly reduce overall complexity, removing the device specific sections, be more clear and efficiently and so. Also, I did some trial and error and noticed that the device is kicked out during binding due to that particular code:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/6a1356751f5c939743a8ea8a2169a57da4a73f44/bindings.cpp#L1672-L1675
Removing it for testing however still resulted in failure afterall with
`failed to send bind/unbind request to 0x0015BC0031004204 cluster 0x0001. drop`
Still investigating it with more custom debug output...

>Note that the REST API plugin doesn't know when you open the network from the GUI. It won't create REST resources, unless, the network is opened through the API (easiest from the Phoscon or old web app).

Thought so. But what should I say, turned out to be the best and fastest way to get that thing working...

closed 2020-01-02 by SwoopX