dresden-elektronik/deconz-rest-plugin #1929: Raspbee never online *stuck in joining*

SSkull995, 2019-10-04:
Hello! 
I just installed my Raspbee, I received it yesterday, on my rasp. The thing is, it doesn’t connect to my network, the red led is always blinking! 
I tried everything already done, I upgrade it to the 300500 version, because I read it helped another person, but not me apparently. 
I just check the network settings, weird thing is that TC address is all 0 (0x0000000) as the network key and tc link key. 
Device type is coordinator, as should be, but I cannot save any thing I change. 
Do you have any idea on how to solve this problem? Thanks a lot! Bye!

manup, 2019-10-04:
>I tried everything already done, I upgrade it to the 300500 version, because I read it helped another person, but not me apparently.

There is no such version, the latest RaspBee firmware version is 0x26330500. See also https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Can you click the "Leave" button in deCONZ? The LED should then stop to blink and we can reconfigure the settings.

SSkull995, 2019-10-04:
Hi! I can assure you there is the 0x26300500, I downloaded it. No all the buttons are grayed out!

SSkull995, 2019-10-04:
As a result, I can’t change anything

manup, 2019-10-04:
0x26300500 (I totally forgot it existed) is older than 0x26330500, please upgrade to the latest version first.

https://www.dresden-elektronik.de/rpi/deconz-firmware/deCONZ_Rpi_0x26330500.bin.GCF

We can then check the configuration and try to repair it manually, for that can you please provide the mac address printed on the RaspBee radio module, it should start with 00-21-2E...

SSkull995, 2019-10-04:
Wait, I passed from the 330500 to the 300500 because even with the 330500 I had the same exact problem. Buttons grayed and nothing can be changed

SSkull995, 2019-10-04:
How do I enable the debug mode? Thanks!!

manup, 2019-10-04:
You really need to use 0x26330500 it has the latest bug fixes. The issue with the buttons can be resolved by reconfiguring afterwards. It's a bit tricky but can be done, I assume currently the firmware is running in circles because no valid configuration is present.

Debugging can be enabled by starting deCONZ from a terminal (but it won't help here much):

    deCONZ --http-port=8080 --dbg-info=1 --dbg-error=1

SSkull995, 2019-10-04:
ok so I installed the latests firmware, I am checking now with the gui

manup, 2019-10-10:
Glad there is progress. The network key can be selected freely and should use some random numbers.
To create new random key you can use the "Reset gateway" function in  the Phoscon App.

https://phoscon.de/en/app/doc#settings-gateway-general-en

Be aware that this will reset all settings including already paired devices.

closed 2019-10-10 by SSkull995