dresden-elektronik/deconz-rest-plugin #1930: Phoscon application on Debian

demoulins, 2019-10-04:
Hello, 
I try to install Phoscom application on my Debian 9 VM (Virtualbox) with a Conbee 2 stick. It seems that the installation gone well but the Deconz cannot be run.
I get this issue when I ask for Deconz status.👍 
$ sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-10-03 19:28:17 CEST; 43s ago
 Main PID: 1425 (deCONZ)
    Tasks: 6 (limit: 4915)
   CGroup: /system.slice/deconz.service
           └─1425 /usr/bin/deCONZ -platform minimal --http-port=80

oct. 03 19:28:17 debian systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
oct. 03 19:28:17 debian deCONZ[1425]: libpng warning: iCCP: known incorrect sRGB profile
oct. 03 19:28:18 debian deCONZ[1425]: This plugin does not support propagateSizeHints()

Is it possible to use Deconz application on Debian ?
I am using the Conbee2 with Jeedom with Deconz plugin.
Thanks

manup, 2019-10-04:
Yes deCONZ also runs on Debian. Did you stop the `deconz-gui` service?

https://github.com/dresden-elektronik/deconz-rest-plugin#headless-support-for-linux

>oct. 03 19:28:17 debian systemd[1]: Started deCONZ: ZigBee gateway -- REST API.

Looks like deCONZ is already running.

What is the output of:

    ps ax | grep deCONZ

demoulins, 2019-10-04:
Thanks for you reply.
Here the output : 👍 
jacques@debian:~$ ps ax | grep deCONZ
 1423 ?        Ss     0:22 /bin/bash /usr/bin/deCONZ-update2.sh
 5995 pts/0    S+     0:00 grep deCONZ
30440 ?        Ssl    4:54 /usr/bin/deCONZ -platform minimal --http-port=80
32534 ?        S      0:00 sudo deCONZ --auto-connect=1 --dbg-error=1 --http-port=8484 --ws-port=20877 --upnp=0 -platform minimal --dbg-info=1 --dbg-aps=0 --dbg-zcl=0 --dbg-zdp=0 --dbg-http=0
32538 ?        Sl    15:32 deCONZ --auto-connect=1 --dbg-error=1 --http-port=8484 --ws-port=20877 --upnp=0 -platform minimal --dbg-info=1 --dbg-aps=0 --dbg-zcl=0 --dbg-zdp=0 --dbg-http=0

demoulins, 2019-10-04:
Deconz-gui service seems to be active.

jacques@debian:~$ systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-10-03 20:30:12 CEST; 19h ago
 Main PID: 30440 (deCONZ)
    Tasks: 6 (limit: 4915)
   CGroup: /system.slice/deconz.service
           └─30440 /usr/bin/deCONZ -platform minimal --http-port=80
jacques@debian:~$ systemctl status deconz-gui
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

manup, 2019-10-04:
There seems be multiple instances running.

The following commands should bring only the headless version to live.
```
sudo killall deCONZ
sudo systemctl disable deconz-gui
sudo systemctl stop deconz-gui
sudo systemctl restart deconz
```

demoulins, 2019-10-04:
Initially Deconz have been installed with the installation of "Jeedom Deconz plugin". And the Deconz Web application is actually operationnal.
Yesteday I just try to install the Deconz application on Debian.  
Did I create a new instance for Deconz ? 
I just want to access the actual instance created by Jeedom but by the Deconz application. 
It's possible that I did not understand something !!

closed 2020-02-08 by stale[bot]