dresden-elektronik/deconz-rest-plugin #1931: Cannot edit switches

weltmeyer, 2019-10-04:
Version: 2.05.67 / 26.8.2019

After Adding Tadfree switch and adding an action within a group of lights, i cant edit the switches anymore. i get a javascript error:

`Uncaught (in promise) TypeError: Cannot read property 'value' of undefined
    at group-switches.js?39f81c:3
    at Array.forEach (<anonymous>)
    at n.render (group-switches.js?39f81c:3)
    at n.initialize (group-switches.js?39f81c:3)
    at n.e.View (backbone-min.js?cedae4:1)
    at new n (backbone-min.js?cedae4:1)
    at n.addResourcelink (group-switches.js?39f81c:3)
    at n.<anonymous> (group-switches.js?39f81c:3)
    at underscore-min.js?2a5156:5
    at Function.m.each.m.forEach (underscore-min.js?2a5156:5)
(anonymous) @ group-switches.js?39f81c:3
render @ group-switches.js?39f81c:3
initialize @ group-switches.js?39f81c:3
e.View @ backbone-min.js?cedae4:1
n @ backbone-min.js?cedae4:1
addResourcelink @ group-switches.js?39f81c:3
(anonymous) @ group-switches.js?39f81c:3
(anonymous) @ underscore-min.js?2a5156:5
m.each.m.forEach @ underscore-min.js?2a5156:5
(anonymous) @ backbone-min.js?cedae4:1
initialize @ group-switches.js?39f81c:3
e.View @ backbone-min.js?cedae4:1
n @ backbone-min.js?cedae4:1
setGui @ group-switches.js?39f81c:3
(anonymous) @ group-switches.js?39f81c:3
Promise.then (async)
(anonymous) @ group-switches.js?39f81c:3
j @ jquery-min.js?0baf29:2
k @ jquery-min.js?0baf29:2
setTimeout (async)
(anonymous) @ jquery-min.js?0baf29:2
i @ jquery-min.js?0baf29:2
add @ jquery-min.js?0baf29:2
(anonymous) @ jquery-min.js?0baf29:2
Deferred @ jquery-min.js?0baf29:2
then @ jquery-min.js?0baf29:2
r.fn.ready @ jquery-min.js?0baf29:2
(anonymous) @ group-switches.js?39f81c:3
`

manup, 2019-10-04:
I think this should be fixed in the latest version 2.05.69.

weltmeyer, 2019-10-04:
> I think this should be fixed in the latest version 2.05.69.

Ohh.. it says i am on the latest version.. how do i upgrade from the webinterface?

weltmeyer, 2019-10-04:
> > I think this should be fixed in the latest version 2.05.69.
> 
> Ohh.. it says i am on the latest version.. how do i upgrade from the webinterface?

Nevermind, i updateds manually via console :)

weltmeyer, 2019-10-04:
The problemn also exists with 2.05.69, but the error looks different:
`TypeError: Cannot read property 'value' of undefined
    at group-switches.js?9ec317:3
    at Array.forEach (<anonymous>)
    at n.render (group-switches.js?9ec317:3)
    at n.initialize (group-switches.js?9ec317:3)
    at n.e.View (backbone-min.js?cedae4:1)
    at new n (backbone-min.js?cedae4:1)
    at n.addResourcelink (group-switches.js?9ec317:3)
    at n.<anonymous> (group-switches.js?9ec317:3)
    at underscore-min.js?2a5156:5
    at Function.m.each.m.forEach (underscore-min.js?2a5156:5)`

This makes it impossible to edit the keys within the group.
[image]

manup, 2019-10-04:
I've updated the online version of the Phoscon App, which hopefully fixes the issue.
Can you please try via http://dresden-elektronik.de/pwa

But I'm not sure why the error occurred in first place, looks like some rule isn't complete.

weltmeyer, 2019-10-05:
yepp error is skipped now.
could this info help you?:
"skip ResourcelinkView render(), btn not found, rl: 20"

manup, 2019-10-07:
Interesting, really looks like the rule isn't complete. Can you please create a backup file in the Phoscon App and send it to mpi@dresden-elektronik.de so I can check what is missing.

weltmeyer, 2019-10-08:
sure. mail is sent.

closed 2020-02-12 by stale[bot]