dresden-elektronik/deconz-rest-plugin #1932: how to pair with a specific router

pergolafabio, 2019-10-04:
hi, on every floor i have an ikea smart plug, to act as router

on 3th floor, have 4 xiaomi pir sensors, 3 of them are connected to the ikea plug, according to VNC
but the 4th, seems i am unable to connect to that router

i cant get it to connect to that ikea smart plug? any tips?
i already restarted conbee, deleted the node from phoscon, restarted ikea

then after restart, i search for devices, i press and fold 4 secs the xiaomi
but it never finds the pir

if i go downstairs, nearby conbee, then it works...

i only have like  8 end devices and 3 ikea routers, so bo big mesh , no limits ot whatsoever

thnx in advance

Smanar, 2019-10-04:
Hello, and if you make tries with the sensor close to the plug ? Not a good solution because the pairing need to be done in definitive place, but can try for test.

pergolafabio, 2019-10-04:
yes, i did it on original place and close to the router like 2 cm

pergolafabio, 2019-10-12:
Pff, still not able to connect to nearby router... Really annoying... I need to walk close to the conbee, to pair my xiaomi... 1 of 4 xiaomi I was able to pair with the Ikea, the others not... While Ikea is very nearby, and conbee is like 20 meters away...

pergolafabio, 2019-10-19:
Anyone can give me some tips how I can repair a xiaomi pir with a Ikea plug?  Seems when I reset the xiaomi by keep and holding the button and doing the add sensor wizard , it always want to search for the conbee, and not the plug that's like 10 cm nearby... 

If I can succeed I can finally ditch another raspberry

closed 2019-11-14 by pergolafabio