dresden-elektronik/deconz-rest-plugin #1933: Eurotronic Spirit z wave plus TRV's

DrDale, 2019-10-04:
Hi everyone , I have installed 9 of these Eurotronic Spirit TRV on my radiator - connected to my home assistant.  All looks good from the automation stand point, with the TRV's changing their target temperature value , but seem not to be controlling the room temperature to align to the target temperature.  I keeps seeing a massive over heat in the room as per attached graph. Any pointers on what I need to do to correct this please ?

The red lines on the graph is the room temperate - seen by the TRV and the green line is the target temperature for the TRV

[image]

[image]

ebaauw, 2019-10-04:
I don't know the z-wave version, but the ZigBee version seems to use a PID algorithm to regulate the room temperature.  It might take some training for the thermostat to learn how quickly the room heats or cools.  The ZigBee version reports the actual valve position, allowing you to monitor its working.  It also allows for an offset, in case the on-device thermometer is influenced too much by the radiator heat.

I'm not sure what heating system you have, but looking at the graphs it appears to start proving (more) heat in the morning around 6.00 and in the evening around 18.00.  This would confuse the PID algorithm in the TRV, since the relationship between valve position and room temperature changes (the TRV doesn't know the water temperature).  You want to control both the central heating setpoint and the TRV setpoint from your automation.  Or synchronise your home automation to your central heating thermostat schedule.

jo-me, 2019-10-05:
I have the Zwave versions of this thermostat and I am seeing the exact same issue. 
They do not close the valve enough to stop overheating the rooms. I saw that it kept a valve on 80% for more than 3 hours with 3°C above setpoint (all values delivered by the thermostat). 
In my opinion this is a bug in the firmware. 

Learning or optimizing for a certain room only works if you have a constant heat supply. We have a "leading room" that has a central thermostat. Only if that room needs needs heat will the boiler turn on. If, by chance, other rooms also need heat, they will get it then, but otherwise not. 

I'm currently playing around with manually controlling the valve to see whether they are forgetting their calibration, but I have no conclusive results yet. I'd rather disable the "optimization" part of the firmware because manually controlling the valves defeats the point of having a thermostat..

If you know a better workaround please let me know ;)

Oh I'm curious: do the temperature reports on the zigbee version work properly? On the zwave version I have to configure to report changes every 0.1°C to even get temperature values. Higher config values (e.g. 0.5°C) will typically not produce temperature reports because the temperature does not change e.g. 0.5° between two measurements and the info is therefore not sent - even if the difference to the last sent temp value is >= 0.5°C.

ebaauw, 2019-10-05:
> In my opinion this is a bug in the firmware.

The ZigBee version supports over-the-air firmware upgrades, but I don't think they publish any firmware.  My email to Eurotronic support went unanswered...

> Oh I'm curious: do the temperature reports on the zigbee version work properly?

ZigBee attribute reporting does both change reporting and periodic reporting.  I've configured it for 0.1°C or 0.2°C change and every 10 minutes.  These settings influence reporting of the temperature to the coordinator, not necessarily measuring the temperature for the TRV's PID algorithm.

> We have a "leading room" that has a central thermostat. Only if that room needs needs heat will the boiler turn on. If, by chance, other rooms also need heat, they will get it then, but otherwise not.

I have a similar setup, with a programmable room thermostat in my living room that controls my modulating central heating boiler.  The boiler actually has a PID algorithm of its own.  The thermostat controls the boiler's water temperature setpoint (based on thermostat's room temperature and setpoint) and the boiler controls the flame intensity and pump (based on water temperature setpoint and the return water temperature).

> If you know a better workaround please let me know ;)

By setting the thermostat a bit (0.5 to 1°C) higher than the living room TRVs, the boiler still supplies heat to the other rooms when the living room is warm enough.  When the other rooms are heated as well, the return water temperature remains close to the outgoing water temperature and the boiler reduces or turns off the flame.  In the morning I set the thermostat and the bathroom TRV half an hour before the living room TRVs, so my bathroom is comfortably warm.  Not sure if it's better, but it seems to do the trick.

I installed an OpenTherm Gateway between my room thermostat and boiler, so I can monitor and control from HomeKit both the TRVs (through homebridge-hue) and the thermostat and boiler (through homebridge-otgw).  This also enables me to feed the outside temperature to the thermostat, even though my boiler doesn't have an outside temperature sensor.  My thermostat actually has a fire place mode, where it temporarily ignores the room temperature, using the outside temperature instead to control the boiler's water setpoint.  I don't have a fire place, but my living room windows face south west, causing the sun to heat the living room in the afternoon.

In the thermostat's hidden installation menu, the fire place logic can be enabled permanently, but I haven't tried that yet.  Also on my to-do list is turning off hot water comfort mode when I'm away (the thermostat will only do that during the night and while on holidays), and replacing the fixed thermostat schedule by HomeKit automations based on presence.

jo-me, 2019-10-06:
> ZigBee attribute reporting does both change reporting and periodic reporting. I've configured it for 0.1°C or 0.2°C change and every 10 minutes. These settings influence reporting of the temperature to the coordinator, not necessarily measuring the temperature for the TRV's PID algorithm.

Hmm, I don't see that feature in the zwave version. If it reports periodically then it is a large period. Maybe the setting is not present or not exposed through OpenZwave...

> By setting the thermostat a bit (0.5 to 1°C) higher than the living room TRVs, the boiler still supplies heat to the other rooms when the living room is warm enough. When the other rooms are heated [...]

Yeah, that was what I tried when I found out that the thermostats don't turn off.. ;-) The living room had 24°C instead of 20°C . So especially if you're pulling tricks like this it is crucial that the TRVs do not overshoot their setpoint.

Your solution is probably viable, although I'm not sure how the boiler deals with that. Will it really turn off or just keep going and going because the setpoint of the central thermostat is never met?

Unfortunately I cannot control my boiler or the central thermostat due to an old, not reverse-engineered protocol. But there is also a brute force kind of way to control the boiler through a certain clamp that overrides the thermostat and - if bridged - just delivers 100% or 0% in an on/off way.  That could be a way to boost a certain room for a short period of time. Not sure how the interaction between that thing and the modulating thermostat is, though.. 

Coming back to a solution to the actual problem: I do get the impression that after a period of manually controlling the valve instead of using the temp setpoint they behave differently then when they were running normal overshooting thing. Only then I actually saw one turning the valve off (0%) when reenabling a setpoint below the current temperature.

[image]

artkrz, 2020-01-01:
I just got couple of these and have same issue. I’m setting it to off with home assistant but water is still flowing trough it :(

logiclabs, 2020-01-01:
@artkrz I'm looking at this at this moment.. I just reset, refitted my TRV and made sure it was screwed on tight. It now seems to be stopping water flow correctly.

Also, with the overshoot issue... has anybody tried just switching the mode to "Manufacturer
Specific".  This allows direct control of the valve.

This manual is a great reference, compared to the manual that came with the device (and couldn't find this in English on Eurotronics website): 
https://www.cd-jackson.com/zwave_device_uploads/710/Spirit-Z-Wave-BAL-web-EN-view-04.pdf

artkrz, 2020-01-05:
Yes, I have this manual. Did You use any tools to tighten TRVs? Regarding "Manufacturer
Specific" mode, how can I control it? I don't see any details how to do it, it only says it's possible via z-wave.

logiclabs, 2020-01-05:
@artkrz No, didn't use tools, just needed to re-sit the valve and then tightened by hand. 

I haven't tried controlling the valve opening directly yet (I have the zwave version), but it seems to indicate in that mode you can send a percentage open value to a switch entity.

artkrz, 2020-01-05:
I've just did following:

1. Unplug=>plug 1 battery so TRV goes into INS mode
2. Tighten carefully with a wrench, I noticed the ones that reported higher valve opening were already pretty tight
3. Press central button to finish installation process

Maybe this will help. I'm using Home Assistant with Z-Wave no idea how can I send this command.

jo-me, 2020-01-06:
To me it seems as if they gradually lose their calibration after installation. Even with manual valve control I experienced that setting it to 0% to close it completeley did not do that. The motor did turn on for a bit but not enough to really close the valve. Only after setting it to 100% and then to 0% did it really close..

electrofloat, 2020-01-06:
> To me it seems as if they gradually lose their calibration after installation. Even with manual valve control I experienced that setting it to 0% to close it completeley did not do that. The motor [...]

Mine does exactly the same. It seems that it forgets where the closed position is only after a day or so.
I also experienced that it can be "fixed" by setting it to "boost" mode (or pressing the middle button), this fully opens the valve, and then setting it to off will fully close the valve. Then the TRV works again for a day or to as expected, then I have to do this again.

It seems that a lot of people has this issue with this TRV and as if the firmware is completely buggy, and the manufacturer (Eurotronic) just does not care about this issue at all.

Ian-Johnston, 2020-02-13:
Hi all,
I am getting the same problem. The room will overheat, or underheat........sometimes!
If when it's overheating for example if I manually drop the temp by a few degrees then the TRV motor does not appear to move (my ear very close to the TRV)......but if I manually set the temp on the TRV up and down a few times then all service is resumed.
I only moved to the Spirit TRV after the Devolo TRV's on my Devolo system were doing the same! I even changed the radiator valve itself incase it was sticking.
Well, I am going after the manufacturer........totally fed up with this!
Ian.

domik82, 2020-02-16:
@Ian-Johnston I was sending them info last year - haven't got any reply. I'm also seeing similar behaviour. Probably I'll finally figure out some kill switch mechanism inside openhab rules to prevent overheating.

electrofloat, 2020-02-16:
If anyone is willing to try Aeotec's version of this TRV (https://aeotec.com/z-wave-home-automation/thermostat-radiator-valve-trv/) for the better support and possible firmware upgrades/fixes, please let us all know about your experience, so we can all learn from it.

Ian-Johnston, 2020-02-17:
> 
> 
> If anyone is willing to try Aeotec's version of this TRV (https://aeotec.com/z-wave-home-automation/thermostat-radiator-valve-trv/) for the better support and possible firmware upgrades/fixes, please let us all know about your experience, so we can all learn from it.

I have just emailed Aeotec and asked them about their unit and pointed them towards this thread.

Ian.

Ian-Johnston, 2020-02-20:
Aeotec Support have told me:

"I can confirm that it is a re-branding of the Eurotronic Spirit TRV in this case, i am unsure about similar issues which i will raise with our team and try to find out of the highlighted problems are present in this current version."

Ian.

electrofloat, 2020-02-20:
Nice. Impatiently waiting for their answer then :)

kovaga, 2020-03-05:
I have 4 Eurotronic spirit zigbee version, controlled via zigbee2mqtt.
I have the same issue experienced here by many people with valve not closing ompletely. it often remains at 1 or 2 % and the radiator keeps getting hot. The only solution is to turn the boost mode, then turn back to auto. I guess will need to write a rule in OpenHAB to do that  for now.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

rbpasker, 2022-10-11:
> @artkrz I'm looking at this at this moment.. I just reset, refitted my TRV and made sure it was screwed on tight. It now seems to be stopping water flow correctly.
>
> Also, with the overshoot issue... [...]

this link is no longer valid. here's the wayback machine capture: https://web.archive.org/web/20220503111532/https://aeotec.com/z-wave-home-automation/thermostat-radiator-valve-trv/

what happened to their TRV?