dresden-elektronik/deconz-rest-plugin #1934: Conbee_I as router

timota, 2019-10-04:
Hi guys, 
dont know is this is a right place to ask... 
How Conbee_I can be used as router ? 
Is it possible to flash it (somehow), and just plug into charger without PC? or i need PC/Raspbe/etc and run deConz anyway to use it as router ?  

Thanks in advance.

ebaauw, 2019-10-05:
You need the deCONZ GUI to configure the ConBee as a router and make it join a network, see #21.  I don’t know if the firmware does anything useful without deCONZ running.

If you just want to extend the reach of your network, it might be easier to get an IKEA Trådfri repeater.

manup, 2019-10-05:
>I don’t know if the firmware does anything useful without deCONZ running.

The routing functionality also works without deCONZ running. Note that due the power-amplifier of the ConBee the signal range should be way higher compared to usual repeaters like lights and power-plugs.

However I'm not sure if the configured Router configuration will auto reconnect after a power-cycle.

timota, 2019-10-05:
Thanks guys. I will try and let you know.

zehrer, 2019-12-08:
I have RaspBee and just updated the firmware. I try to setup it according #21 as a router and join my Hue Network but joining did not work.
In several older issues (e.g. #633 #653) is mentioned it would not work because the TC Link Key will not be stored. I tested this and it works. After pressing save it store the data but show the wrong data in the GUI, but after pressing the read button, it shows the correct setup.  
But what is the detail setup? (e.g. which Security Mode, NWK Update ID? ...)
I use the famous TC Link Key 5a...39, is this still valid?

ebaauw, 2019-12-08:
You need to use the ZLL link key to connect to the Hue bridge network, see #21.

zehrer, 2019-12-08:
No i have the same issue, i can not change anymore the TC Link Key and i don't know why.

ebaauw, 2019-12-08:
I'm afraid because something is broken in the RaspBee firmware.  You might try and flash it to ancient firmware version 0x26050500.  See http://deconz.dresden-elektronik.de/deconz-firmware/.  You might need to downgrade deCONZ as well to a version that will work with the old firmware.

zehrer, 2019-12-10:
Exists somewhere an offical problem report for this issues? 
At least the product don't work according the manual, which is quite old (v1.06 from 31.10.2012)

carbonelle, 2019-12-13:
Hello

I have the conbee II and can not change the TC Link Key any more !  I use the last firmware 0x264a0700.

Have you an idea?

rkjnsn, 2020-04-12:
I also have a ConBee II and need to change the TC Link Key to join a ZLL network. Is there any extant version of firmware for this device that allows setting the TC key? If not, is there a plan to release a new firmware version that fixes the bug where it cannot be set? Or even a way to modify the firmware so the unchangeable key is the one I want?

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix