dresden-elektronik/deconz-rest-plugin #1936: Senic Friends of Hue

daendaen, 2019-10-05:
Hi,

today my senic switches have come. But deconz can't find the switch.

Can not the switch be integrated?

Greetings

daendaen, 2019-10-09:
Now the switch is connected to the conbee stick. But the switch doesn't work. 
I have assigned the switch a Osram Light, but nothing happens

Important infomation for connect the switch comes from dresden elktronik Mr. Preuß. Thank you.

I little manual:
1) Lock in the Gateway which Zigbee channel you use.
2) Go in the App to Switch and select other switch
3) Then press the button from the richt channel on the switch with the energy bar for 7-12 seconds

Now The switch is recognized.

 
[image]

closed 2019-10-12 by daendaen

fazerize, 2019-10-29:
Hey, did you get the switch to work?
I asigned the buttons but they won't work.
Idk what I'm doing wrong...

manup, 2019-10-29:
We've just created a small FAQ entry on how to join the Friends of Hue switches. Note that step 5) is very important to keep the switch on the required Zigbee channel.

https://phoscon.de/en/support#pairing-friends-of-hue-switch