dresden-elektronik/deconz-rest-plugin #1940: New Ikea filament bulbs not showing up on scans

roaringbits, 2019-10-05:
I've bought three of these bulbs at Ikea:

https://www.ikea.com/nl/nl/p/tradfri-led-lamp-e27-250-lumen-draadloos-dimbaar-warm-schijnsel-globe-bruin-helder-glas-70455676/

After resetting them with a Hue dimmer swith and then powercycling the device after starting the search for lights, they seem to be totally invisible to the Conbee II - Although the bulb is right next to the Conbee. Although not (yet?) supported, they should at least show up as new nodes right?

Any approaches I can try for these bulbs?

manup, 2019-10-05:
Usually the bulbs can be reset as described in https://phoscon.de/en/support#faq-ikea1

roaringbits, 2019-10-05:
Seems I was not resetting correctly. Thanks!

closed 2019-10-05 by roaringbits