dresden-elektronik/deconz-rest-plugin #1941: "Creating backup failed" in Phoscon - 2.5.69 fw 26490700

jose-d, 2019-10-05:
I can not backup using the Phoscon web app with "Creating backup failed" message.

Problem is reproducible also with fresh session (in Private window) of Firefox as I originally suspected just some session timeout.

here is IMO relevant entry from log:

```

21:47:58:390 HTTP API POST /api/909DDC09DC/config/export - 192.168.1.220
21:47:58:390 API error 1, /config/export, unauthorized user
21:47:58:390    host: 192.168.1.220
21:47:58:390 [{"error":{"address":"/config/export","description":"unauthorized user","type":1}}]
21:47:58:458 Incr. ZDP retry count 2 on item 7
21:47:58:853 HTTP client hdr detected:
GET /api/909DDC09DC/config?_=1570304848325 HTTP/1.1
Host: 192.168.1.166
User-Agent: Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:69.0) Gecko/20100101 Firefox/69.0
Accept: vnd.ddel.v1
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
Referer: http://192.168.1.166/pwa/settings-gateway2.html?_v=efc3de87f3aa8666865a4cdee64d46b76795afb8,9,6,21&gwid=00212EFFFF0487DB
X-Requested-With: XMLHttpRequest
DNT: 1
Connection: keep-alive
```
collected with 

```
/usr/bin/deCONZ -platform minimal --dbg-info=2  --dbg-http=2 --http-port=80 > deconz.stdout1
```

platform: conbee II in debian buster LXC container on top of TurrisOS@turris Omnia router.
```
root@lxc-deconz:~# cat /proc/cpuinfo | grep ARM | uniq
model name      : ARMv7 Processor rev 1 (v7l)
root@lxc-deconz:~# cat /etc/debian_version 
10.1
root@lxc-deconz:~# uname -r
4.4.191-a890a5a94ebb621f8f1720c24d12fef1-0
root@lxc-deconz:~# dpkg -l | grep deconz
ii  deconz                        2.05.69-raspbian-buster-beta armhf        A generic ZigBee monitoring and control tool
root@lxc-deconz:~#
```

if needed, i can provide more logs.

jose-d, 2019-10-05:
possibly related #1892 ?

jose-d, 2019-10-05:
Problem was between chair and keyboard.
When architecting the LXC deconz stack I haven't realized consequences of running the deconz as root user first. App itself created

```
./.local/share/dresden-elektronik/
```

with ownership of root and app itself was not able to update the config.

so not a bug.

closed 2019-10-05 by jose-d

nunoary, 2019-10-09:
@jose-d
 I noticed you already solved this, you can help me solve mine. I read the answer in your post and I think my problem is the same as yours but I can't understand how you solved it.

My deCONZ is installed on Windows 10 in

C:\Program Files\deconz

and the Zll.db it's on

~/local/dresden-elektronik/deCONZ

I think i have the userrights for this folders, how can i confirm that? In the folder properties i have the permissions all.

jose-d, 2019-10-10:
hi;

> I can't understand how you solved it.

in my case the problem was, that deconz was trying to create backup file, but it had not enough permissions in the relevant directory - in my case `~/.local/share/dresden-elektronik/` - I must emphasize that it was my mistake when I was doing custom LXC installation (it's running inside of my Wi-Fi router) - so **NOT bug in deconz**.

I'm not sure if windows is right platform for deCONZ, especially if you're not advanced user. Windows has quite complex system of permissions so TBH, I'd suggest to invest few € into raspberryPi3, guys at Deconz did great work when testing and tuning their software for such a platform and - last thing but not less important Linux and its environment is easy to debug and the it has great community support.

nunoary, 2019-10-10:
Thanks,
You're right about windows, complicate some things too much.
Thanks for the suggestion, I will ponder ... but I haven't given up yet.

Markkuuss, 2019-11-26:
I don't understand how to set the permissions? I have deCONZ running on a Raspberry Pi under the user "Pi", who has the necessary permissions for this directory.  Nevertheless it does not work.

jose-d, 2019-11-26:
> I don't understand how to set the permissions? I have deCONZ running on a Raspberry Pi under the user "Pi", who has the necessary permissions for this directory. Nevertheless it does not work.

yes, but that was not my case. I have dedicated user for this service and he had no access to the `~/.local/share/dresden-elektronik/`

Markkuuss, 2019-11-26:
Mhhh... that's weird. My user definitely has access. Is it possible to change the path for the backup in a config file?

I have also tried to read something from the log file: deCONZ --dbg-info=2 --dbg-http=2

However, I do not find any error message.

jose-d, 2019-11-27:
if you have some linux knowledge, you can use `strace` to realize what is going on. If not, then I strongly recommend to use the oficial raspberryPi image.

Markkuuss, 2019-11-27:
I was recording a trace while trying to make a backup. However, I don't find any error in this trace file.

Here is a link to the trace, if someone can interpret it better. :-) 
https://drive.google.com/file/d/1s75jm5CYjoWZ_ha63o3JoP-8dod5pKWd/view?usp=sharing

jose-d, 2019-11-28:
> I was recording a trace 
mhmm, which strace options have you used?
Please use also the "-f" to trace the child processes too.

Markkuuss, 2019-11-30:
I have now executed strace again with the -f parameter.

I think from line 65118 (`read(22, "POST /api/42683BC9DE/config/expo"..., 565) = 56"`) the relevant trace follows. However, I could not derive an error from it again.

https://drive.google.com/file/d/1s75jm5CYjoWZ_ha63o3JoP-8dod5pKWd/view?usp=sharing

tnTano, 2020-01-24:
hi to all, i have the same problem...
i use:
- raspberry 4 (4gb ram)
- homeassistant on docker (0.104.3)
- deconz addon (5.1)
- conbee2 (2.05.76)

This is some log i have in HA plugin page:
> try to reconnect to network try=4
> Device protocol version: 0x010B
> Current channel 11
> CTRL ANT_CTRL 0x03

if i use this command `/usr/bin/deCONZ -platform minimal --dbg-info=2  --dbg-http=2 --http-port=80 > deconz.stdout1`
i have this output:
> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
> libpng warning: iCCP: known incorrect sRGB profile
> This plugin does not support propagateSizeHints()
> This plugin does not support propagateSizeHints()
> This plugin does not support propagateSizeHints()

and i cannot create backup...

i need this because that config not working anymore, all the sensors and lights not responding...
i try to install deconz in standalone mode in another raspberry and i need backup because i have some light that i cannot rebind...

thanks for your help