dresden-elektronik/deconz-rest-plugin #1942: reachable: false status message slow?

Appelg, 2019-10-06:
**What I do:**
1. Physically turn off my IKEA light
2. Wait for the Websocket connected to deconz to receive a message 

It then takes around 20 minutes before I get this message:

```
{ 
   "e":"changed",
   "id":"5",
   "r":"lights",
   "state":{ 
      "alert":null,
      "bri":254,
      "colormode":"ct",
      "ct":454,
      "on":false,
      **"reachable":false**
   },
   "t":"event",
   "uniqueid":"00:0b",
   "_session":{ 
      "type":"websocket",
      "id":"2a57a"
   },
   "_msgid":"fa2a85",
   "_event":"node:5fe"
}

```
Why does it take so long for deconz to figure out that the light has been physically turned off?
Is there a setting to improve that (polling?) time?
I would like to get this info in at least less than 60 seconds or so...

Smanar, 2019-10-06:
I m interested too, it's managed by deconz or the API ?

closed 2020-02-10 by stale[bot]

ebaauw, 2020-03-15:
`reachable` is utterly useless and best ignored, see #2590.