dresden-elektronik/deconz-rest-plugin #1945: Firmware update issue and deConz dependensies

Luigi-01, 2019-10-07:
Hi,
I have send you an email two weeks ago, only no reply till now.
After an attempt to update the firmware, my Combee II stick crashed. Useless after that.
Second after installation of deConz my Ubuntu computer, there is a message regarding 
missing dependencies. Which dependencies are required for deConz.
deConz is not able to install these dependencies without Super User rights.
The problem will never be solved this way.
Greetings,

SwoopX, 2019-10-07:
The dependencies can be installed as follows:

`sudo apt install -f`

Make sure the sources have been updated before.

manup, 2019-10-07:
Can you please provide more details on your Ubuntu version and platform?
Currently only 16.04 and 18.04 are supported via APT repository, 19.04 will be added soon.

https://phoscon.de/conbee2/install#ubuntu

Other versions can be installed manually via .deb package from:
https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages-for-manual-installation

    wget https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.05.69-qt5.deb
    sudo dpkg -i deconz-2.05.69-qt5.deb
    sudo apt install -f

Luigi-01, 2019-10-08:
Hi,
I am running Ubuntu 18.04.
Something else I noticed, the stick is fighting with all other USB sticks. (Z-wave, P1 USB.)
The Conbee II only want to be connected with ttyusb0.
For that reason I have moved all other USB devices.
See: https://www.domoticz.com/wiki/PersistentUSBDevices
All programs wants to use port 80 and port 443 and ttyusb0.
It is a never ending story. 
Now I have to wait, if I get a new Conbee II stick.  When I get a new stick
I will inform you have things are going.

manup, 2019-10-08:
>Something else I noticed, the stick is fighting with all other USB sticks. (Z-wave, P1 USB.)

The default name for ConBee II always is /dev/ttyACMx (x = 0, 1, ...).
/dev/ttyUSB0 is certainly not the ConBee II.

Do you use an USB extension cable?

Luigi-01, 2019-10-08:
Hi,
Than the Conbee II has been fighting with the P1 connection.
However, when the stick was connected with my desktop computer. 
He was located on ttyusb0. and according to deConz no issues.
And NO, he was not connected by an extension usb cable.
He was connected with an USB3 port. (directly)

Luigi-01, 2019-10-16:
Hi,
After receiving a new stick, I  installed deConz again. 
However: sudo apt install -f is not the missing dependencies. 
One improvement, deConz now asks for the SU password and continues.
Deconz is asking for this password every times it starts.

closed 2020-02-20 by stale[bot]