dresden-elektronik/deconz-rest-plugin #1947: Unable to upgrade IKEA remote

StoricU, 2019-10-07:
Hi,

I have downloaded the firmware to my computer, and tried to update two remotes, but I'm not sure how the prosess goes.
When I'm trying to update, the OTAU plugin only states Queued or Paused.
Anybody have a guide to what I should do to upgrade?
Or what I have to to to manually start the update process?

I'm running Deconz on a Windows now just for the upgrade, but normally I run on a Raspberry Pi with Conbee. So a guide on any of the OS's I can work with :D

Philje123, 2019-10-09:
You need to remove the battery and re-insert before it will initiate the upgrade.

I've updated a few Ikea dimmers and remotes successfully via deconz.

StoricU, 2019-10-09:
> You need to remove the battery and re-insert before it will initiate the upgrade.
> 
> I've updated a few Ikea dimmers and remotes successfully via deconz.

Thanks. Do you initiate it with pressing the update button, or the guery button?

StoricU, 2019-10-09:
Tried one more time now.
But the status is only "Queued". Have tried the battery trick also now.
[image]

Philje123, 2019-10-10:
I found it a little hit and miss to get going. What I did was hit the query button then update a few times, then took the battery out and put it back in.

I think there was also an issue that I saw mentioned elsewhere too where it helps to restart deconz prior to doing an OTA too.

WhistleMaster, 2019-10-11:
This is how I updated mine:

1. Download the latest firmware and put it in ~/otau
2. Remove battery from Remote and put it back
3. Click on "OTAU" attribute and click on "Read" in "Cluster info" panel
4. "Image upgrade status" should change to "Download in progress"

[image]

StoricU, 2019-10-11:
Thanks. Will try this.
But for a newbee in Linux as myself, where should I put the otau folder?
What does ~/ mean? :)

closed 2020-02-15 by stale[bot]