dresden-elektronik/deconz-rest-plugin #1948: Deconz don't react on IKEA remote signal

StoricU, 2019-10-08:
I have two IKEA remotes that I'm trying to use with Deconz on a Raspberry Pi running Raspbian.
The problem is that 80% of the time i press a button, there is no reaction.
I'm using Deconz with Homeseer to control my garage door, but sadly since it is just sometimes it works, the WAF is going down!

Anyone with an idea what to try?

I'm still using Deconz .65, since .67 and .69 disturbs my whole network.

Philje123, 2019-10-09:
It sounds like a routing problem. How far away is the conbee and/or the nearest router (bulb etc)?

StoricU, 2019-10-09:
The conbee is approx 2 meter away from the remote. Just one floor up.
Have also tried setting up a new light beside the remote, so that the routing goes thru that one, but no difference. 
Have also moved the remote so that the distance have been approx 1 meter, but stil there is problems.

nunoary, 2019-10-10:
I'm not an expert, but what equipment are you trying to control with the remote?
Is the action controlled by deCONZ or Homeseer?
If it is by Homeseer, you have disabled in the command equipment the option "Do not update device last change time if device value does not change"?

I don't know if this is the case with Ikea commands, but with Xiaomi I have come to the conclusion that it is better to do control with deCONZ instead of Homeseer.

StoricU, 2019-10-10:
> I'm not an expert, but what equipment are you trying to control with the remote?
> Is the action controlled by deCONZ or Homeseer?
> If it is by Homeseer, you have disabled in the command equipment the [...]

The action is controlled by Homeseer, and yes, that option it disabled.
It work some times. If I just keep presseing the same button, suddenly it reacts. And then it can react several times in a row.
Looking at the deconz display, there is just no activity on the remote when pressing. Just once in a while when I press, I see the blue flashing colour on the remote device.

manup, 2019-10-10:
>I'm still using Deconz .65, since .67 and .69 disturbs my whole network.

How do you mean this? .69 should run way smoother than .65.

StoricU, 2019-10-10:
> > I'm still using Deconz .65, since .67 and .69 disturbs my whole network.
> 
> How do you mean this? .69 should run way smoother than .65.

When I tried .67 and then .69 everything stoped.
None of the connected devices would work properly.
Just sporadic connectivity resulting in no reaction when trying to control them.
Even though the lights where turned on, they showed offline in Phoscon. And suddenly they where online, but after a while, they where offline again.
Drove me nuts.

nunoary, 2019-10-10:
You need to rescan the devices. With the Phoscon App start the option to add light/switch or sensor then reset the device. Do this to all devices, you don't need to put the devices near conbee/rasp, just do it where the device is.

V .69 as much more stability for me.

StoricU, 2019-10-10:
> You need to rescan the devices. With the Phoscon App start the option to add light/switch or sensor then reset the device. Do this to all devices, you don't need to put the devices near conbee/rasp, just do it where the device is.

Tried this.. But nothing.. After upgrading to .69 I started the option to add light. Reset 3 of my lights. Don't think there was any change in the Deconz app. Tried to adjust the lights when search was finished. Nothing..
No reaction on any of the lights.
Just to test, I completely removed one light fra deconz/phoscon and tried to add it from scratch. But now I'm not able to add the light.
As long as the inclusion rutine hasn't changed from .65, something is not working correctly! :(

StoricU, 2019-10-10:
I downgraded to .65 again, but this time it didn't go as smooth as last time.
Had to try several times to include the lights again, and now that I managed to include one of them, I can control it. It shows connected, but nothing happens when I adjust it via Phoscon...

Is there a big master reset? That I can use to delete everything from Raspbian and ConBee so that I can start completely fresh?
I'm starting to loose my hair with all this trouble.

svenove, 2019-10-11:
This sounds like a HomeSeer “problem”. Are you using “a device had it’s value set to”? Or are you using “a device changed and becomes”? The last won’t work when you keep pressing the same button without any other button presses in between.

StoricU, 2019-10-11:
Have checked this.
And I'm using "a device had it’s value set to".
Anyway, that is related to events.
I'm just happy if the child device updates at all. :/
I have also been monitoring deconz gui when pressing the buttons on the remote.
And when the device reacts(adjust light) and updates in Homeseer, the remote flashes on the deconz-gui.
When nothing happens (that is almost all the time now) there is no blue flashing on the gui.
So I'm guessing there has to be some connection issue and not in Homeseer.

StoricU, 2019-10-15:
No one have any idea how I can do a complete reset of everything?
My setup seems to not work on any version at the moment...

WhistleMaster, 2019-10-29:
I have something similar with the IKEA remote (5 buttons). After a non-using period, the first press is not taken into account, I have to press a second time, and then it works fine. I'm using Phoscon to configure the actions of each button of the remote.

DerOetzi, 2019-10-29:
Same behavior here, when not using remote for a while it sometimes reacts only after several clicks. But after this initial lack it works fine again.

DerOetzi, 2019-10-29:
Oh I forgot, I did some testing. The lamps can be controlled with Hue Binding in openhab like a charm and it doesn't matter how long the have been off before but remote needs again some clicks before reacting. Remote is always shown as connected in Deconz network view.

StoricU, 2019-10-30:
Then I'm not the only one with this problem.
Since the remote is configured to open my garage door, that's out of view, this solution is not good enough for me. Have to work on the first click.

DerOetzi, 2020-02-17:
Anything new to this issue?

Mimiix, 2020-06-05:
@DerOetzi Is this still a issue?

DerOetzi, 2020-06-05:
I'm not sure. I switched to use deconz only as gateway and to the control rules in OpenHAB. This works as expected. So I can't say anything on this.

Mimiix, 2020-06-05:
@DerOetzi Thanks for your quick reply. For now i'll close the issue, if anyone still has this problem we'll re-open it. Thank you!

closed 2020-06-05 by Mimiix