dresden-elektronik/deconz-rest-plugin #1950: Sonoff Zigbee

Klagio, 2019-10-09:
I just noticed that Sonoff ZIGBEE version is out. Fantastic news for me. IS this already working with Conbee?

https://www.itead.cc/sonoff-basiczbr3.html

WhistleMaster, 2019-10-11:
Would be great to have it working with DeConz indeed !

Philje123, 2019-10-12:
I saw someone on the Home Assistant community boards had tried one with Deconz and it worked straight away.

https://community.home-assistant.io/t/sonoff-basiczbr3-zigbee-diy-smart-switch/140105/4

andyjenkinson, 2019-10-27:
If anyone who has one could post which profile it is supporting that would be great. Interested to know if it is zigbee 3.0 or ZHA.

Klagio, 2019-10-28:
but noi meshing so no repeater function right?

cobirnm, 2019-11-23:
> but noi meshing so no repeater function right?

I have just installed a Sonoff basicczbr3. It works also as repeater.

Tanddant, 2019-11-27:
Mine appears as a lightbulb, how can I change this?

Smanar, 2019-11-27:
I think you can't, it something normal on phoscon.

Tanddant, 2019-11-27:
@Smanar Do you know of any way to map it as a switch when exposed to Home assistant?

WhistleMaster, 2019-11-27:
It will be automatically mapped to a switch then in HA.

Tanddant, 2019-11-27:
Mine appears as a light in home assistant

Smanar, 2019-11-27:
IDK for Home assitant but in the API.
Battery switch are detected as switch because they can only send command
Wired switch are detected as bulb because they can receive command.

zladukas, 2019-12-05:
It looks like supports groups 

"20": {
    "etag": "f7eac08e0f573a42a2b7bb673a46bd82",
    "hascolor": false,
    "manufacturername": "SONOFF",
    "modelid": "BASICZBR3",
    "name": "Sonoff1",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": null,
    "type": "Unknown",
    "uniqueid": "00:12:4b:00:1f:XX:XX:a0-01"
  },

[image]

[image]

but when reading Groups and Scenes cluster, deCONZ GUI (2.05.70) crashing with:

Dec  5 18:49:21 ddpi systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV
Dec  5 18:49:21 ddpi systemd[1]: deconz-gui.service: Unit entered failed state.
Dec  5 18:49:21 ddpi systemd[1]: deconz-gui.service: Failed with result 'signal'.

Gamester17, 2019-12-10:
Anyone verified and confirmed which Zigbee profile it is using exactly?

zladukas, 2019-12-10:
@Gamester17  tell me how to find it, I will verify.
My guess it's: On/Off Light | 0x0100

Smanar, 2019-12-10:
ZHA with DEV_ID_MAINS_POWER_OUTLET           0x0009 // Mains power outlet ???

realwax, 2019-12-17:
Hi, just found out about Sonoff zigbee. So who has it successfully integrated into deconz? I don't worry using the GUI but will it be present in the Lights/Switch section and will it be exposed to e.g deconz iobroker instance? Thanks for you infos... Would be really great having this. (Avoid WIFI ones with tasmota.)

cobirnm, 2019-12-17:
> Hi, just found out about Sonoff zigbee. So who has it successfully integrated into deconz? I don't worry using the GUI but will it be present in the Lights/Switch section and will it be exposed to [...]

I have had it for a month now, and it integrates as a light and for me it gets exposed to Home Assistant, so it should work in the same way to you

SnaKKo, 2020-01-06:
@cobirnm (or anyone who knows the answer): I would like to know how this device works in case of powercycle: restore of previous state, power on or power off?

Thanks and sorry for this slightly offtopic question!

SnaKKo, 2020-01-07:
Just to help other people with the same question/doubt: it seems to restore the previous state.

Gamester17, 2020-01-08:
@SnaKKo Thanks! By the way, then the technical term for that is "Bistable Switch" or "Latching Relay"

damienclaveau, 2020-01-08:
Hello. 
I have also installed one basicczbr3. 
Successfully discovered, in a couple of seconds, as a light by Deconz through a Conbee 2.
I also confirm it behaves as a Latching Relay.
Works nicely out of the box so far.

tomburke74, 2020-01-13:
Hi, anyone had any luck with Deconz and a Raspbee?

cs42, 2020-03-03:
Hi, BASICZBR3 is basically working for me, but **scenes** are not working at all. When I try to read the scenes cluster in GUI, deCONZ crashes completely. Scenes defined in Phoscon aren't stored as well. Can I help with some debug information somehow?

serrnovik, 2020-04-17:
Running Phoscon in docker on synology, 2.05.75 / 3/8/2020, Conbee II. **Can't discover/pair them**. I'm trying through Phoscon UI. VNC on docker is not accessible (probably due to docker port mapping and can be fixed - will try later, had no need for it before). Home assistant doesn't see sonof, so it is not just UI issue. 

Has anyone who reported them working together is using docker version?

pippo73, 2020-05-30:
I have the same problem as @serrnovik .
I have a conbee2 stik on a RBPi 4 with rasbian/Home Assistant (0.110.3) and deconz (2.05.77 FW 264A0700)

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

If I try to connect the sonoff to it, it does not work as light or switch.
If I try to connect it to Alexa, it is detected as switch (sonoff smart device)

what is the procedure to make it detectable?

I've tried to push the "reset" button for 5 sec untill the red lighr starts to bllink and then I've started the research on deconz interface...

Do I miss something?

Mimiix, 2020-06-12:
@pippo73 Do other devices work? Please open a seperate issue on your problem. Thanks!

closed 2020-06-12 by Mimiix

morfei1, 2020-06-18:
@serrnovik , @pippo73 try to connect the Basiczbr3 with the workaround described in #2110 

regards!

sunshineautomate, 2022-04-20:
My experience with this is that it is not a latching relay. It will resume the previious state if the power is cycled back on within a short period of time but if power is dropped for a few minutes it will revert to off. Can anyone else confirm or deny?