dresden-elektronik/deconz-rest-plugin #1952: Philips Hue Presence sensor - lastupdated

mplogas, 2019-10-10:
**System**

- RaspBerry 2B with RaspBee
- Raspbian Stretch Headless Image
- Version: 2.5.67 / 26.08.2019
- Firmware: 26330500

**Describe the bug**

The `lastupdated` field of the Hue motion sensor is showing the timestamp of the last movement correctly, but only if this happened within the last ~5 minutes. When the presence detection is older than said 5 minutes, a timestamp within the last 5 minutes is shown. 

**To Reproduce**

1. Set up a Hue presence sensor with Deconz
2. Make sure the sensor is detecting and reporting presence (use REST API)
3. Put your sensor somewhere where it won't be able to detect any presence (e.g. closet, empty & closed room, ...)
4. do another REST call and note the timestamp in the `lastupdated`field
4. wait >5 minutes
5. call the REST API again, the `lastupdated` property of the sensor will have changed without any presence being detected

**Expected behavior**

The `lastupdated` field only updates when presence is detected, e.g. with Tradfri sensors the `lastupdated` field is only updating when a presence is detected.

ebaauw, 2019-10-10:
The Hue motion sensor sends a message to the coordinator every five minutes; `state.lastupdated` reflects the time the last message was received.  Note that different brands of motion sensors work very differently: the Hue motion sensor sends periodic messages and a message when motion is no longer detected; most other motion sensors only send a message when motion is detected.

mplogas, 2019-10-10:
Thanks for the info. I wonder, is there a way to retrieve the last real presence update through deconz, regardless of the brand?

hacklst, 2020-01-01:
The problem is: This issue renders existing software-features useless, since other software interprets the "lastupdated" field as the "time of last movement detected".

I've tested with original HUE app <-> DIYHUE <-> deConz <-> HUE motion sensor. 
AFAIK DIYHUE uses the REST-API in combination with the WS to obtain the data from deConz.

When setting a standard "after X minutes without motion"-action, the action never fires since every few minutes a fake update (from "presence 'false'" to "presence 'false'") is fired, making the rest of the software chain believe that a movement has been detected recently.

Since I cannot compare to an original HUE gw, I do not know if the deConz-API acts wrongly or if the DIYHUE's interpretation of the data is wrong.

Can you please clarify? If the faulty behaviour is in deConz-REST-API, a fix would be highly appreciated.
If the deConz-REST-API acts as the original API would, then I'd get in touch with the DIYHUE-Developers and/or fit the behaviour there myself. Thanks.

Sideremark: Switching things on works correctly though.

closed 2020-05-07 by stale[bot]