dresden-elektronik/deconz-rest-plugin #1953: ubuntu conbee 2 installation instructions

daniel-lo, 2019-10-10:
Followed the instructions (hopefully) precisely but I cant't install deconz in my Proxmox VM (Intel NUC). I assume the instructions need an update:

`daniel@deconz:~$ sudo sh -c "echo 'deb [arch=amd64] http://phoscon.de/apt/deconz \$(lsb_release -cs)-beta main' > \/etc/apt/sources.list.d/deconz.list"
daniel@deconz:~$ sudo apt update
Ign:1 http://phoscon.de/apt/deconz $(lsb_release InRelease
Fehl:2 http://phoscon.de/apt/deconz $(lsb_release Release
  404  Not Found [IP: 81.169.237.235 80]
OK:3 http://archive.ubuntu.com/ubuntu bionic InRelease
Holen:4 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88,7 kB]
Holen:5 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74,6 kB]
Holen:6 http://archive.ubuntu.com/ubuntu bionic-security InRelease [88,7 kB]
Paketlisten werden gelesen... Fertig
E: Das Depot »http://phoscon.de/apt/deconz $(lsb_release Release« enthält keine Release-Datei.
N: Eine Aktualisierung von solch einem Depot kann nicht auf eine sichere Art durchgeführt werden, daher ist es standardmäßig deaktiviert.
N: Weitere Details zur Erzeugung von Paketdepots sowie zu deren Benutzerkonfiguration finden Sie in der Handbuchseite apt-secure(8).
daniel@deconz:~$ sudo apt install deconz
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
E: Paket deconz kann nicht gefunden werden.
daniel@deconz:~$`

Please help!

daniel-lo, 2019-10-10:
it returns "codename: bionic" .. don't know whether that is meaningful or not.

But my most pressing probelm si that I cannot install deconz "no packet found".

manup, 2019-10-10:
>Das Depot »http://phoscon.de/apt/deconz $(lsb_release Release«

Looks like the file /etc/apt/sources.list.d/deconz.list doesn't contain the correct content.

On your system after invoking:
```
sudo sh -c "echo 'deb [arch=amd64] http://phoscon.de/apt/deconz \
            $(lsb_release -cs)-beta main' > \
            /etc/apt/sources.list.d/deconz.list"
```

/etc/apt/sources.list.d/deconz.list should become:

```
deb [arch=amd64] http://phoscon.de/apt/deconz bionic-beta main
```

daniel-lo, 2019-10-10:
edited deconz.list .. but still doesn't work:

`daniel@deconz:~$ sudo apt install deconz
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
E: Paket deconz kann nicht gefunden werden.
daniel@deconz:~$ cat /etc/apt/sources.list.d/deconz.list
deb [arch=amd64] http://phoscon.de/apt/deconz $(lsb_release -cs)-beta main
`

daniel-lo, 2019-10-10:
regrettably no:

daniel@deconz:/$ cat /etc/apt/sources.list.d/deconz.list
deb [arch=amd64] http://phoscon.de/apt/deconz bionic-beta main
daniel@deconz:/$ sudo apt install deconz
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
E: Paket deconz kann nicht gefunden werden.
daniel@deconz:/$

daniel-lo, 2019-10-10:
I removed the -beta and now it works .. don't know if I messed up .. but right now it is installing. Hope I can figure out my way forward now. Is it ok to not use -beta?

daniel-lo, 2019-10-10:
Thank you .. Phoscon up and running!

closed 2019-10-11 by daniel-lo