dresden-elektronik/deconz-rest-plugin #1956: When is next stable release planned?

osos, 2019-10-11:
I am trying to stick to the stable channel.

When is the next stable release from master planned?

yamlCase, 2019-10-13:
@osos Are there other channels besides release/stable?  I figured I'd just have to clone and build, but if they have releases I'd like to save some effort.

manup, 2019-10-13:
Currently there are two channels, stable and beta, which can be installed via APT on Linux.

https://phoscon.de/en/conbee2/install

The next version 2.05.70 will be released within the coming week.

osos, 2019-10-15:
Thanks for the update.

closed 2019-10-15 by osos

mitch-dc, 2019-10-22:
Any update about the release?

Bladestar2105, 2019-10-24:
Also waiting for the update :-)

caro-coder, 2019-10-24:
Oh yes, an update would be amazing!

bratanon, 2019-10-29:
@manup: any updates her? Were the week supposed to be "weeks"?

manup, 2019-10-29:
Yes as usual time planning has gone south ;)

The 2.05.70 APT-repositories for beta channel is online now, I'd like to wait until weekend before also pushing that to stable channel. 

The 2.05.70 release notes and copy of packages to manual installation directories are in progress. Copying is still done manually, I'm working on to automate that within the auto-build process. The Windows packaging is now automated, this took a while but works really smooth now. Next steps is the integration into Jenkins tooling and also creation of the macOS deCONZ.app package (which made some progress in the last days).

ebaauw, 2019-10-29:
> also creation of the macOS deCONZ.app package (which made some progress in the last days).

Yeah!

osos, 2019-10-29:
Just upgraded.

Now all lights seem to have lost pairing.

Raspberry PI - headless.

Running GUI shows the lights etc (all yellow) but no routes

osos, 2019-10-29:
Found this: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1226

Performing a "Scan for devices" in Webapp crashes/restarts deConz gui.

Please advise how to get the lights back online

osos, 2019-10-29:
ConBee II

manup, 2019-10-29:
Ok, I guess it's not related to the release but an already known error (which we'll fix with an later firmware update).

Please try:

1. Stop deCONZ service
2. Unplug ConBee II
3. Wait 10 seconds
4. Plug-in ConBee II again
5. Wait 10 seconds
6. Start deCONZ service again

osos, 2019-10-29:
Unfortunately not solving it.

However, it did seem that permit join made it recognize the ikea repeater name, but still no routes showing

manup, 2019-10-29:
Hmm strange, is the firmware version shown in the *Phoscon App > Menu > Settings > Gateway*?

osos, 2019-10-29:
Yes, firmware: 264A0700

Not sure about the permit join did anything, maybe the range extender was simply named due to deconz internal recognizing it as such.

osos, 2019-10-29:
I have Ikea bulbs and remotes, and Xiamo temp/hum sensors and Cube.

The Xiami sensors are getting routes in deCONZ after pushing the button, while not updating temp/hum in Phoscon. Xiami Cube suddenly adds route, but no event update in Phoscon.

osos, 2019-10-29:
starting deConz using systemd non-gui, the Phoscon firmware says "not connected"

osos, 2019-10-30:
Could the Conbee II have been bricked... 

In dmesg I notice this:

```
[28042.538216] usb 1-1.4: USB disconnect, device number 21
[28042.843188] usb 1-1.4: new full-speed USB device number 22 using dwc_otg
[28042.977655] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[28042.977669] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[28042.977679] usb 1-1.4: Product: ConBee II
[28042.977689] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[28042.977698] usb 1-1.4: SerialNumber: DE2120472
[28042.980158] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[28046.378217] usb 1-1.4: USB disconnect, device number 22
[28046.673251] usb 1-1.4: new full-speed USB device number 23 using dwc_otg
[28046.817662] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[28046.817679] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[28046.817684] usb 1-1.4: Product: ConBee II
[28046.817689] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[28046.817702] usb 1-1.4: SerialNumber: DE2120472
[28046.818976] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[31146.026207] usb 1-1.4: USB disconnect, device number 23
[31146.333756] usb 1-1.4: new full-speed USB device number 24 using dwc_otg
[31146.468144] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[31146.468156] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[31146.468161] usb 1-1.4: Product: ConBee II
[31146.468166] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[31146.468172] usb 1-1.4: SerialNumber: DE2120472
[31146.470397] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[31149.866173] usb 1-1.4: USB disconnect, device number 24
[31150.163726] usb 1-1.4: new full-speed USB device number 25 using dwc_otg
[31150.298044] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[31150.298059] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[31150.298069] usb 1-1.4: Product: ConBee II
[31150.298078] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[31150.298087] usb 1-1.4: SerialNumber: DE2120472
[31150.300544] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
```
Several disconnect and reconnects on USB. Didn't see them before. Raspberry PI 3

manup, 2019-10-30:
>Could the Conbee II have been bricked...

No it can't be bricked, the software is protected against that.

But the bootloop must have an reason.
Currently known related issues:
*  ConBee II is connected ,without USB extension cable, directly to the Raspberry Pi   
   (recommend always use extension cable)
* USB 3.0 devices like disks or flash sticks nearby
* ConBee II boot sequence is disrupted by deCONZ or another service writing to the USB port

Can you please provide the output of:

1. `ps ax | grep deCONZ`
2. `systemctl status deconz`
3. `systemctl status deconz-gui`
4.  `lsusb`

It would also be helpful to see the dmesg output when connecting the ConBee II without deCONZ running.

osos, 2019-10-30:
I will provide this, once home again. I do not have a USB extension cable so this could be one of the problems.

What I tried before leasving this morning was.

- manually reflash conbee II with latest firmware - worked no problems
- Phoscon app still have problems connection to Conbee II in headless mode - ok in gui mode
- remove one ikea light bulb and trying to repair it - not working
- uninstall deconz beta (.70)
- install stable (.67)
- Headless mode - Phoscon app connects fine
- still network did not solve itself
- the previously removed lightbulb not re-paired and connected.

Hope this gives something. Will continue the search tonight or tomorrow.

arnonel, 2019-10-31:
Same thing happened to me. Did apt-get upgrade and now lost all connection to lights

manup, 2019-10-31:
:/ hmmm, have you tried the steps mentioned above?

arnonel, 2019-10-31:
> :/ hmmm, have you tried the steps mentioned above?

Yup, didnt work

arnonel, 2019-10-31:
scratch that - 2mins later - its appeared again

manup, 2019-10-31:
> scratch that - 2mins later - its appeared again

Do you mean it works or it doesn't?

arnonel, 2019-10-31:
> > scratch that - 2mins later - its appeared again
> 
> Do you mean it works or it doesn't?

Works again. Steps above worked. Not immediately - seems to have resolved a few mins later. Tx