dresden-elektronik/deconz-rest-plugin #1958: Xiaomi Magic cube not visible in editor

KlausHeynen, 2019-10-12:
I've successfully added a Xiaomi Magic Cube (Zauberwürfel) to Phoscon it is displayed but in Details showing 10.13° all the time.
But even worse it does not Show on the "Schalter Editor" at all.
I allready apt-get remove deconz and installed again (2.0.5.69 / 26330500) but did not help.

Smanar, 2019-10-12:
Try to include it a new time, without deleting it, but with moving it during inclusion
> https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1846

KlausHeynen, 2019-10-12:
Dont't really understand, is German language okay?

happenpappen, 2019-10-12:
> Dont't really understand, is German language okay?

Du sollst den Würfel neu verbinden ohne ihn vorher zu löschen. Während des Verbindens sollst Du ihn etwas bewegen (Ich habe meinen Würfel vor 5 Minuten ausgepackt und stehe vor dem selben Problem ;-))

KlausHeynen, 2019-10-12:
Hi, hab jetzt den ganzen Mittag rum probiert. Null Erfolg.
Vielleicht kann am Montag jemand von Dresden Elektronik das mal erklären, schiesslich ist es explizit im Change log aufgeführt. Beste Grüße Klaus

Smanar, 2019-10-12:
"Versuchen Sie, Ihre Cuve während der Aufnahme zu bewegen" (Sorry google transalte)

KlausHeynen, 2019-10-12:
@Smanar, I did but no Change.

daniel-lo, 2019-10-14:
Ich häng mich hier mal ran. Wollte meiner Tochter grad Double-Tap vorführen und klar .. ging nicht. In die Objekte geschaut und Seitenkipper funktionerten 1a aber Double-Tap nicht .. tot. 2 Minuten später gings auf ein Mal wieder. Sind das Hänger die der Würfel selber erzeugt (Energiemanagement oder was auch immer) .. oder liegt das eher an deCONZ?

KlausHeynen, 2019-10-14:
@sdaniel-lo
Wie hast du denn das Ding installiert?

daniel-lo, 2019-10-14:
@KlausHeynen 
Wird in Phoscon als Schalter erkannt. Und dort kann man auch die Werte sehen .. wenn man auf die Gerätedetails schaut. Die Objekte les ich über ioBroker aus. Hoffe ich hab die Frage richtig verstanden.

manup, 2019-10-14:
@KlausHeynen hast du bereits die aktuelle deCONZ Version 2.05.69 installiert, hiermit sollte die Integration eigentlich gut klappen. Falls der Würfel mit einer älteren Version eingebunden wurde, diesen bitte löschen und neu hinzufügen.

KlausHeynen, 2019-10-14:
Ja, habe seit ca. 14 Tagen die 2.05.69 drauf und erst danach den Würfel hinzugefügt.Was mir auf gefallen ist das im Git einige Dateien vom 11. Oktober sind aber in der Release History ein Datum für die 2.05.69 vom 8. Sept. steht. Grüße Klaus Von: Manuel Pietschmann [mailto:notifications@github.com] Gesendet: Montag, 14. Oktober 2019 17:39An: dresden-elektronik/deconz-rest-plugin <deconz-rest-plugin@noreply.github.com>Cc: K. Heynen <KHe@futaba-eu.com>; Mention <mention@noreply.github.com>Betreff: Re: [dresden-elektronik/deconz-rest-plugin] Xiaomi Magic cube not visible in editor (#1958) @KlausHeynen hast du bereits die aktuelle deCONZ Version 2.05.69 installiert, hiermit sollte die Integration eigentlich gut klappen. Falls der Würfel mit einer älteren Version eingebunden wurde, diesen bitte löschen und neu hinzufügen.
Registered Office: Willich; Commercial Registry: Krefeld HRB 6061 Managing Director: Motoki TamuraUSt.-ID-Nr DE119263774
Important Note: This e-mail and any attachments are confidential, may contain trade secrets and may well also be legally privileged or otherwise protected from disclosure. If you have received it in error, you are on notice of its status. Please notify us immediately by reply e-mail and then delete this e-mail and any attachment from your system. If you are not the intended recipient please understand that you must not copy this e-mail or any attachments or disclose the contents to any other person. Thank you.

daniel-lo, 2019-10-14:
@KlausHeynen Wenn Du Probleme beim Paring hast .. bringe das Gerät in den Pairing-Modus durch längeres gedrückhalten des Knopfes (blinkt dann irgendwann 3 mal) .. und dann etwa alle 5 Sekunden einen deutlichen Druck auf die Tast .. wird dann meist mit einem Blinken bestätigt .. bis das Device in Phoscon erscheint und die Erfolgsmeldung in grün erscheint.

KlausHeynen, 2019-10-14:
@daniel-Io, habe es gemacht wie du es beschrieben hast - es funktioniert, vielen Dank!

weltmeyer, 2019-10-15:
Bei mir findet er den Schalter, in der webapp wird der Schalter unter Geräte/Schalter auch angezeigt.

Leider wird er nicht auf der Haupseite angezeigt.
Beim Verknüpfen mit einer Lichtergruppe wird er auch nicht angezeigt, dafür aber eine JS-Debugmeldung: " TODO: Handle Xiaomi Cube not both sensors available" ...
Was kann ich hier tun?

SwoopX, 2019-10-15:
Könnte vielleicht daran liegen, dass ein Sensor fehlt. Einfach noch einmal nach Sensoren suchen lassen und dabei den Würfel auf alle möglichen Arten bewegen.

Btw, passt hier vielleicht nicht ganz rein, aber fehlt nicht noch eine Geste, "drop"?

weltmeyer, 2019-10-16:
Okay. ich habe ihn nach einigen Gateway-Neustarts nun im Schalter-Editor für Gruppen.

Nun funktioniert außer drehen rechts/links aber nichts. 
Habe an/aus auf schütteln und auf Doppeltippen gelegt. aber er reagiert nicht darauf. 
Schalte ich die Gruppe ein kann ich manchmal durch rechts/links drehen die Helligkeit ändern... sind die Würfel mies oder geht der Rest einfach noch nicht?

closed 2020-02-20 by stale[bot]