dresden-elektronik/deconz-rest-plugin #1960: [Device support request] Develco water leak detector FLSZB-110

jukmanty, 2019-10-12:
I have attempted to add Develco water leak detector to hass.io home automation setup with deCONZ add-on. I've managed to get visible in hass.io, but state changes are not reflected to Home assistant correctly. Device is detected as Light. deCONZ REST API also reports in /api/<apikey>/lights/2

`    "state": {
        "alert": "none",
        "reachable": true
    }`

even though "Alarm 1" is true:

[image]

I see that the device is not in the list of Supported Devices, but recently base support for the Develco products has been added. Here is the basic info for this device:

[image]

[image]

Link to the technical documentation of the device:

https://www.develcoproducts.com/media/3804/flszb-110-technical-manual-water-leak-detector.pdf

SwoopX, 2019-10-15:
Hey there!

I might be able to help you out (on the weekend) but then you'd need to play my lab rat ;) In that sense, you must compile the plugin to have the code changes tested before issuing a PR...

I'd assume that you also have 2 sensor nodes, temperature and water? Do they report values via API?
Also, did you have any issues while pairing? Just asking since you got the values in I never was able to get with my smoke sensor and I don't get the point

jukmanty, 2019-10-19:
Yes, one device with water leak alarm (23 HA) and temperature measurement (26 HA). REST API reports alarm state, but incorrectly (always "none"). REST API does not report temperature at all. Pairing as Light always succeeded without problems, but I'm not sure if this is the right way for this kind of device. I could not add as as Sensor.

SwoopX, 2019-11-01:
You might wanna give it a try with the changes from my repo
[https://github.com/SwoopX/deconz-rest-plugin](https://github.com/SwoopX/deconz-rest-plugin)

You must compile the plugin with the given instructions.

jukmanty, 2019-11-02:
Thanks! I'll attempt to build it and try to pair the device.

jukmanty, 2019-11-06:
I was able to compile it to armhf architecture in raspbian/stretch docker container, but could not get it running in my RPi 3B+ + Raspbee + Hass.io environment. I modified official addon_core_deconz docker container https://github.com/home-assistant/hassio-addons/tree/master/deconz and copied compiled plugin to /usr/share/deCONZ/plugin/libde_rest_plugin.so Then I stopped and started container. Phoscon webapp starts, but REST API does not. Requests to /api are routed to webapp and it responds in html `<p>The file /api couldn't be found</p>`. I cannot login to Phoscon app, it does not accept any password. Also if VNC server port is configured then docker container exits with error code at VNC server startup.

If I add dbg-error=1 to deCONZ parameters I can see from the logs some errors during plugin lookup:

> 2019-11-06T18:53:20.963558000Z 20:53:17:688 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
2019-11-06T18:53:20.963868000Z 20:53:17:781 sub element size does not match real size
... 40 similar rows removed ...
2019-11-06T18:53:20.976747000Z 20:53:20:761 sub element size does not match real size
2019-11-06T18:53:20.977036000Z 20:53:20:764 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin

Can you assist how compiled plugin can be tested in my environment? Or is there something in the plugin code that prevents loading the library?

SwoopX, 2019-12-07:
@jukmanty with the changes committed, you have to wait for .72 release to give it another try. I'd also be interested in seeing if your described issue is resolved:

> Device is detected as Light. deCONZ REST API also reports in /api/apikey/lights/2
>
>"state": { "alert": "none", "reachable": true }
>
>even though "Alarm 1" is true

It might be required to fiddle around with the values a bit more. Develco documentation is at time not as specific as I'd wish it to be...

Btw, do you get an audio alert while "alarm 1" is set to true in deconz GUI? I'd assume the sensor gives audio alert?

jukmanty, 2019-12-08:
Hi! I'll give it a retry after next release finds it way to the hass.io deconz add-on. It usually happens quite soon after deCONZ REST plugin release.

For some reason I not able to write "Alarm 1" value. When Develco water leak detector is NOT joined to Zigbee network then I can read 23 0500 id 0x0002 (IAS Zone > IAS Zone Cluster > Zone Status > read) values. Changing "Alarm 1" to true and executing Zone Status Change Notification does not work, it gives "status 93". I quess sensor needs to be connected to network.

After setting group for "Warning device" in Phoscon app, link between raspbee controller and water leak detector appears in deCONZ GUI (VNC app). But after that I'm not able to read or write cluster attributes. Operations always timeout.

SwoopX, 2019-12-09:
> For some reason I not able to write "Alarm 1" value. When Develco water leak detector is NOT joined to Zigbee network then I can read 23 0500 id 0x0002 (IAS Zone > IAS Zone Cluster > Zone Status > read) values.

That doesn't really make sense. Devices must be paired and online to get reasonable values. Btw, Zone Status is a read only attribute if I'm not mistaken and describes what the device supports. If you want to have the alert value in the API to be changed, have you tried sprinkling it with water?

SwoopX, 2019-12-09:
Argh, missed it. Forget about the alert value for the light node. This is pretty much to trigger audio alerts. I'd assume that would go for the flood sensor as well. Check #1593 for more details.

What really will give you the indication if it's wet or not will be in the water sensor, as it has an alarm value (in contrast to the alert you mentioned). That one will get its info from the zone status attribute ...

jukmanty, 2019-12-09:
Previously I saw that HA 23 0500 0x0002 16-bitmap changed from 0x0030 to 0x0031 - i.e. "Alarm 1" bit is set to true when water leak sensor gets wet (yes, I did it manually dropping it on top of small amount of water).

Yes, it seems very strange that values cannot be read anymore when it's paired. I think before I managed to do this (looking at the screenshots in the first post in this issue). Perhaps I need to shutdown all other zigbee devices if it is a network communication issue.

I could not add this device as sensor previously, only as Light. Is it possible to pair water leak detector as sensor and get the "is wet" value via REST API if your pull request included in the next release?

jukmanty, 2019-12-09:
Triggering audio alert via REST API is not that relevant. Wet warning status (yes or no) is valuable.

SwoopX, 2019-12-09:
> I could not add this device as sensor previously, only as Light. Is it possible to pair water leak detector as sensor and get the "is wet" value via REST API if your pull request included in the next release?

Ah, now we're getting to the point :) I initially thought you would have had the sensors available. Have you tried that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2052#issuecomment-552202908 ? Anyway, the commit should do the trick. The ZHAwater sensor is the important one then.

> Previously I saw that HA 23 0500 0x0002 16-bitmap changed from 0x0030 to 0x0031 - i.e. "Alarm 1" bit is set to true when water leak sensor gets wet (yes, I did it manually dropping it on top of small amount of water). I'm not sure if this the one that indicates wet status or not.

In deconz GUI, that makes perfect sense.

jukmanty, 2019-12-10:
> Ah, now we're getting to the point :) I initially thought you would have had the sensors available. Have you tried that #2052 (comment) ? Anyway, the commit should do the trick. The ZHAwater sensor is the important one then.

No, I haven't. I'll try that procedure.

SwoopX, 2019-12-12:
New beta is out which contains the commits. Feel free to give it a try.

jukmanty, 2020-03-01:
Phoscon app now found a new sensor after resetting the device couple of times. It does not appear in the list of Connected sensors. But I see it now in deconz REST API as type ZHAWater sensor and also nicely in Home Assistant. It responded correctly by changing water value to true after dropping it into little amount of water.

But I cannot change water value back to false (Alarm 1 in deconz GUI) without resetting the device... I tried to execute some command via deconz GUI and also tried PUT "water": false via deconz REST API but none of these seem to work. Also reading state via deconz GUI seems to work only randomly.

I'm using deconz rest plugin 2_05_74 now.

jukmanty, 2020-03-01:
[image]

[image]

[image]

[image]

SwoopX, 2020-03-01:
Thaks for the feedback. As I see it, there's also alight being created and that's what you need to interact with in terms of the alert. Could you please also do a REST query against the corresponding light and provide the output as above for the sensor?

You cannot set any values for sensors. If I recall correctly, you must set the alert state from the light to none (I'll try to dig it out). Eventually, some attitional tweaking might be required here.

jukmanty, 2020-03-01:
Device is not exposed as light in deconz REST API. I see only the Raspbee Gateway and a pile of OSRAM plugs in `/api/<apikey>/lights`. I cannot add device as Light in Phoscon app anymore, it is found only as Sensor. In Home Assistant I can see unavailable, automatically added entity "switch.light_1" related to Water sensor in deCONZ integration, but value cannot be read or written.

SwoopX, 2020-03-01:
Ok, then that might be the bit of tweaking required. I can check the code to learn how that is handled for other water leak sensors.

For the smoke sensors, as you can use the siren also independently, a light had to be created where you can youse the alert attribute (see https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1593#issue-288454463 for reference).

jukmanty, 2020-03-02:
Thanks, I appreciate your effort! Previously it was found as Light in Phoscon app and it was exposed as light in REST API with alert attribute. Now it is found as Sensor and is exposed as sensor in REST API with water attribute, but not as light.

Device gives very loud audio alert when it gets wet.

SwoopX, 2020-03-03:
Ok, looks like the device is not considered from the lights perspective. Shouldn't be such a big deal to implement it according to the smoke sensors as well. When done, you should at least be able to turn of the alarm via REST API (at least I hope so).

Checked documentation and the technically proper way to disable the alarm would require some more effort and a specific zigbee packet to be send...

jukmanty, 2020-05-25:
Retested this now using deCONZ 2.05.76. I found it as sensor in Phoscon app. Water status can be read via REST API, as a sensor:

    "config": {
        "battery": 100,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 35,
    "etag": "480212ce4c0da81712b7f3aa4d26e1d7",
    "lastseen": "2020-05-25T20:07:54.784",
    "manufacturername": "Develco Products A/S",
    "modelid": "FLSZB-110",
    "name": "Water Sensor",
    "state": {
        "lastupdated": "2020-05-25T20:07:54.784",
        "lowbattery": false,
        "tampered": false,
        "water": true
    },
    "type": "ZHAWater",
    "uniqueid": "00:15:bc:00:33:00:03:a6-23-0500"

It is not exposed as light anymore and alert attribute is not there anymore, but that's fine.

Home Assistant integration works:

[image]

"light 1" is some dysfunction integration

Status changes to wet when it gets wet :)

Only thing that is missing is a way to set status back to `water: false`. I'm not able to do it via REST API nor deCONZ GUI. Even physical audio alarm off button does not do it. Resetting the sensor and re-adding it to Zigbee mesh does the job. That's ok too if disabling the alarm and wet status programmatically is complex. I hope there no need to see `water: true` status during the actual use of the device :) -> no need to set it back to the original state many times.

SwoopX, 2020-05-25:
Thanks for the feedback. It's strange that the light disappeared since no changes have been made in this regard. Maybe you just run light search again while reading the basic cluster in deconz GUI or re-read the simple descriptors.

If you'd send, to the associated light, an API request with payload `alert: none`, the alarm condition should be ceased. That should be similarly doable via deconz GUI as well (if I'm not mistaken, there's a button for that).

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix