dresden-elektronik/deconz-rest-plugin #1962: Innr light strips flash shortly when dimming up

Openhabber, 2019-10-14:
Hi,

i have some innr light strips. 
When dimming them via openhab (or dirwctly through rest-api) from off to a value, they instantly go to the set value, then go back off and dim up again.

It looks like they are flashing beforw dimming up. 

It does not happen if
-they are just turned on back to previous value
-they are dimmed through the phoscon web app
-transitiontime is set to 0 in the REST request

The request is as follows:
http://localhost:5100/api/A697051788/lights/15/state

Body:
{"on":true,"bri":170,"transitiontime":4}

it does not happen if coupled to the philips hue bridge.

Openhabber, 2019-10-15:
Is there anything else needed? Thid is clearly a bug.

ChrisHae, 2019-10-22:
Hi, there are a few things that you could try:

-  transition time is in 1/10 seconds. If you want to dimm up in 4 seconds then the request should be: {"on":true,"bri":170,"transitiontime":40}

- you could use 2 requests if that is possible:
{"on":true,"bri":10,"transitiontime":0}
{"on":true,"bri":170,"transitiontime":40}

- perhaps it is needed to also turn down the brightness to 0 when you put your lights off. That could prevent the flashing when turning them on again. so when putting them off add the "bri" parameter:
{"on":false,"bri":0}

Openhabber, 2019-10-22:
Thanks for your feedback. 

The problem is, i am using openhab.
Openhab always uses a transitiontime of 4. All other lights (osram, philips, tradfri) behave correctly. I have no chance in changing that.

They worked perfectly on the hue bridge before. 

Of course i could try those commands for debugging, but it will not help with that they behave different than other lights. 

Do you need me to try any of those commands so you can fix this issue?

ChrisHae, 2019-10-22:
It is possible to add a hotfix to deCONZ that will treat innr lights as though the command would have a transition time of 0. But some more testing and research is needed so it could take some time.

If you want to help could you try some commands and post the behavior of the light:
{"on":true,"bri":170}
{"on":true,"bri":170,"transitiontime":0}
{"on":true,"bri":170,"transitiontime":10}
{"on":true,"bri":170,"transitiontime":30}

kasteleman, 2019-10-22:
What command is used to dim the lights?
I had some “issues” with some INNR devices and at the end used
MoveToLevelWithOnOff To prevent the behavior as described.

Oh and another thing: Since INNR introduced the Zigbee 3.0 bulbs there is
also a attribute OnOffTransitionTime that defines a transition when bulb is
turned on or off, but does not apply when dimlevel is changed If bulb is
already on.

ChrisHae, 2019-10-22:
thx @kasteleman, this is some good info. We will consider this when researching the problem.

kasteleman, 2019-10-22:
If you need any other info or have specific questions regarding INNR
products, let me know. I can always drop those at the INNR product
developers.

ebaauw, 2019-10-22:
My (ZLL) innr PL 100 and UC 110 have had _OnOffTransitionTime_ since I got them, several years ago.  Unfortunately, they're not firmware upgradable.  The firmware hangs when _Breathe_ is followed by _Stop_ too early (what the Hue bridge and deCONZ send on setting `state.alert` to `select`), see #100.  I used to have some (ZLL) innr RB 162 bulbs, but replaced them by Hue bulbs, as their routing seemed incompatible with deCONZ', see #806.

The deCONZ REST API plugin handles `state.on`, `state.bri`, and `state.transitiontime` differently than the Hue bridge, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1111#issuecomment-456464872.  Still haven't found the time to refactor the deCONZ logic.  Note: I haven't checked the latest version of the Hue bridge firmware.

kasteleman, 2019-10-22:
> state.transitiontime differently than the Hue bridge, see #1111 (comment)

That’s the reason I used the other command as INNR told me that there isn’t
a new firmware available

Openhabber, 2019-11-02:
> It is possible to add a hotfix to deCONZ that will treat innr lights as though the command would have a transition time of 0. But some more testing and research is needed so it could take some [...]
Flashes shortly, turns off and dims up to 170
> {"on":true,"bri":170,"transitiontime":0}
Instantly turns on to 170
> {"on":true,"bri":170,"transitiontime":10}
Flashes, turns off and dims up slow
> {"on":true,"bri":170,"transitiontime":30}
Flashes, turns off and dims up even slower

I hope you guys can fix this :/

ebaauw, 2019-11-03:
We really need to know how the innr reacts to the various ZigBee commands, notably _Move to Level_ and _Move to Level (with On/Off)_.  Can you play around in the deCONZ GUI (_Cluster Info_ panel for the _Level Control_ cluster)?

Does the light actually turn off after the flash?  I would expect it to turn on at max level (or maybe previous level?), then instantly move to minimum brightness, and then transition to given brightness.  Might be hard to see the minimum brightness when you've just been blinded by max (previous?) brightness.

I've seen various lights that don't implement _Move to Level (with On/Off)_ correctly and the REST API translates `{"on": true, "level": 170}` to this, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1111#issuecomment-456464872.  I would like a generic workaround, rather than a specific one for this innr light, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1111#issuecomment-495895013.

Openhabber, 2019-11-03:
I think the best approach would be to do it the same way the hue brudge does?
From your comment:
The Hue bridge handles {"on": true, "bri": 1} by sending a Move to Level (with On/Off) (2, 0) to turn the light on at minimum brightness, and then a Move to Level (1, 4) to set the brightness to 1 (see above table). The deCONZ REST API plugin sends a single Move to Level (with On/Off) (1, t).

As the innr lighrs behaved correctly with the hue bridge, and most brands will make surw to be compatible with the hue bridge. 

What do you think?

kasteleman, 2019-11-03:
Keep in mind that not all INNR lights behave the same. There are some
products that “suffer” from a firmwarebug That can’t be fix by upgrading
the firmware, there is no update available.

ebaauw, 2019-11-03:
> What do you think?

Please replay the Hue bridge commands using the deCONZ GUI to confirm that this fixes (or rather circumvents) the issue.

> Keep in mind that not all INNR lights behave the same.

Different types of lights have different firmware.  The _Breathe_ bug only appears on the xx 110 controller firmware.  The routing issues only surface on the RB 162.

Openhabber, 2019-11-03:
Okay, i played around a little. It seems the flashing is at previous on brightness.

However, move to level to 0/1 with and without on/off did not do the trick.

I could reproduce it only, if the lamp was turned off via "off" command. After that, there was no way to prevent the flickering. (allthough an api call with transitiontime 0 does indeed turn them on without flickering? **What commands is sent then?**)

It also flickered if movetolevel with onoff (0,0) was send before, allthough the lamp reported a currentstate of 1. It did flicker with previous brightness nontheless. 

If the lamp was turned off via movetolevel with onoff(0,t), it did turn on correctly with movetolevel with onoff(b, t) under all circumstances. 
The transitiontime only affects the dimup time after the short flashing. I cannot tell if it turns back off, i think it just goes to minimum brightness after flashing and then dims up. 

Maybe turning it off is different than the hue bridge handles it? 

By the way, these are FL 130C Strips. Sw build is 0x20001470.

ebaauw, 2019-11-12:
Another thought: Do your innr lights support the _On Level_ attribute (0x0011) in the _Level Control_ cluster (0x0008).  If this is set to 254 (which seems to be the default for mine), this causes the light to turn on at maximum brightness.  Set it to 1 for minimum brightness, or to 255 for previous brightness (before turning it off).  Unfortunately, this attribute isn't stored in non-volatile memory, so you need to redo this setting after power-cycling the light.  The attribute is not exposed by the REST API plugin.

kasteleman, 2019-11-12:
To add some additional info, don’t know if I already mentioned it, the
Zigbee 3.0 INNR bulbs store the last state. So if you cut the power on them
and restore the power, they go to the previous dim or color.

Openhabber, 2019-11-12:
Why is it not implemented the way philips hue does it? Wouldnt that be an option?

kasteleman, 2019-11-12:
I don’t know, but why should Philip’s Hue be the standard? 😉

Openhabber, 2019-11-12:
Because the innr light behaves correctly with it, and most likely is the one every manufacturer checks against.

I dint have a bridge anymore or the possibility to track it, but the suggestions were try this and that, but we already know it works if it is used the way other bridges speak with it.

I wrote before:
1. it does not flicker, if only the movetolevel with onoff(0,t) is used for turning it off
2. It flickers every time, if turned off via off command exept 3
3. It does not flicker if turned off via ofd command and turned on via api call, sith transition time 0. I do not know. What commands are sent to the lamp then.
4. It does mot happen if using the sliders in the phoscon ui

So, there are three things you could find out:
1. What commands are sent on api call with transitiontime =0
2.is it possible the hue bridge only uses movetolevel with on off for turning off and does not use the off command
3.is it an option to not use the off command on this innr light, but the movetolevel with onoff (0,t) instead

The option mentioned before does not seem to be relevant, as the light did not flash with max brightness, but the last brightness before turning off via off command.

Hope it helps :)

lukicsl, 2019-11-14:
I closed my issue.
But just to mention this behavior can be reproduced In Phoscon group view, see the closed issue.

Openhabber, 2019-11-19:
Any update? Can you fix it? Please see my last comment.

Openhabber, 2019-12-01:
Problem still persists.....

Openhabber, 2019-12-09:
Im a little sad this is not getting fixed...

Openhabber, 2019-12-18:
This is sad. I thought it would be a good idea to buy your zigbee solution. My bed flashes every time i turn it on like it are old fluorescent tubes. This is so annoying.
And I thought innr lights are quite common. 

:(

lukicsl, 2019-12-19:
@Openhabber  try a q workaround with linking openhab with nodered.
I  made it working without flickering by modifying  transitionTime to 0 before switching on:
[image]
`[{"id":"8d895c51.2a165","type":"hue-group","z":"2758b5a.fbf574a","name":"Arbeitszimmer","bridge":"825931a5.ed46b","groupid":"26","colornamer":true,"x":600,"y":1720,"wires":[["af7180eb.669cd","44db5ec9.e52a6"]]},{"id":"af897ac0.9f6db8","type":"function","z":"2758b5a.fbf574a","name":"lightDimmFnc","func":"if (msg.payload === 0)\n{\n    msg.payload={ \"on\":false}\n}\nelse \n{\n    msg.payload={\"on\": true,\n                 \"brightness\":msg.payload, \n                 \"transitionTime\":\"0\"}\n}\nreturn msg;\n","outputs":1,"noerr":0,"x":400,"y":1720,"wires":[["8d895c51.2a165"]]},{"id":"12448261.77272e","type":"function","z":"2758b5a.fbf574a","name":"dimmer-feedback-fnc","func":"\nif (msg.payload.on)\n    return {payload: msg.payload.brightness};\nelse\n    return {payload: 0};\n","outputs":1,"noerr":0,"x":280,"y":1800,"wires":[["d4e91d1e.aad8e","8cf3f439.f70bc8"]]},{"id":"c8be68eb.1edee8","type":"link in","z":"2758b5a.fbf574a","name":"DiningGroup","links":["af7180eb.669cd"],"x":135,"y":1800,"wires":[["12448261.77272e"]]},{"id":"af7180eb.669cd","type":"link out","z":"2758b5a.fbf574a","name":"","links":["c8be68eb.1edee8"],"x":815,"y":1760,"wires":[]},{"id":"f4a75126.dc6e8","type":"openhab2-in","z":"2758b5a.fbf574a","name":"StudyRoomLightTop","controller":"e4499e9e.9fd89","itemname":"StudyRoomLightTopHG","x":210,"y":1720,"wires":[["af897ac0.9f6db8"],[]]},{"id":"d4e91d1e.aad8e","type":"openhab2-out","z":"2758b5a.fbf574a","name":"StudyRoomLightTop","controller":"e4499e9e.9fd89","itemname":"StudyRoomLightTopHG","topic":"ItemUpdate","payload":"","x":520,"y":1800,"wires":[[]]},{"id":"44db5ec9.e52a6","type":"debug","z":"2758b5a.fbf574a","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":870,"y":1720,"wires":[]},{"id":"8cf3f439.f70bc8","type":"debug","z":"2758b5a.fbf574a","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":530,"y":1900,"wires":[]},{"id":"825931a5.ed46b","type":"hue-bridge","z":"","name":"vevedock-09","bridge":"192.168.3.152","key":"72B99A4ECE","interval":"6000"},{"id":"e4499e9e.9fd89","type":"openhab2-controller","z":"","name":"vevedock-64","protocol":"http","host":"vevedock-64","port":"8080","path":"","username":"","password":""}]`

Alternatively what I tried is setting lights on and off by rules. I the rule then you always set light to 1% before setting it to 0% (OFF), and never setting it directly to OFF.

hope this helps

Openhabber, 2019-12-19:
I do not have node red.
Is this a bug that cannot be fixed or what is going on?
As said, philips hue bridge handled these strips without issues. I gave you every information i can.

manup, 2019-12-21:
My co-worker was looking into this issue, the main reason that this particular innr problem isn't resolved yet is that we would need to check such a change against all lights from all vendors which is very time consuming. Other lights like IKEA and Xiaomi have bugs related to transition time as well, some are pretty serious that the device can be brought in a non responding state until the device gets power-cycled.

So we need to be really careful with these kind of changes, it may look simple at first but the whole picture is more complex. We can add small quirks for unicast commands, but for group commands which are received by all devices broad testing is inevitable.

Openhabber, 2019-12-21:
Thanks for the update. I was hoping you would implement such changes device-specific.
Im not sure about group commands, as i do not use deconz groups. 

Also im still unsure why it does not happen if dimmed from the phoscon frontend then?

manup, 2019-12-21:
When my memory serves me right the Phoscon App sends only plain on/off command without transition time. This can be viewed in the browser developer tools network tab (F12).

closed 2020-04-26 by stale[bot]