dresden-elektronik/deconz-rest-plugin #1963: deCONZ app not connecting

daniel-lo, 2019-10-14:
Hi, I managed to run a dektop via VNC. I can start deCONZ app and the web apps work (buttons top right). ConBee II is displayed as device but I cannot connect .. nothing happens when I press the Connect button and the top bar shows Not Connected in red.

Xieldan, 2019-10-14:
Same issue here - cannot connect to my Raspbee in Deconz anymore.

I am running Deconz 2.05.69, I noticed that Deconz did a firmware upgrade yesterday. In Phoscon, it shows the gateway as "not connected".

Curiously, some sensors and a tradfri remote are still visible in Homekit/Homebridge, but all lights are gone. I have tried loading my latest gateway backup, but without any luck.

Xieldan, 2019-10-17:
Fixed it now by flashing the latest firmware again as per:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

LorisAlbanese, 2019-10-20:
Hi, Same issue here.
Buyed and installed raspbee 2 day ago. After HA integration, last evening no one Zigbee device was available. Reboot, reset and than landed here.
via deCONZ gui i can't connect to device. via Phoscon App only login work, every action end ina failure.
So i tried to follow step to reset raspbe but i can't go onward this output:

$GCFFlasher_internal -f deCONZ_Rpi_0x26330500.bin.GCF
GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
action: update firmware after 2 ms
� (104 ms)
�  (112 ms)
� � (120 ms)
� �  (127 ms)
� � � (135 ms)
� � �  (143 ms)
� � � � (150 ms)
� � � �  (158 ms)
� � � � � (166 ms)
� � � � �  (173 ms)
� � � � � � (181 ms)
� � � � � �  (189 ms)
� � � � � � � (196 ms)
� � � � � � �  (204 ms)
� � � � � � � � (212 ms)
� � � � � � � �  (220 ms)
could not sync with bootloader

Phoscon App say is not connected.
I'm using Pi4 , Wiring-pi 2.52,  over 4.19.79-v7l+ kernel.

daniel-lo, 2019-10-20:
In the meantime I have figured out that you can start either a gui or a non-gui version (see documentation a few pages up). So the problem on my side was that the deconz-gui was not started. But due to problems with the initial setup I obviously also had multiple installations .. which I had to clean up (somehow .. I'm not so much into Linux) .. and ultimately it worked. Though .. if you look at issue 1980 you will notice that I had quite some struggle .. now IMO everything is clean .. phoscon works .. gui works .. cpu load is low and objects react. On my way I did not do any flashing.

LorisAlbanese, 2019-10-24:
thank you. Problem is quite simple. deCONZ gui does not work if systemd application "decons" is running. Stopping (systemctl  stop deconz.service) deCONZ gui work like a charm.
problem solved.

MichaelHeimann, 2020-02-04:
> Fixed it now by flashing the latest firmware again as per:
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

I had to do the same - pretty frightning.
I was able to see all switches and sensors in the webgui but "no lights connected" and deCONZ couldn't connect.
The Connect button in deCONZ just went grey and came back without success.
Reflashing and it works again.

WAF level decreased today :(

Xieldan, 2020-04-19:
Issue is back today. This time, flashing does not work ("Could not sync with bootloader)". Hope I can fix this, but I am running out of ideas. :-(

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

@Xieldan if this persists for you, please open a new issue.

closed 2020-06-05 by Mimiix

gregorschulz, 2020-06-23:
> I had to do the same - pretty frightning.
> I was able to see all switches and sensors in the webgui but "no lights connected" and deCONZ couldn't connect.
> The Connect button in deCONZ just went grey [...]

Updated to 2.05.78 / 26610700

after Update no lights/switches/sensors connected anymore. had to connect everything new by hand. very bad for devices not reachable easily.

WAF was in extrem danger too.

Absolutely unacceptable. This destroys confidence in this solution.

Mimiix, 2020-06-23:
@gregorschulz As your issue lacks context and is hijacking another issue, i'd like to ask you to open a [Bug Report](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=Bug+report&template=bug_report.md&title=)

henkkakay, 2021-01-23:
A quick note for anyone who may have a similar setup that I seem to have spotted a pattern: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1719#issuecomment-766088679

ben69uaq, 2023-04-15:
Hi,
I got exactly the same issue and my problem was there was already one software connected to my ConBee 2.
It seems that there cannot be 2 software connected to ConBee 2.

Since zigbee2mqtt was connected to ConBee 2, I was not able to connect using deconz.
Once I stopped zigbee2mqttt, I'm able to connect ConBee 2 using deconz.

I hope it will help.