dresden-elektronik/deconz-rest-plugin #1971: API Key and change deConz standard port 80

Luigi-01, 2019-10-16:
Hi,
I have been reading your manual regarding the API Key.
Not a success. when I follow your instruction, the outcome is always **"unauthorized user**".
I can unlock whatever I want, but nothing helps. Please advice.
Presently the Conbee II is useless without API Key.
Second where can I change the port setting. Every program wants to use port:80
That does not work.

Smanar, 2019-10-16:
To change port https://github.com/dresden-elektronik/phoscon-app-beta/issues/26

Witch one url are you using to make request for API key ?

Luigi-01, 2019-10-17:
Sorry, that does not work:
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
qt.qpa.screen: QXcbConnection: Could not connect to display 
Could not connect to any X display.

I assume that most people are running headless. which file can I edit.
Second how to solve the API key. This is a basic requirement. 
Why not use a login name and password as all the others do. Simple but it works.
-----------------------------------------------------------------------------
    Status Code: 403 Forbidden
    access-control-allow-origin: *
    content-length: 70
    content-type: application/json; charset=utf-8
---------------------------------------------------------------------------------
[{
  "error": {
    "address": "/",
    "description": "unauthorized user",
    "type": 1
  }
}]
----------------------------------------------------------------------------------

Please advice,

Luigi-01, 2019-10-17:
Sorry, for bold text, this was not my intention.

Smanar, 2019-10-18:
Yep, it's for the GUI version but have you take a look in the headless system file ?
>To overwrite the headless mode the deconz.service file instead of deconz-gui.service must be overwritten.

Have you the request send ? with the url and the data ? Take a screenshoot if you can.

You don't forget to "authenticate app" ?

closed 2019-10-24 by Luigi-01