dresden-elektronik/deconz-rest-plugin #1973: Device Support Request HEIMAN Gaz Sensor Heiman HS1CG

laurent941, 2019-10-16:
Hi,

Could you add support for EIMAN Gaz Sensor Heiman HS1CG ( european version )

[image]

[image]

[image]

[image]

thank you in advance

laurent941, 2019-11-18:
no answer ?

andriej, 2019-11-18:
I'm also interested, just have ordered it.

andriej, 2019-12-03:
It just arrived. Paired but no further options to use it...

michis0806, 2020-01-17:
> It just arrived. Paired but no further options to use it...

How did you pair it? In Phoscon?

andriej, 2020-01-17:
No, through the VNC UI. I can see it in mesh but even without name. 
Currently unusable :(

michis0806, 2020-01-17:
Okay, thanks though. ;)

andriej, 2020-02-02:
FYI: [Here is the instruction](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2403#issuecomment-581164140) from different user how to pair it so at least it can be seen in API, not in Phoscon tho

closed 2020-06-09 by stale[bot]