dresden-elektronik/deconz-rest-plugin #1974: Device Support Request HEIMAN Smart motion Sensor HS1MS-E

laurent941, 2019-10-16:
Hi,

Could you add support for EIMAN Smart motion Sensor HS1MS-E ( european version )

[image]

[image]

[image]

[image]

laurent941, 2019-11-18:
no answer ?

Deinara, 2019-12-02:
I also bought one of these thinking they were supported as shown on the [Supported Devices](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#supported-sensors) page. Seems I have the -E variant which is not supported. 

Is there a point in doing any verification of the data you presented with my device? (and how would I go about creating those screenshots? I can't seem to find a guide on the wiki)

Deinara, 2019-12-05:
I can confirm that my device is the exact same as the one in this issue.

For extra info, I've included the cluster info below:
Basic:
[image]

Power configuration:
[image]

Identify:
[image]

IAS Zone:
[image]

OTAU:
[image]

Hypfer, 2019-12-17:
Same device. Same issue.
I suppose it's just a matter of whitelisting this variant.

It might be interesting to compare IDs etc with a non -E version.

Harry-1976, 2020-03-26:
Would like to see this sensor being supported. 
Heiman probably only ships out the E-version to Europe now anyway.

sylviepipi, 2020-04-17:
same problem, any progress ?

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

Mimiix, 2020-06-06:
Re opening until merged.

laurent941, 2020-06-06:
it's really sad that dresden-elektronik is not compatible with most of the heiman products in european version none of my tickets were managed

laurent941, 2020-06-06:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1973
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1974
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1972
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1975

Mimiix, 2020-06-07:
@SwoopX

SwoopX, 2020-06-07:
deconz can handle most of the Heiman devices and therefore supports them. An exception might be the switch mentioned before.

laurent941, 2020-06-07:
Thank you for your response SwoopX
but I assure you the Heiman devices I mentioned above are not yet supported. 
How can we help to support what information do you need?

#1973
#1974
#1972
#1975

SwoopX, 2020-06-07:
So, the first two is ok. For the motion sensor, this one came just in lately as the modelID had changed for newer versions. However, as HA downgraded thwir addon and subsequently deconz as well, support git kicked out again.

So that leaves us with the switch. It requires some debugging and ideally the ability to compile a dev version of deconz REST API plugin for testing.

laurent941, 2020-06-07:
hello swoopx  I'm not a developer but I own both versions of switch tell me how to do and I'll give you all the connexion sceenshot I have.  :)

SwoopX, 2020-06-07:
Well, the challenge is that you're using HA if I understood correctly. Is deconz running in docker or on bare metal?

laurent941, 2020-06-07:
Hello SwoopX

I have a synology and a dockworker.
I'm using Conbee II  Zigbee Gateway  and t phoscon with
https://registry.hub.docker.com/u/marthoc/deconz/

And HA with 
https://registry.hub.docker.com/u/homeassistant/home-assistant/

an Aqara Homekit gateway thanks to API to HA
so I'm combining 

Zigbee conbee 2 deconz
Aqara in homekit version 
and HA in homekit and zigbee with conbee 2

and in the end I manage to connect many manufacturers together.

What I find a pity and that heiman is currently one of the only ones to have products in the European format and with very attractive prices.

For exemple I have been using Smart Metering Plug HEIMAN ( french version ) for almost 1 year at less than 30 USD per plug and the power monitoring works perfectly. in HA
.
Look at their smart scene switch.
http://www.heimantech.com/product/?type=detail&id=22

I think many people will be happy to use more Heiman products as they are often the same product in the UK / european / German and French / south european format.

Mimiix, 2020-06-08:
@SwoopX @laurent941  As this original issue is solved, 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1972 

I re-opened that one.

closed 2020-06-08 by Mimiix

SilverWx, 2020-08-26:
Doesn't work for me, pairing ok but the state keep on RAS even if a move is detected, any tips ?